hoylabs/OpenDTU-OnBattery #379: MQTT victron HA auto config for sensor.panel_yield_yesterday is wrong

alexz707, 2023-08-11:
### What happened?

In the HA logs I have found an entry:

`Entity sensor.panel_yield_yesterday (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

I guess "stat_cla":"measurement" should be "stat_cla":"total_increasing" ?

### To Reproduce Bug

Add a victron MPP Tracker to dtu with enabled ha auto discovery.
Have a look into the logs.

### Expected Behavior

Sensor gets added without an error log entry.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	026bca9

### Relevant log/trace output

```Shell
Auto discovery config of Panel yield yesterday:
`{"name":"Panel yield yesterday","stat_t":"solar/victron/123/H22","uniq_id":"123","unit_of_meas":"kWh","dev":{"name":"Victron(123)","ids":"123","cu":"http://10.10.10.35","mf":"OpenDTU","mdl":"SmartSolar MPPT 150|35","sw":"g026bca9"},"exp_aft":15,"dev_cla":"energy","stat_cla":"measurement"}`

Auto discovery config of Panel yield yesterday:
`{"name":"Panel yield today","stat_t":"solar/victron/123/H20","uniq_id":"123_panel_yield_today","unit_of_meas":"kWh","dev":{"name":"Victron(123)","ids":"123","cu":"http://10.10.10.35","mf":"OpenDTU","mdl":"SmartSolar MPPT 150|35","sw":"g026bca9"},"exp_aft":15,"dev_cla":"energy","stat_cla":"total_increasing"}`
```

madmartin, 2023-08-21:
I will investigate this.

philippsandhaus, 2023-08-22:
total_increasing is also wrong I guess. The reading will not necessarily increase from day to day. And as it is also not a continues measurement like Yield total. I guess the state class should be "total"

From https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes for state Classe measurement:

_The state represents a measurement in present time, not a historical aggregation such as statistics or a prediction of the future. Examples of what should be classified measurement are: current temperature, humidity or electric power. Examples of what **should not be classified as measurement**: Forecasted temperature for tomorrow, **yesterday's energy consumption** or anything else that doesn't include the current measurement. For supported sensors, statistics of hourly min, max and average sensor readings is updated every 5 minutes._

madmartin, 2023-08-22:
>  I guess the state class should be "total"

That's my opinion, too. Pullrequest will come soon.

madmartin, 2023-08-22:
see #394

closed 2023-08-28 by helgeerbe