hoylabs/OpenDTU-OnBattery #387: Huawei R4850 CAN Kommunikation bricht ab

indie89, 2023-08-16:
### What happened?

Seit der ersten Version im August (und seitdem bei allen weiteren) habe ich nach ein paar Minuten nach Start von openDTU, in denen die CAN Kommunikation mit dem Huawei R4850 gut funktioniert, einen Verbindungsabbruch. Und auf die "Sending Requests" kommt keine Antwort mehr.

Wenn ich wieder die letzte Version im Juli installiere (ist glaub die 24.7.23 Version) funktioniert alles wieder super. Ist mit der August Version etwas verändert worden, was dieses Verhalten hervorrufen kann? Log siehe unten (ist aber wenig aussagekräftig). Wichtig beim LOG: ich habe hinten raus mein Skript, das das Laden der Batterie steuert, pausiert, um zu schauen, ob es daran liegt. Daher kommt dann kein "Limit Voltage" und "Limit Current" mehr...

Grundsätzlich komme ich mit der Version 2023.07.24 gut zurecht, ich möchte nur nicht auf alle Ewigkeit von Weiterentwicklungen abgeschnitten sein ;-).

### To Reproduce Bug

Ab den August Versionen bricht die CAN Kommunikation mit dem R4850 nach ein paar Minuten automatisch ab.

### Expected Behavior

Die CAN Kommunikation sollte stabil aufrechterhalten werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

seit 2023.08.01 bis inkl 2023.08.10.post1 (also der neuesten Version)

### Relevant log/trace output

```Shell
09:15:13.127 > [HUAWEI********************* Sending request
09:15:13.242 > [HuaweiCanClass::onReceive] In:  229.94V, 1.27A, 408.52W
09:15:13.242 > [HuaweiCanClass::onReceive] Out: 52.91V, 5.95A of 5.90A, 315.96W
09:15:13.242 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 36.0C09:15:15.598 > Admin AP remaining seconds: 20 / 180
09:15:18.119 > [HUAWEI********************* Sending request
09:15:22.253 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:15:22.253 > Limit Voltage: 55.400002 V
09:15:22.342 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:15:22.342 > Limit Current: 6.000000 A
09:15:23.125 > [HUAWEI********************* Sending request
09:15:23.197 > [HuaweiCanClass::onReceive] In:  229.94V, 1.27A, 357.72W
09:15:23.197 > [HuaweiCanClass::onReceive] Out: 52.91V, 6.06A of 6.00A, 318.37W
09:15:23.197 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 36.0C09:15:25.621 > Admin AP remaining seconds: 30 / 180
09:15:28.132 > [HUAWEI********************* Sending request
09:15:28.286 > [HuaweiCanClass::onReceive] In:  229.94V, 1.27A, 361.84W
09:15:28.286 > [HuaweiCanClass::onReceive] Out: 52.91V, 6.02A of 6.00A, 318.37W
09:15:28.286 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 36.0C09:15:32.186 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:15:32.186 > Limit Voltage: 55.400002 V
09:15:32.354 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:15:32.354 > Limit Current: 6.000000 A
09:15:33.134 > [HUAWEI********************* Sending request
09:15:33.191 > [HuaweiCanClass::onReceive] In:  229.94V, 1.27A, 361.84W
09:15:33.191 > [HuaweiCanClass::onReceive] Out: 52.91V, 6.06A of 6.00A, 319.61W
09:15:33.191 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 36.0C09:15:35.627 > Admin AP remaining seconds: 40 / 180
09:15:38.139 > [HUAWEI********************* Sending request
09:15:38.223 > [HuaweiCanClass::onReceive] In:  229.31V, 1.27A, 363.25W
09:15:38.223 > [HuaweiCanClass::onReceive] Out: 52.91V, 6.11A of 6.00A, 325.61W
09:15:38.223 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 36.0C09:15:42.221 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:15:42.221 > Limit Voltage: 55.400002 V
09:15:42.372 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:15:42.372 > Limit Current: 6.100000 A
09:15:43.140 > [HUAWEI********************* Sending request
09:15:45.647 > Admin AP remaining seconds: 50 / 180
09:15:48.311 > [HUAWEI********************* Sending request
09:15:48.462 > [HuaweiCanClass::onReceive] In:  229.12V, 1.27A, 371.47W
09:15:48.462 > [HuaweiCanClass::onReceive] Out: 52.91V, 6.15A of 6.10A, 324.47W
09:15:48.462 > [HuaweiCanClass::onReceive] Eff: 88.0%, Temp in: 25.0C, Temp out: 37.0C09:15:52.166 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:15:52.167 > Limit Voltage: 55.400002 V
09:15:52.379 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:15:52.379 > Limit Current: 6.100000 A
09:15:53.292 > [HUAWEI********************* Sending request
09:15:55.663 > Admin AP remaining seconds: 60 / 180
09:15:58.297 > [HUAWEI********************* Sending request
[... 188 lines cut ...]
09:20:52.258 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:20:52.258 > Limit Voltage: 55.400002 V
09:20:52.258 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:20:52.258 > Limit Current: 7.100000 A
09:20:57.251 > [HUAWEI********************* Sending request
09:21:02.282 > [HUAWEI********************* Sending request
09:21:02.282 > Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
09:21:02.282 > Limit Voltage: 55.400002 V
09:21:02.282 > Received MQTT message on topic: solar/huawei/cmd/limit_online_current
09:21:02.282 > Limit Current: 7.200000 A
09:21:07.256 > [HUAWEI********************* Sending request
09:21:12.263 > [HUAWEI********************* Sending request
09:21:17.274 > [HUAWEI********************* Sending request
09:21:22.270 > [HUAWEI********************* Sending request
09:21:27.280 > [HUAWEI********************* Sending request
09:21:32.305 > [HUAWEI********************* Sending request
09:21:37.291 > [HUAWEI********************* Sending request
09:21:42.302 > [HUAWEI********************* Sending request
09:21:47.303 > [HUAWEI********************* Sending request
09:21:52.308 > [HUAWEI********************* Sending request
09:21:57.308 > [HUAWEI********************* Sending request
09:22:02.321 > [HUAWEI********************* Sending request
09:22:07.324 > [HUAWEI********************* Sending request
09:22:12.335 > [HUAWEI********************* Sending request
```

bf4ed, 2023-08-16:
Kann ich bestätigen, ich hab das gleiche problem mit August Versionen.

MalteSchm, 2023-08-18:
@indie89 @bf4ed 
Wenn ihr möchtet könnt ihr mal diesen Branch bauen und testen: https://github.com/MalteSchm/OpenDTU-OnBattery/tree/huawei_interval_fix

In dieser Version läuft die Huawei CAN Bus Kommunikation in einem eigenen Thread. Damit werden bei mir typischerweise alle Daten empfangen. Ich habe beim Testen allerdings dieses Stabilitätsproblem festgestellt. https://github.com/helgeerbe/OpenDTU-OnBattery/issues/381 
Ich habe damit normalerweise so Uptimes von mehreren Stunden aber es ist nicht so stabil wie vor der Änderung. Ich vermute das diese Abstürze mit der Anzeige des Webinterfaces zusammenhängen.
Ich konnte aber da noch nicht weiter machen.

Der Modus bei dem das Huawei Netzteil automatisch geregelt wird ist ungetestet. Allgemein habe ich nur das setzen von Werten über das Webinterface geprüft, per MQTT sollte aber auch gehen. Bitte nur unter Aufsicht nutzen.

indie89, 2023-08-29:
Danke, malte, ich probiere das gerne aus! Leider hab ich keine Ahnung, wie ich den branch baue / kompiliere... Hättest du da ne Link zu einer Erklärung/kurzem Tutorial? Ich nutze sonst immer die pre compiled versions...

MalteSchm, 2023-08-30:
[firmware.zip](https://github.com/helgeerbe/OpenDTU-OnBattery/files/12479084/firmware.zip)
Ich habe das mal hier angehängt. Das ist meine aktuelle FW. Achtung: Diese unterstützt keine HTTPS requests beim Power Meter (nur HTTP). Weil: https://github.com/helgeerbe/OpenDTU-OnBattery/issues/381

Ich hatte damit in den letzten 3 Tagen einen Restart den ich leider nicht geloggt habe.

MalteSchm, 2023-10-04:
Fixed in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/454
I think we can close this issue

indie89, 2023-10-05:
The combination of 3 actions fixed the problem (where action 3 might also work stand alone as a solution):
1. Installing a capacitor 220yF to the power connection.
2. Use Vin instead of USB-C to power the ESP (see https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/458)
3. Move Huawei CAN communication to separate thread (see https://github.com/helgeerbe/OpenDTU-OnBattery/pull/454)

commit 40203f73 (2022-12-05): Fix #387: Allow a little bit more loss before HASS shows unavailable
closed 2023-10-05 by indie89