hoylabs/OpenDTU-OnBattery #399: Publish load corrected voltage via MQTT

swingstate, 2023-08-25:
### Is your feature request related to a problem? Please describe.

Feature request: 

I would like to track the difference between the measured DC voltage at the inverter channels and the calculated corrected voltage, to optimise the calculation. Ideally in HA, so data should be published via MQTT.

I appears to me that a different factor seems to be suitable depending on load, SOC/Voltages. To understand that better, a possibility to monitor these values would be appreciated.

### Describe the solution you'd like

Publish the data via MQTT as follows:

loadcorrectedVoltage under $invertername$/dc/lc_dc

load correction factor under $invertername$/dc/lc_f

schlimmchen, 2024-09-07:
The load-corrected voltage is not inverter-specific. Would this work:

Publish unter `.../powermeter/status/`:
* inverter serial used by DPL to calculate load-corrected voltage (this is a setting)
* inverter input number used by DPL to calculate load-corrected voltage (this is a setting)
* calculated load-corrected voltage in each DPL loop

schlimmchen, 2024-12-01:
Depends on #1431, as otherwise this would need to be implemented in an unacceptable manner (buried in the DPL code which should only be concerned with calculations and decisions, not publishing to MQTT).

commit ae023674 (2022-12-05): Fix #399: Move pin assignment to generic environment to prevent compilation errors