hoylabs/OpenDTU-OnBattery #420: Too many releases

schlimmchen, 2023-09-04:
### What happened?

It seems that for every tag that is *merged*, a release is built and published.

### To Reproduce Bug

See releases posted today, 2023-09-04.

### Expected Behavior

Only one release will be built and published once @helgeerbe merges development into master.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

n/a

### Anything else?

Thanks for promptly merging pull requests and releasing firmware for everybody to enjoy, @helgeerbe. However, I find it confusing for people that multiple releases seem to include similar changes. I think that this is not what you intended? Do you want one release per tag on the master branch **after** merging development into master? Currently it seems that also other tags (from other forks?!) cause releases to be built and published...

Also, not that it was important, but the JK BMS serial support is sorted under section "Under the hood". Is there something I need to do so that such a feature is listed in the respective section of the changelog/release info?

schlimmchen, 2023-09-05:
Nice!

The following releases were created because I created respective tags for them in my fork:
* https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2023.08.30-vedirect-debugging
* https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2023.08.29-jkbms

Why does that happen? Why are those tags even part of this repo? Actually... the same question applies for the tags made in the upstream project. I am confused... Does github automagically create them when merging pull requests? I really don't think that this is a pure GIT feature.

And those two tags from my fork would still trigger releases based on the `startsWith` approach...

helgeerbe, 2023-09-06:
Tags points to a specific commit. So yes, they are merged. To build actions on them, we have to use patterns which we can filter.

helgeerbe, 2023-10-02:
@schlimmchen If it is fine for you, I would close this issue. The actual script is not perfect, but it works so so far.

schlimmchen, 2023-10-02:
what about v23.9.18? wasn't that compiled by accident as well?

to be honest, I don't care too much, I mainly wanted to bring this to your attention.

helgeerbe, 2023-10-02:
It's build, but not released.

I coud try:

````yaml
  build:
    name: Build Enviornments
    runs-on: ubuntu-latest
    needs: get_default_envs
    strategy:
      matrix:
        environment: ${{ fromJSON(needs.get_default_envs.outputs.environments) }}
    if: !startsWith(github.ref, 'refs/tags/v2')
````

helgeerbe, 2023-10-02:
Checking the yaml code, I'm wondering why this does not work?

````yaml
on:
  push:
    paths-ignore:
      - docs/**
      - '**/*.md'
    branches:
      - master
      - development
    tags-ignore:        
      - v*
  pull_request:
    paths-ignore:
      - docs/**
      - '**/*.md'
````

schlimmchen, 2023-10-03:
> It's build, but not released.

https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/v23.9.18 :man_shrugging: 

I don't know YAML and I dont know github workflows, but this (already implemented) should do the trick:

```yaml
  release:
    name: Create Release
    runs-on: ubuntu-latest
    needs: [get_default_envs, build]
    if: startsWith(github.ref, 'refs/tags/2')
```

> Checking the yaml code, I'm wondering why this does not work?

I don't know either. I agree that the whole workflow should be skipped entirely for tags that start with a "v". There might be other events than "push" that trigger the workflow, in which case the filter is not working. I don't know...

> I coud try:

Yes, this would be my next try as well. Preventing a build for tags that start with a "v". You forgot double quotes, which are needed due to the exclamation point (as per chatGPT, I don't know any YAML):

```yaml
  build:
    [...]
    if: "!startsWith(github.ref, 'refs/tags/v')"
```

schlimmchen, 2024-01-03:
@helgeerbe The tags starting with a "v" (the tags tbnobody creates) do not trigger a realease any more. This is very obvious when looking at the tags from December.

I don't know what happens if I push another tag starting with 202... to my fork and if you merge a commit that is tagged as such. I will simply not do going forward.

Feel free to close this issue as resolved.

helgeerbe, 2024-01-04:
Unfortunately when those tags are merged to master, they still trigger a release build.

schlimmchen, 2024-03-18:
The Internet does not seem to know about this problem. This makes me believe that the tags are not part of this repo because a fork created them. I mean, it would be weird that a fork could create arbitrary matching tags and they would somehow end up in this repo or would be shared among forks.

I think the following happened with my tags: You locally added my fork as a GIT remote to test pull-requests or to merge them via the CLI. That also adds the tags from my fork to your local repo. That makes sense to me. You then create a tag to trigger a release and then you use

`git push --tags # <- bad`

which pushes foreign tags as well, rather than

`git push <tag you just created> # <- better`

I guess the solution is to always use `git push <tag you just created>` to avoid pushing tags from other remotes/forks. And also avoid pushing tags from upstream.

This all makes sense to me now. I mean, there is no reason why this repo should contain the upstream tags, do you agree?

For testing purposes I will created a tag that would match the filter in my repo. I assume that it will not be part of this repo and not be listed at https://github.com/helgeerbe/OpenDTU-OnBattery/tags.

[image]

Appears in my fork:
[image]

Does **not** appear in your upstream repo, as expected. And I am pretty confident that this test tag will not trigger a release in this repo unless you push it.

@helgeerbe Please let me know if you read this.

helgeerbe, 2024-03-19:
@schlimmchen good finding. But at the moment, I use the version tag from upstream to create the badge on the readme file, which points to the base release. 

Beside this I think it is not a bad idea to have the upstream version tag in our code, when we merge.

schlimmchen, 2024-03-19:
@helgeerbe Understood. Then let's use two commands, `git push v24...` (the latest upstream tag) and `git push 2024....` (the new release tag for this project) explicitly, which will prevent other foreign tags from appearing in this project.

I will take the liberty to clean up (delete) tags that are neither upstream tags not release tags. There are some, and they probably are all from my fork originally.

closed 2024-03-19 by schlimmchen