hoylabs/OpenDTU-OnBattery #435: [Request] Add support for non-regulated AC-charger

philippsandhaus, 2023-09-11:
### Is your feature request related to a problem? Please describe.

Besides directly charging my battery with a Victron MPPT I also have a Victron Blue Smart IP 22 AC-charger where I cannot dynamically regulate the output Amp. Currently I am dynamically switch on and the charger through a Smart-Plug controlled by Homeassistant based on excess PV energy. It would be great if this could instead be controlled by OpenDTU-OnBattery

### Describe the solution you'd like

A new "fixed amps" AC-charger should available besides the now available Huawei R4850G2. Configuration  in the web interface could be similar to how different batteries are available currently. The maximum current of the charger should be configurable as well to be able to only switch the charger on in DPL when enough energy is available.

Switching the charger could be done in several ways. I would suggest either a HTTP-Request or a MQTT-call, both configurable. This makes it quite flexible for different switches (Tasmota, es-home, Shelly, ...).

schlimmchen, 2024-09-07:
What exactly is the threshold to enable this type of charger? "Is full-solar-passthrough active (battery is full) and solar power greater than AC charger power"? I assume this is only working in a system where there are solar charge controllers and those do charge a battery? I am kind of getting a headache thinking about how to implement the logic that enables disables this type of charger, and in what setups it shall be supported... Please clarify.

Snoopy-HSS, 2025-02-16:
Maybe could be implemented in same way as Shelly/SmartPlug ...

You give start/stop threshold for the charger and user should be able to calculate the needed watts for the charger
 as the gap beetween

Snoopy-HSS, 2025-02-16:
Checked it.
It's only 12/24V
is it really suitable to use 24V Battery for Inverters?
Its only 24V + 11,5 A so up to less than 200 Watt per channel...

PR #1806 (2025-05-21): Feature: Improved support for R4850 and support for R48xx in general
This changeset improves upon the integration for R4850 power supplies by implementing new features/functions like

* reading board properties
* setting fan control modes
* shutdown over CAN
* processing ACKs from the PSU
* auto-setting offline limits (user-configurable)

and adds general support for (hopefully) all Huawei R48xx power supplies.

[image]

Closes #1544 (set (user-configurable) offline limits automatically)
Closes #1440 (support for R48xx in general, here: R4875)
Closes #435 (there is now a MQTT topic that allows [... cut]
closed 2025-05-21 by schlimmchen