hoylabs/OpenDTU-OnBattery #436: [Request] Add support for Victron AC chargers (Phoenix / BlueSmart)

philippsandhaus, 2023-09-11:
### Is your feature request related to a problem? Please describe.

Besides the Huawei charger it would great to also support Victron AC chargers. This the Phoenix line of AC chargers (expensive), which are controllable via VE.direct. There is however also [evidence](https://github.com/wasn-eu/Victron_BlueSmart_IP22) that the much more affordable BlueSmart line could be controlled via VE.Direct through a hidden PIN-header which make them a very interesting alternative to the often used Huawei charger. Another benefit ist, that there are also 12V and 24V (I have a 24V/16A) versions available. 

### Describe the solution you'd like

I envision a similar integration as described in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/435#issue-1891270558 and as the current integration of the Huawei charger.

philippsandhaus, 2023-09-24:
I have some first findings, also as a note to myself: 

My Victron blue charger provides a VE.DIrect port with 3.3 V as described in https://github.com/wasn-eu/Victron_BlueSmart_IP22, but it seems to speak only the HEX protocol. This is a sample log output:

```
:A0E2000011C
:4AAAAFD
:A0920000022
:A0E2000011C
:4AAAAFD
:A0920000022
:A0E2000011C
:4AAAAFD
:A0920000022
:A0E2000011C
:A0920000022
:A0E2000011C
:4AAAAFD
:A0920000022
:A0E2000011C
:4AAAAFD
:A0920000022
:A0E2000011C
:A0920000022
:A0E2000011C
:A0920000022
:A0E2000011C
:A8DED00220BA4
:A8FED000000CF
:A8CED0000000000D2
:AD5ED00220B5C
:AD7ED00000087
:ADBED00B80BC0
:A16EE00FFFF49
:A2001000100000029
:A30EC00012E
:A31EC00015EB4049A36A300FF4203007FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7FFFFFFFFF75
:A16EC000445
:A20EC00ECEDFFFFFFFFFFFF8DEDFFFFFFFFFFFF8CEDFFFFFFFFFFFF3EECFFFFFFFFFFFF61
:A70100000F0FE0C0001000000000000000000000000000000000000000000000000000000000000000000000000000000FFFFFFFFFFFF00D6
:A43100000F1FE0C005E2800004600000002000000BC0000000000000073
:A42100000F1FE0C005E2800004600000002000000BC0000000000000074
:A6F100028A4
:A99100080D90C003D
```

This means that the current VE.Direct implementation would have to be extended to also support HEX-Frames. This project provides a basic implementation for this: https://github.com/giacinti/VeDirectFrameHandler/blob/master/VeDirectFrameHandler.cpp

What is however missing is a concrete implementation of the HEX-based protocol. I suspect it is similar or the same as for the Phoenix-chargers as described here: https://www.victronenergy.com/upload/documents/VE.Direct-HEX-Protocol-Phoenix-Inverter.pdf

helgeerbe, 2023-09-24:
Is your Tx port set correctly in the victron app?

The functionality of the TX port can be set at:

Normal communication:
This is the default setting. Use this function when connecting to a GX device, a VE.Direct Bluetooth Smart dongle, or any other device that needs to communicate with the solar charger via the VE.Direct port.

schlimmchen, 2023-09-24:
A little encouragement: Given what you did to support the Victron SmartShunt, I have no doubts you can make this work. You already know your way around the VeDirectFrameHandler. Decoding the hex protocol is simply a matter of reading the field type and decoding the value that follows as defined by Victron and/or tested against your device.

However, if you do implement this, please be so kind and keep in mind that the Victron MPPT also speaks HEX and that it carries more and interesting data in those HEX messages, compared to the currently supported text-based protocol. Just keep it in mind.

philippsandhaus, 2023-09-24:
@helgeerbe Sadly this is not configurable via the victron app, you can try yourself by selecting one of the Blue Smart Chargers from the demo library. I guess the reason is that the port is not "officially" exposed in form of a normal VE-Direct connector, but only as an internal PIN-header I had to route outside.

@schlimmchen Thanks for the encouragement! Yes, I guess the HEX-protocol can also be interesting for other Victron devices. It also seems that when wanting to send commands to a Victron devices, you have to use the HEX protocol. It could for example be interesting to be able to control Load output relay which is present on some MPPT chargers.

schlimmchen, 2023-09-24:
> It also seems that when wanting to send commands to a Victron devices, you have to use the HEX protocol.

Yes. One can request historical data in particular. Only "max power today" and "max power yesterday" are included in the unsolicited text-encoded message. This datum ("may power") can however be requested for a whole bunch more days, which would be convenient if they could be displayed within OpenDTU-OnBattery.

> Sadly this is not configurable via the victron app

I think he had the same trouble as me: I did not understand at first that you are **not** talking about an MPPT charger, but an AC charger that also has "blue smart" in its name.

philippsandhaus, 2023-09-26:
I ha to correct myself: The BlueSmart also sends text messages together with Hex messages (Hex Messages starting with ':A' are status messages which are periodically sent and are not answers to commands sent to the device). This is good news and also in line with the Victron documentation which says that all newer Victron devices will also send the Text protocol for status messages. In fact, when connecting the device as a MPPT Charger in OpenDTU, basic measurements are already shown in the Webview.

'''
11:16:21.030 > [VE.Direct] hex frame received: :A0920000022
11:16:21.030 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.031 > [VE.Direct] hex frame received: :A0A20008E0F000084
11:16:21.031 > [VE.Direct] hex frame received: :A0920000022
11:16:21.031 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.032 > [VE.Direct] hex frame received: :A0920000022
11:16:21.032 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.033 > [VE.Direct] hex frame received: :A0320002D05F6
11:16:21.137 > [VE.Direct] hex frame received: :A132000B040000028
11:16:21.139 > [VE.Direct] hex frame received: :A0920000022
11:16:21.139 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.244 > [VE.Direct] hex frame received: :A0220005E0AC1
11:16:21.245 > [VE.Direct] hex frame received: :A0920000022
11:16:21.245 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.348 > [VE.Direct] hex frame received: :A0A20008E0F000084
11:16:21.350 > [VE.Direct] hex frame received: :A0920000022
11:16:21.350 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.351 > [VE.Direct] hex frame received: :A0320002D05F6
11:16:21.455 > [VE.Direct] hex frame received: :A0920000022
11:16:21.456 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.560 > [VE.Direct] hex frame received: :A8DED006E0A59
11:16:21.562 > [VE.Direct] hex frame received: :AD5ED006E0A11
11:16:21.562 > [VE.Direct] hex frame received: :AECED00E06F23
11:16:21.563 > [VE.Direct] hex frame received: :A16EE005E0ADF
11:16:21.563 > [VE.Direct] hex frame received: :A2001001F0100000A
11:16:21.563 > [VE.Direct] hex frame received: :A0920000022
11:16:21.564 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.564 > [VE.Direct] hex frame received: :A30EC00042B
11:16:21.668 > [VE.Direct] hex frame received: :A31EC00015EB4049A36A300FF4203007FFFFFFFFF1758B79989A30014040000C6FFFFFFFF20
11:16:21.670 > [VE.Direct] hex frame received: :A32EC000179EBAD9760A00061010000BEFFFFFFFF11B92899A5A300FF130100CDFFFFFFFFB9
11:16:21.670 > [VE.Direct] hex frame received: :A20EC00ECED000811B928998DED000C1758B7998CED000C1758B7993EEC000F79EBAD9769
11:16:21.775 > [VE.Direct] hex frame received: :A0220005E0AC1
11:16:21.776 > [VE.Direct] hex frame received: :A0320002D05F6
11:16:21.776 > [VE.Direct] hex frame received: :A0920000022
11:16:21.777 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.777 > [VE.Direct] hex frame received: :A0A20008E0F000084
11:16:21.777 > [VE.Direct] hex frame received: :A0920000022
11:16:21.778 > [VE.Direct] hex frame received: :A0E200021FC
11:16:21.802 > [Victron MPPT] Received Text Event PID: Value: 0XA336
11:16:21.908 > [Victron MPPT] Received Text Event FWE: Value: 0342FF
11:16:21.909 > [Victron MPPT] Received Text Event SER: Value: HQ2250QG2U6
11:16:21.910 > [Victron MPPT] Received Text Event V: Value: 26700
11:16:21.910 > [Victron MPPT] Received Text Event I: Value: 10000
11:16:21.911 > [Victron MPPT] Received Text Event T: Value: 13
11:16:21.911 > [Victron MPPT] Received Text Event ERR: Value: 0
11:16:21.912 > [Victron MPPT] Received Text Event CS: Value: 3
11:16:21.912 > [Victron MPPT] Received Text Event HC: Value: 851696
'''
[image]

DDvO, 2024-03-29:
Would this be worth the effort?

I doubt that with an AC-coupled system one could get
a (real, not based on vendor claims) efficiency of the storage system better than 80%
(total energy ratio: AC output over time / AC input over same period).

philippsandhaus, 2024-04-02:
It is for me and I can imagine siilar scenarios for others. Currently I have two independent sets of solar panels in my home. One set is connected to a Hoymiles inverter. The other set is connected to a Victron MPPT and a battery and the battery to a small Hoymiles inverter. If there is a lot of sun, the set with the Victron does charge the battery and the other (bigger) set produces more power then I can consume and is therefore fed back to the grid. Even if the efficiency is comparably bad, it is better for me to use at least a fraction of the otherwise fed back energy to additionally charge my battery via an AC charger.