hoylabs/OpenDTU-OnBattery #438: Shelly: Digest or RPC Channel changes since Shelly Pro Firmware above 1 (new stable Versions)

OlSteph, 2023-09-12:
### What happened?

Error(http://192.168.179.129/rpc/Shelly.GetStatus): connection lost
-> No actual Value for /em:0/total_act_power from my Shelly Pro 3em

### To Reproduce Bug

auto/manual Update of Shelly Gen 2 Devices to Stable Version 1 and higher

### Expected Behavior

https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/247
-> Same Settings (Password and IP hasn't changed)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

8dd96c4

### Relevant log/trace output

```Shell
23:03:35.972 > [HttpPowerMeter] Getting the power of phase 1 failed. Error: Error(http://192.168.179.129/rpc/Shelly.GetStatus): connection lost
23:03:36.030 > [HttpPowerMeter] Couldn't find a value with Json query "/em:0/total_act_power"
23:03:36.102 > [HttpPowerMeter] Getting the power of phase 1 failed. Error: Error(http://192.168.179.129/rpc/Shelly.GetStatus): connection lost
23:03:36.165 > [HttpPowerMeter] Couldn't find a value with Json query "/em:0/total_act_power"
23:03:36.349 > PowerMeterClass: TotalPower:  0.00
23:03:36.471 > [HttpPowerMeter] Getting the power of phase 1 failed. Error: Error(http://192.168.179.129/rpc/Shelly.GetStatus): connection lost
23:03:36.647 > [HttpPowerMeter] Couldn't find a value with Json query "/em:0/total_act_power"
```

### Anything else?

https://shelly-api-docs.shelly.cloud/gen2/General/RPCChannels/#get-with-query-string
-> Maybe this can help with the RPC Channel "Design"

-> https://shelly-api-docs.shelly.cloud/gen2/General/Authentication
Maybe there are changes in Digest Auth - i am not sure.
(if i change the Password to any wrong - the message is identical:
Error(http://192.168.179.129/rpc/Shelly.GetStatus): connection lost)

OlSteph, 2023-09-12:
After removing Authentication:
[image]

helgeerbe, 2023-09-13:
I don't see any changes in the spec. Unfortunately I don't own a Shelly Pro and implemented digest auth against the Shelly documentation.

What do you get if you run:

````bash
curl -v -i --digest -u admin:MY_PASSWORD "http://192.168.179.129/rpc/Shelly.GetStatus"
````

OlSteph, 2023-09-14:
C:\Users\user>curl -v -i --digest -u admin:my_passwort "http://192.168.179.129/rpc/Shelly.GetStatus"

*   Trying 192.168.179.129:80...
* Connected to 192.168.179.129 (192.168.179.129) port 80 (#0)
* Server auth using Digest with user 'admin'
> GET /rpc/Shelly.GetStatus HTTP/1.1
> Host: 192.168.179.129
> User-Agent: curl/8.0.1
> Accept: */*
>
< HTTP/1.1 401 Unauthorized
HTTP/1.1 401 Unauthorized
< WWW-Authenticate: Digest qop="auth", realm="shellypro3em-ec62609773d8", nonce="1694691801", algorithm=SHA-256
WWW-Authenticate: Digest qop="auth", realm="shellypro3em-ec62609773d8", nonce="1694691801", algorithm=SHA-256
< Content-Length: 0
Content-Length: 0
< Server: ShellyHTTP/1.0.0
Server: ShellyHTTP/1.0.0
< Connection: close
Connection: close

<
* Closing connection 0
* Issue another request to this URL: 'http://192.168.179.129/rpc/Shelly.GetStatus'
* Hostname 192.168.179.129 was found in DNS cache
*   Trying 192.168.179.129:80...
* Connected to 192.168.179.129 (192.168.179.129) port 80 (#1)
* schannel: InitializeSecurityContext failed: SEC_E_QOP_NOT_SUPPORTED (0x8009030A) - Die Eigenschaft "Quality of Protection" für Einzelnachrichten wird vom Sicherheitspaket nicht unterstützt.
* Connection #1 to host 192.168.179.129 left intact
curl: (94) An authentication function returned an error

OlSteph, 2023-09-14:
Test with raspberry OS (Raspi4)

`pi@raspberrypi:~ $ curl -v -i --digest -u admin:my_password"http://192                                                                              .168.179.129/rpc/Shelly.GetStatus"
*   Trying 192.168.179.129:80...
* Connected to 192.168.179.129 (192.168.179.129) port 80 (#0)
* Server auth using Digest with user 'admin'
> GET /rpc/Shelly.GetStatus HTTP/1.1
> Host: 192.168.179.129
> User-Agent: curl/7.74.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 401 Unauthorized
HTTP/1.1 401 Unauthorized
< WWW-Authenticate: Digest qop="auth", realm="shellypro3em-ec62609773d8", nonce=                                                                              "1694692867", algorithm=SHA-256
WWW-Authenticate: Digest qop="auth", realm="shellypro3em-ec62609773d8", nonce="1                                                                              694692867", algorithm=SHA-256
< Content-Length: 0
Content-Length: 0
< Server: ShellyHTTP/1.0.0
Server: ShellyHTTP/1.0.0
< Connection: close
Connection: close

<
* Closing connection 0
* Issue another request to this URL: 'http://192.168.179.129/rpc/Shelly.GetStatu                                                                              s'
* Hostname 192.168.179.129 was found in DNS cache
*   Trying 192.168.179.129:80...
* Connected to 192.168.179.129 (192.168.179.129) port 80 (#1)
* Server auth using Digest with user 'admin'
> GET /rpc/Shelly.GetStatus HTTP/1.1
> Host: 192.168.179.129
> Authorization: Digest username="admin", realm="shellypro3em-ec62609773d8", non                                                                              ce="1694692867", uri="/rpc/Shelly.GetStatus", cnonce="YTRhN2NhYWI3YzJjMTk0N2ZlNz                                                                              diMDBiNjQxMzdiNGE=", nc=00000001, qop=auth, response="e0fd6bf55a992e30c1747fffb0                                                                              58c44a6545b25feb032c68af635b41bbb86e81", algorithm=SHA-256
> User-Agent: curl/7.74.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
HTTP/1.1 200 OK
< Content-Type: application/json
Content-Type: application/json
< Content-Length: 1215
Content-Length: 1215
< Server: ShellyHTTP/1.0.0
Server: ShellyHTTP/1.0.0
< Connection: close
Connection: close

<
{"ble":{},"cloud":{"connected":true},"em:0":{"id":0,"a_current":2.008,"a_voltage                                                                              ":233.8,"a_act_power":-462.4,"a_aprt_power":469.3,"a_pf":0.98,"a_freq":50.0,"b_c                                                                              urrent":1.532,"b_voltage":234.0,"b_act_power":-328.7,"b_aprt_power":358.3,"b_pf"                                                                              :0.92,"b_freq":50.0,"c_current":1.486,"c_voltage":234.2,"c_act_power":283.4,"c_a                                                                              prt_power":347.6,"c_pf":0.81,"c_freq":50.0,"n_current":null,"total_current":5.02                                                                              6,"total_act_power":-507.723,"total_aprt_power":1175.244, "user_calibrated_phase                                                                              ":[]},"emdata:0":{"id":0,"a_total_act_energy":118417.69,"a_total_act_ret_energy"                                                                              :102565.19,"b_total_act_energy":92184.11,"b_total_act_ret_energy":148274.46,"c_t                                                                              otal_act_energy":156209.96,"c_total_act_ret_energy":0.00,"total_act":366811.76,                                                                               "total_act_ret":250839.64},"eth":{"ip":"192.168.179.129"},"modbus":{},"mqtt":{"c                                                                              onnected":true},"sys":{"mac":"EC62609773D8","restart_required":false,"time":"14:                                                                              01","unixtime":1694692867,"uptime":132913,"ram_size":256448,"ram_free":105488,"f                                                                              s_size":524288,"fs_free":204800,"cfg_rev":11,"kvs_rev":0,"schedu* Closing connec                                                                              tion 1
le_rev":0,"webhook_rev":0,"available_updates":{}},"temperature:0":{"id": 0,"tC":                                                                              55.4, "tF":131.7},"wifi":{"sta_ip":null,"status":"disconnected","ssid":null,"rss`

helgeerbe, 2023-09-14:
OK, that looks fine. Have to think about, how to test it. I have a Shelly pm1 plus which also should support digest authentication. I will update it and run som tests. Maybe I will see the same issue.

Github-Nutzer, 2023-09-14:
Using OpenDTU-onBattery 2023.09.13 with a Shelly Pro3EM on Firmware 1.0.3, digest auth is working for me as before

OlSteph, 2023-09-14:
Strange, i flashed the same DTU version again and get the same Error. Same Shelly Firmware Version. - resetted the Shelly again - no Luck..

helgeerbe, 2023-09-14:
Could be related to #381

I get connection lost for the second call in digest auth

````c++
httpCode = httpClient.GET();
````

MalteSchm, 2023-09-14:
> After removing Authentication: [image]

Can you post the other settings?

Github-Nutzer, 2023-09-15:
[image]

MalteSchm, 2023-09-16:
@helgeerbe 
You could try to let httpClient create/handle the wifiClient by not providing it externally. This works with http only though

helgeerbe, 2023-09-18:
@MalteSchm 

This works.

````c++
    // secureWifiClient MUST be created before HTTPClient
    // see discussion: https://github.com/helgeerbe/OpenDTU-OnBattery/issues/381
    //std::unique_ptr<WiFiClient> wifiClient;

/*     if (urlProtocol == "https") {
      auto secureWifiClient = std::make_unique<WiFiClientSecure>();
      secureWifiClient->setInsecure();
      wifiClient = std::move(secureWifiClient);
    } else {
      wifiClient = std::make_unique<WiFiClient>();
    } */

    WiFiClient *wifiClient = new WiFiClient;
````

So what is exactly the difference to `std::make_unique<WiFiClient>();` and why did you use it?

helgeerbe, 2023-09-18:
OK, should be fixed with ed2a189a6136ddd1a412e88586ba7abfd0d363c9

Looks like you can't reuse the connection. You need to call `httpClient.end() and again httpClient.begin()'

helgeerbe, 2023-09-19:
@OlSteph can you test the fix with release 2023.09.19

OlSteph, 2023-09-19:
@helgeerbe 
Looks good 
[image]

commit f2893220 (2023-09-18): fix: #438
closed 2023-09-19 by helgeerbe