hoylabs/OpenDTU-OnBattery #442: [Request] network configuration - allow definition of Access Point MAC address

swingstate, 2023-09-16:
### Is your feature request related to a problem? Please describe.

The DTU I am using is installed next to a wireless repeater, however the device keeps connecting to another repeater which is more distanced.
The result is that the DTU's WIFI link quality isn't really good (and stable). In average the link quality is -80db / below 50%. 

I have another esp in that cabinet which is running ESPHome and where I configured to use the Accesspoint which is installed in the cabinet. The Wifi link is really good, -15db. 

### Describe the solution you'd like

The UI should support entering a preferred access point by defining the MAC address into a text field. The text field should also support basic checks, e.g. the Mac address should be in the right format. e.g. 0C:72:74:C6:22:6E.

The UI should also support a fallback in case the connection to the AP can not be made any longer. In that case the ESP should connect to any AP in the WIFI/Mesh.

Mockup of the settings: 

[image]

### Describe alternatives you've considered

I brought the AP as close as possible to the ESP, however it keeps connecting to other, more distanced APs.

### Additional context

This is a more general feature, hence might also benefit the OpenDTU (non -onBattery) project.

swingstate, 2023-09-18:
I dopend a request in the original project.

closed 2023-09-18 by swingstate

schlimmchen, 2023-09-21:
This is the upstream issue: https://github.com/tbnobody/OpenDTU/issues/1354