hoylabs/OpenDTU-OnBattery #457: [Request] Get value for PV power via MQTT

stanpete78, 2023-09-25:
### Is your feature request related to a problem? Please describe.

The way it is now, you have to wire a Victron Charger to the ESP in order to get its values for PV power production and to be able to use solar passtrough. So if you need to use two chargers (east/west, for example) can't fully utilize the potential of openDTU.

### Describe the solution you'd like

It would be beneficial if you could calculate the total PV power separately and have the DTU use this external value. This would instantly allow the use of multiple chargers and also chargers other than victron. And all this without more GPIOs, more CPU cycles. It would be scalable basically for free.

### Additional context

The preferred way I'd do this with my 100/20 Victrons would be to attach an ESP with KinDR007/Victron-MPPT-ve.direct-to-MQTT to each of the Victrons and either have Home Assistant calculate the value, or better, have openDTU-onbattery just read and add the values by itself.

iamdevnull, 2024-10-21:
@schlimmchen kann man theoretisch damit rechnen, dass der DPL Mode 2 ((Full) Solar-Passthrough - gemäß https://opendtu-onbattery.net/firmware/solar_passthrough/#solar-passthrough-is-off ) auf Basis von MQTT ermittelten Daten arbeiten könnte beim nächsten Release. Falls ja, kannst eine zeitlich Prognose vielleicht geben. Vielen Dank.

Bei mir stehen z.B wie folgt die Daten bereit:

[image]

AndreasBoehm, 2024-10-21:
@iamdevnull Ich hab vor ein paar Wochen damit angefangen das feature `MPPT Daten via MQTT` zu implementieren. Dafür ist eine sehr umfangreiche anpassung des aktuellen MPPT codes nötig, das wird sich noch ein bisschen ziehen. Ein Datum oder eine Prognose kann ich dir nicht geben.

PR #1457 (2025-01-18): Restructure solarcharger classes
Restructure solarcharger classes to allow simple addition of new providers.

Based on the structure that @schlimmchen introduced here: https://github.com/hoylabs/OpenDTU-OnBattery/pull/1451

Closes #1408.
Closes #457.
closed 2025-01-18 by schlimmchen