hoylabs/OpenDTU-OnBattery #467: Support for JBD BMS [Request]

futzicomp, 2023-09-26:
### Describe the solution you'd like

Über ein Support für die gebräuchlichen und weit verbreiteten JBD und Darly BMS würde ich mich sehr, sehr freuen!

D A N K E ! Tolles Projekt

schlimmchen, 2023-09-26:
Dann lass doch mal ein paar Links da: zu Produktseiten, zu Dokumentation, etc. Am besten auch Hinweise darauf, welches Protokoll die sprechen (vermutlich proprietär) und welches Medium die nutzen (RS232, RS485, UART, CAN, Bluetooth). Und am besten überlegst du dir gleich zusätzlich, wie du ein solches BMS für die Entwicklung zur Verfügung stellen kannst...

futzicomp, 2023-09-26:
Hilft dir das hier weiter?
https://github.com/syssi/esphome-jbd-bms

schlimmchen, 2023-09-26:
Hm, jaaaaa, also copy'n'paste ist es leider nicht, aber es gibt ein gutes Dokument https://github.com/syssi/esphome-jbd-bms/blob/main/docs/Jiabaida.communication.protocol.pdf und der Code kann im Zweifel helfen, Unklarheiten zu beseitigen.

Ist halt Fleißarbeit... Der Überbau (Web UI, Abstraktion) ist fertig seit dem Support für JK BMS, es wäre "nur" ein Controller für genau dieses BMS zu entwickeln, der sich da einfügt.

> Wenn du es wert findest, kann ich gerne ein günstiges Modell bestellen

Es ist sehr gut, dass du dazu grundsätzlich bereit wärst, das erhöht die Chancen dramatisch, dass sich dem jemand annehmen möchte. Ich fänds auch durchaus sehr spannend, aber leider kann ich nicht absehen, dass ich mir die Zeit nehmen kann.

Ich kann gerne Tipps geben und beim Implementieren helfen. Das war mit @philippsandhaus in #452 sehr fruchtbar :wink:

futzicomp, 2023-09-26:
Sorry, das was ihr da zusammenprogrammiert ist reine Magie für mich ( ich bin nur ein Elektrotechnik und Software Sympatisant) Du müsstest dich mit einem DAU rumschlagen. Das mag ich Dir nicht zumuten, da du verm. schon viel zu viel von deiner Freizeit opferst - Aber Danke für das Angebot. Ich such was bei Ali und schicke es dir oder deinen Mitstreiter wenn es angekommen und verdrahtet ist.

Viel Erfolg beim weiteren Verbessern! Wirklich tolles Projekt!

Artikel14, 2024-01-31:
Hier ein Beispiel zum JBD BMS in C++. 
Wie wäre es mit einem solchen Ansatz?
Könnte man auf dieser Basis etwas einbinden?

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

// Bluetooth UUIDs für das JBD BMS
#define SERVICE_UUID "0000ff00-0000-1000-8000-00805f9b34fb"
#define READ_CHARACTERISTIC_UUID "0000ff01-0000-1000-8000-00805f9b34fb"

BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(BLEUUID(SERVICE_UUID))) {
      BLEDevice::getScan()->stop();
      advertisedDevice.getScan()->getScan()->clearResults();
      BLEClient* pClient = BLEDevice::createClient();
      pClient->connect(&advertisedDevice);

      BLERemoteService* pRemoteService = pClient->getService(BLEUUID(SERVICE_UUID));
      BLERemoteCharacteristic* pReadCharacteristic = pRemoteService->getCharacteristic(BLEUUID(READ_CHARACTERISTIC_UUID));

      // Menü für die Datenabfrage
      int choice;
      do {
        Serial.println("1. State of Charge  abfragen");
        Serial.println("2. Leistung abfragen");
        Serial.println("3. Spannung der gesamten Batterie abfragen");
        Serial.println("4. Stromstärke abfragen");
        Serial.println("5. Zyklenanzahl abfragen");
        Serial.println("0. Beenden");
        Serial.print("Auswahl: ");
        while (!Serial.available()) {}
        choice = Serial.parseInt();
        Serial.println();

        switch (choice) {
          case 1:
            queryAndPrintSOC(pReadCharacteristic);
            break;
          case 2:
            queryAndPrintPower(pReadCharacteristic);
            break;
          case 3:
            queryAndPrintTotalVoltage(pReadCharacteristic);
            break;
          case 4:
            queryAndPrintCurrent(pReadCharacteristic);
            break;
          case 5:
            queryAndPrintCycleCount(pReadCharacteristic);
            break;
        }
      } while (choice != 0);

      pClient->disconnect();
    }
  }

  void queryAndPrintSOC(BLERemoteCharacteristic* pReadCharacteristic) {
    // Hier implementieren Sie den Code zum Abfragen und Ausgeben des SoC
    std::string readValue = pReadCharacteristic->readValue();
    uint8_t rsoc = readValue[15];
    Serial.println("SoC: " + String(rsoc) + "%");
  }

  void queryAndPrintPower(BLERemoteCharacteristic* pReadCharacteristic) {
    // Hier implementieren Sie den Code zum Abfragen und Ausgeben der Leistung
    // Beispiel: Leistung wird nicht direkt gemeldet, kann aber aus Spannung und Strom berechnet werden
    std::string readValue = pReadCharacteristic->readValue();
    uint16_t totalVoltage = (readValue[5] << 8) | readValue[6];
    uint16_t totalCurrentRaw = (readValue[7] << 8) | readValue[8];
    int16_t totalCurrent = (totalCurrentRaw & 0x8000) ? -(totalCurrentRaw & 0x7FFF) : totalCurrentRaw;
    float power = totalVoltage * totalCurrent / 1000.0; // Leistung in Watt
    Serial.println("Leistung: " + String(power) + " W");
  }

  void queryAndPrintTotalVoltage(BLERemoteCharacteristic* pReadCharacteristic) {
    // Hier implementieren Sie den Code zum Abfragen und Ausgeben der Spannung der gesamten Batterie
    std::string readValue = pReadCharacteristic->readValue();
    uint16_t totalVoltage = (readValue[5] << 8) | readValue[6];
    float totalVoltageFloat = totalVoltage / 100.0; // Umrechnung in Volt
    Serial.println("Spannung der gesamten Batterie: " + String(totalVoltageFloat) + " V");
  }

  void queryAndPrintCurrent(BLERemoteCharacteristic* pReadCharacteristic) {
    // Hier implementieren Sie den Code zum Abfragen und Ausgeben der Stromstärke
    std::string readValue = pReadCharacteristic->readValue();
    uint16_t totalCurrentRaw = (readValue[7] << 8) | readValue[8];
    int16_t totalCurrent = (totalCurrentRaw & 0x8000) ? -(totalCurrentRaw & 0x7FFF) : totalCurrentRaw;
    float totalCurrentFloat = totalCurrent / 100.0; // Umrechnung in Ampere
    Serial.println("Stromstärke: " + String(totalCurrentFloat) + " A");
  }

  void queryAndPrintCycleCount(BLERemoteCharacteristic* pReadCharacteristic) {
    // Hier implementieren Sie den Code zum Abfragen und Ausgeben der Zyklenanzahl
    std::string readValue = pReadCharacteristic->readValue();
    uint16_t cycleCount = (readValue[13] << 8) | readValue[14];
    Serial.println("Zyklenanzahl: " + String(cycleCount));
  }
};

void setup() {
  Serial.begin(115200);
  Serial.println("JBD BMS Bluetooth Steuerung");
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5);
}

void loop() {
  pBLEScan->start(5);
  delay(5000);
}

Was meint ihr? Hilft das weiter?

closed 2024-10-30 by schlimmchen