hoylabs/OpenDTU-OnBattery #468: Victron LR über Bluetooth zugreifen [Request]

futzicomp, 2023-09-26:
### Describe the solution you'd like

Der Victron sendet neben Hardware Schnittstelle auch über Bluetooth seine Werte, siehe   https://github.com/Fabian-Schmidt/esphome-victron_ble/tree/main. Dans funktioniert bei mir momentan sehr gut.
Damit könnte man sich verkabelungsaufwand ersparen und eh vorhandene Bluetooth Hardware nutzen. Wäre doch  chick wenn das auch noch in den Ressourcen des ESP platz hätte?

schlimmchen, 2023-09-26:
> Wäre doch chick wenn das auch noch in den Ressourcen des ESP platz hätte?

Stimmt. Hat es aber nicht. Wenn der Bluetooth Stack mit in die Firmware kompiliert wird, platzt die Firmware sofort aus dem Rahmen. Das lässt sich auch nicht schönrechnen oder hinbiegen. Dann ist es erforderlich, das Partitionsschema zu ändern. Und man muss sich eine neue Update-Strategie überlegen (weil nicht zwei volle Firmware gleichzeitig im Flash sein können). Oder man muss den Support für Geräte mit 2MB Flash einstellen. Oder noch irgendetwas anderes...

Bluetooth benutzen in OpenDTU-OnBattery ist kein einfaches Thema wegen dieser Beschränkung.

swingstate, 2023-09-26:
ist hier "nur" die Größe des Flashspeichers limitierend? Neben der Komplexität BT zu implementieren.

schlimmchen, 2023-09-26:
> ist hier "nur" die Größe des Flashspeichers limitierend? Neben der Komplexität BT zu implementieren.

Ich hab übel zusammengezimmerten Code für das Auslesen des JK BMS über Bluetooth an einem ESP32 gesehen, der laut Autor funktionierte, und der sah im Grunde alles andere als kompliziert aus. Lästig ist nur das Koppeln, was schlimmstenfalls interaktive Benutzereingaben erfordert. Der Rest scheint eher simpel zu sein. Ob das zuverlässig ist: :man_shrugging:

Also ja, der Flashspeicher ist in erster Linie was uns hier aufhält.

swingstate, 2023-09-26:
Könnte man dann einen "branch" aufmachen, mit dem man das mal testet um die Zuverlässigkeit zu ermitteln? 

Food for thought. Könnte eine sinnvolle Ergänzung sein.

futzicomp, 2023-09-26:
cool wie ihr euch kümmert! Dann mach ich wieder auf!

schlimmchen, 2023-09-26:
> Könnte man dann einen "branch" aufmachen, mit dem man das mal testet um die Zuverlässigkeit zu ermitteln?

Klar geht das. Wann startest du die Exploration?

swingstate, 2023-09-26:
Läuft doch schon :)

helgeerbe, 2023-09-28:
Ich hatte das schon mal implementiert.
https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/129#discussioncomment-5730807

Code habe auf einen gist bereitgestellt. Prinzipiell geht es, aber dann bleibt keine Speicher mehr übrig.

swingstate, 2023-09-28:
@helgeerbe bezieht sich Deine Aussage auf die Modelle mit 2MB Flash bzw. worauf genau?

helgeerbe, 2023-09-28:
@swingstate Ich nutze esp32 D1 Mini. Die haben 4 MB Flash. Da bei OTA immer 2 Firmware Stände im Flash abgelegt werden, hast du aber nur ca. 2MB zur Verfügung. Jede zusätzliche lib macht das Image größer.

Das eigentliche Image ist leider schon so groß, dass kein Debug build mehr so einfach geflasht werden kann.

swingstate, 2023-09-28:
Ja der D1 is ein schönes Teil, aber eben mini :) 
Ist ein Wechsel bzw. eine Erweiterung des Support für 8 oder 16MB Modelle denkbar, so dass sich der Funktionsumfang ausbauen ließe?

schlimmchen, 2023-09-28:
@swingstate Du bist Produktmanager, stimmt's?

Ja, natürlich geht das. Der ESP32-S3 auf dem OpenDTU-FUSION Board hat z.B. mehr Flash. Der kann auch jetzt schon OpenDTU, der halbe Flash ist halt ungenutzt. Aber hast du dir schon Gedanken um die Implikationen gemacht? Wie soll denn das von statten gehen? Wie viele Varianten eines Releases gibt es denn dann? Und wie werden die gebaut? Muss ja einmal mit und einmal ohne Bluetooth Stack sein. Und die Leute müssen begreifen, welche Version sie nehmen müssen. Ist ja jetzt schon kompliziert genug. Und sie müssen auch begreifen, dass es die Version für den D1 mini nur ohne Bluetooth Stack gibt. Oder sollen ESP32 mit weniger als 8MB Flash einfach nicht mehr unterstützt werden mit dem nächsten Release? Oder stellst du dir vielleicht eine minimal-Firmware im zweiten Firmware-Slot vor, so wie es Tasmota macht? Und wenn ja, wie bauen wir die denn wiederum? Das wird ja ein `#ifdef`-Gemetzel. Oder stellen wir den Support für OTA Updates ein und man muss halt Updates mit dem Kabel einspielen?

Wie klären wir all diese Fragen? Welche Antworten hast du?

swingstate, 2023-09-28:
bin ich nicht.

madmartin, 2023-09-28:
Hi,
meiner Meinung nach sind die bisher angesprochenen Details (Größe des Programcode vs. OTA) alle irgendwie lösbar.
Problematischer sehe ich - daß ich an verschiedenen Stellen im Netz gelesen habe  daß die gleichzeitige Nutzung von BT und WLAN schwierig ist, weil sich beide Technologien die Antenne und den RF-Teil teilen müssen.

Detail stehen hier: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/coexist.html

Das sehe ich als vorrangiges Problem. 

Den Platz im Flash würde ich dann so wie bei Tasmota "vergrössern" - Stichwort "Safeboot". https://tasmota.github.io/docs/Safeboot/ - das wäre allerdings eher in Feature was in OpenDTU implementiert werden sollte.

Und bis dahin flasht man eben nur seriell. Ist dann eben so.

schlimmchen, 2023-10-03:
Guter Punkt, @madmartin. Der Tatsache geschuldet, dass die Anbindung über UART so gut funktioniert (extrem stabil), wäre es wirklich ungeschickt sich die Wi-Fi-Verbindung kaputt zu machen mit Bluetooth. Nochmal andere sind allerdings mit Ethernet unterwegs und könnten durchaus den RF-Teil für BT nutzen.

>  das wäre allerdings eher in Feature was in OpenDTU implementiert werden sollte.

Das sehe ich nicht, dass das passiert. Das Upstream-Projekt hat davon nichts außer Nachteilen.

> Und bis dahin flasht man eben nur seriell. Ist dann eben so.

Müsste doch möglich sein die aktuelle bzw. die einizige Firmware-Kopie im Flash zu überschreiben. Wenn das fehlschlägt, kann man immer noch seriell flashen. Anders als bei Tasmota-Geräten, an die man meistens *sehr* schlecht drankommt, hat man bei unserem Anwendungsfall normalerweise recht leicht physischen Zugriff und kann das bei jeden zehnten Update, das irgendwie schief geht, seriell wiederbeleben.