hoylabs/OpenDTU-OnBattery #471: Falscher Channel für Wechstromleistung im Modul "PowerLimiter"

skippermeister, 2023-09-30:
### What happened?

In der Funktion PowerLimiterClass::inverterPowerDcToAc(....
in den Zeilen 345 und 346

float inverterEfficiencyPercent = inverter->Statistics()->getChannelFieldValue(TYPE_AC, static_cast<ChannelNum_t>(config.PowerLimiter_InverterChannelId), FLD_EFF);

wird die Wechselstromleistung über die Funktion getChannelFieldValue abgefragt. 
Der Channel Parameter für den TYPE_AC muss aber immer CH0 sein (siehe Inverter Tabellen z.B. HM_2CH.cpp).

Der gleiche Fehler ist in der Funktion PowerLimiterClass::calcPowerLimit(....
in der Zeile 443

acPower = static_cast<int>(inverter->Statistics()->getChannelFieldValue(TYPE_AC, (ChannelNum_t) config.PowerLimiter_InverterChannelId, FLD_PAC));

und in der Funktion PowerLimiterClass::getLoadCorrectedVoltage()
in der Zeile 586

float acPower = _inverter->Statistics()->getChannelFieldValue(TYPE_AC, channel, FLD_PAC);

### To Reproduce Bug

Ich habe diesen Fehler gefunden, weil ich den 2. MPPT Kanal eines HM-600 an meine Pylontech Batterie angeschlossen habe und fehlerhafte Werte erhalten hatte.

### Expected Behavior

Bitte den 2. Parameter in den Funktionsaufrufen in der Datei PowerLimiter.cpp

inverter->Statistics()->getChannelFieldValue(TYPE_AC, ......

auf CH0 ändern.   

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

f0a55ea

schlimmchen, 2023-09-30:
Das würde erklären, warum ich Probleme hatte einen anderen Kanal zu wählen für meinen HM-1500, obwohl da alle DC Kanäle an der Batterie hängen und es mit allen hätte gleich funktionieren sollen.

Guter Fund!

Kann ich reparieren, falls sonst niemand Zeit findet bis Ende nächster Woche. Oder du magst selbst einen PR versuchen? Den betroffenen Code hast du ja schon gefunden.

skippermeister, 2023-09-30:
Ich habe heute mich das erste mal bei github registriert, bin also noch nicht so firm mit dem github System. Bislang habe ich alle meine Entwicklungen lokal auf meinem PC durchgeführt. 

@schlimmchen: Wenn Du die Korrektur vornehmen könntest wäre das super.

Die OpenDTU-onBattery Software habe ich im Source Code auf meinem PC in weiten Teilen an meine Bedürfnisse und Zwecke angepasst und geändert. 

Meine Setup sieht so aus, dass ich einen Hoymiles HM-600 Wechselrichter direkt von den Solar Panels betreibe. Ich habe kein Victron MPPT Charger, sondern ein effizienten Wechselstrom Lader von Meanwell (ja ich weiß, Verluste im Vergleich zur Gleichstromladung). Die Ladung erfolgt ebenfalls über einen Software Regler, so daß nur überschüssige Leistung, welche normalerweise ins Grid gehen würde zur Ladung herangezogen wird. Ich lade die batterie also nur mir Solar Energie.

Ein zweiter HM-600 (hatte ich übrig) befindet sich mit dem Kanal 2 direkt an der Batterie und wird über eine Softstart Hardware Schaltung automatisch nachts eingeschaltet (MOSfet Schaltung) und morgens wieder ausgeschaltet. Dazu habe ich im Modul PowerLimiter eine entsprechende Steuerung eingebaut. Die beiden MOSfets (1x precharge für 5 Sekunden über 40 Ohm Lastwiderstand) und 1x Full Power) sind über Opto-Koppler an den ESP32 verbunden. dadurch verhindere ich den Peak-Inrush und schone die Eingangskondensatoren des HM-600.
Statt des HUWAEI Laders habe ich ein Meanwell NPB1200-48 per CAN-Bus realisiert.
Die Pylontech Batterie wird nicht mehr über CAN-Bus sondern RS485 abgefragt. Dadurch erhalte ich von der Batterie wesentlich mehr Informationen wie z.B. die Zellspannungen.  
Weiter habe ich eine Zero-Export PI Regler für den Solar Regler realisiert. So kann die OpenDTU nahezu autark arbeiten. 
Als nächstes will ich mein SmartMeter basierend auf tasmota per RS485 anschließen. Aktuell habe ich einen shelly3EM für die Leistungsmessung. 
Meine Absicht ist es das gesamte System autark ohne mein Netzwerk zu realisieren.
Ich habe festgestellt, dass der shelly3EM einmal am Tag (meistens gegen 3:00 nachts) für mehr als 30 Sekunden nicht erreichbar ist und häufiger auf die HTTP Requests nicht antwortet. Dazu habe ich auch hier ein Erweiterung in den PowerLimter programmiert, der mir nach 30 Sekunden zunächst eine Warnung gibt und nach weiteren 30 Sekunden den Hoymiles an der Batterie abschaltet (shutdown).
Die Abhängigkeit vom Netzwerk stört mich, deshalb möchte ich das Power-Meter auf RS485 umstellen.

Um alle meine Änderungen auch im Web Fronend zu sehen und einstellen zu können, sind auch hier weitreichende Änderungen von mir vorgenommen worden.
 
Bei der ganzen Programmierung ist mir der gefundene Fehler aufgefallen.

Meine Änderungen sind so vielfältig, dass ich wohl einen Fork machen, oder ein eigenes Repository starten müsste.

helgeerbe, 2023-10-02:
Sollte mit 7fb26e1e815b28ec623a460aab87149b6d67fd78 in development enthalten sein

schlimmchen, 2023-10-06:
@skippermeister Inzwischen ist die Reparatur in einem [Release](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2023.10.04) enthalten. Kannst du das Issue schließen?

skippermeister, 2023-10-06:
Problem gelöst

closed 2023-10-06 by skippermeister