hoylabs/OpenDTU-OnBattery #482: JK BMS incomplete MQTT data in Home Assistant

NevelSavage, 2023-10-06:
### What happened?

JK BMS only shows 'SoC' and 'Manufacturer' in Home Assistant.
All other items are 'unbekannt'.

[image]

In the OpenDTU-OnBattery web interface, other values are displayed correctly, e.g. temperature, current, etc.

MQTT Explorer shows slightly more entities (although not all of the possible ones that were published in Home Assistant), but these are rarely updated, it seems to me. The entities SoC and Manufacturer are updated cyclically. Could it possibly have something to do with that?

### To Reproduce Bug

Removing the device MQTT Battery in Home Assistant.
Reboot Home Assistant and OpenDTU-OnBattery.
After the restart, the battery is recognized again, but the values are still not fully displayed.

### Expected Behavior

All available entities could be displayed in Home Assistant. The cell voltages would also be interesting.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

f0a55ea

schlimmchen, 2023-10-06:
Yes, I am aware. The reason is this: The HomeAssistent integration in OpenDTU-OnBattery still only knows about the Pylontech Battery and publishes all the sensors the Pylontech would feed with data. I started working on fixing this, but have not found the time to finish it...

Thanks for using the JK BMS interface and giving feedback :blush:

NevelSavage, 2023-10-06:
This is good news, it means it will work in the future.
No stress! I appreciate your work and can only thank you for it.
If I knew how I could support you in your work, I would do it. Unfortunately, my Git experience was a while ago.

NevelSavage, 2023-10-28:
I have added a case to the file “MqttHandlePylontechHass.cpp”. Now, depending on the BMS selected (Pylontech or JK BMS) in the config, the correct data will be published to Home Assistant. I have tested it successfully and would like to make it available to the general public. What I don't really like is that I implemented it in the file for "Plyontech" due to inexperience. Maybe someone can rename the file to "MqttHandleBMSHass.cpp"

https://github.com/NevelSavage/OpenDTU-OnBattery/blob/development/src/MqttHandlePylontechHass.cpp

```
// distinction PYONTCH / JK BMS / 
    switch (config.Battery_Provider) {
        case 0: // Pylontech
            publishSensor("Battery voltage", NULL, "voltage", "voltage", "measurement", "V");
            publishSensor("Battery current", NULL, "current", "current", "measurement", "A");
            publishSensor("Temperature", NULL, "temperature", "temperature", "measurement", "°C");
            publishSensor("State of Charge (SOC)", NULL, "stateOfCharge", "battery", "measurement", "%");
            publishSensor("State of Health (SOH)", "mdi:heart-plus", "stateOfHealth", NULL, "measurement", "%");
            publishSensor("Charge voltage (BMS)", NULL, "settings/chargeVoltage", "voltage", "measurement", "V");
            publishSensor("Charge current limit", NULL, "settings/chargeCurrentLimitation", "current", "measurement", "A");
            publishSensor("Discharge current limit", NULL, "settings/dischargeCurrentLimitation", "current", "measurement", "A");
        
            publishBinarySensor("Alarm Discharge current", "mdi:alert", "alarm/overCurrentDischarge", "1", "0");
            publishBinarySensor("Warning Discharge current", "mdi:alert-outline", "warning/highCurrentDischarge", "1", "0");
        
            publishBinarySensor("Alarm Temperature low", "mdi:thermometer-low", "alarm/underTemperature", "1", "0");
            publishBinarySensor("Warning Temperature low", "mdi:thermometer-low", "warning/lowTemperature", "1", "0");
        
            publishBinarySensor("Alarm Temperature high", "mdi:thermometer-high", "alarm/overTemperature", "1", "0");
            publishBinarySensor("Warning Temperature high", "mdi:thermometer-high", "warning/highTemperature", "1", "0");
        
            publishBinarySensor("Alarm Voltage low", "mdi:alert", "alarm/underVoltage", "1", "0");
            publishBinarySensor("Warning Voltage low", "mdi:alert-outline", "warning/lowVoltage", "1", "0");
        
            publishBinarySensor("Alarm Voltage high", "mdi:alert", "alarm/overVoltage", "1", "0");
            publishBinarySensor("Warning Voltage high", "mdi:alert-outline", "warning/highVoltage", "1", "0");
        
            publishBinarySensor("Alarm BMS internal", "mdi:alert", "alarm/bmsInternal", "1", "0");
            publishBinarySensor("Warning BMS internal", "mdi:alert-outline", "warning/bmsInternal", "1", "0");
        
            publishBinarySensor("Alarm High charge current", "mdi:alert", "alarm/overCurrentCharge", "1", "0");
            publishBinarySensor("Warning High charge current", "mdi:alert-outline", "warning/highCurrentCharge", "1", "0");
        
            publishBinarySensor("Charge enabled", "mdi:battery-arrow-up", "charging/chargeEnabled", "1", "0");
            publishBinarySensor("Discharge enabled", "mdi:battery-arrow-down", "charging/dischargeEnabled", "1", "0");
            publishBinarySensor("Charge immediately", "mdi:alert", "charging/chargeImmediately", "1", "0");
            break;
        case 1: // JK BMS
            // numeric sensors
            publishSensor("Battery voltage", NULL, "BatteryVoltageMilliVolt", "voltage", "measurement", "mV");
            publishSensor("Battery current", NULL, "BatteryCurrentMilliAmps", "current", "measurement", "mA");
            publishSensor("BMS temperature", NULL, "BmsTempCelsius", "temperature", "measurement", "°C");
            publishSensor("Battery temperature 1", NULL, "BatteryTempOneCelsius", "temperature", "measurement", "°C");
            publishSensor("Battery temperature 2", NULL, "BatteryTempTwoCelsius", "temperature", "measurement", "°C");
            publishSensor("State of Charge (SOC)", NULL, "stateOfCharge", "battery", "measurement", "%");
    
            // binary sensors
            publishBinarySensor("Charge enabled", "mdi:battery-arrow-up", "BatteryChargeEnabled", "yes", "no");
            publishBinarySensor("Discharge enabled", "mdi:battery-arrow-down", "BatteryDischargeEnabled", "yes", "no");
            publishBinarySensor("Balancing enabled", "mdi:battery-sync", "BalancingEnabled", "yes", "no");
    
            // common sensors        
            publishSensor("Battery Cycles", "mdi:counter", "BatteryCycles");
            publishSensor("Battery Cycle Capacity", "mdi:counter", "BatteryCycleCapacity");
            publishSensor("Status", "mdi:list-status", "StatusBitmask");
            publishSensor("Alarm", "mdi:bell", "AlarmsBitmask");
        break;
    }
```

ButterBetzi, 2024-02-04:
is this issue still being worked on/is there a solution?

schlimmchen, 2024-02-05:
@ButterBetzi @NevelSavage Please have a look at #640. You can find respective artifacts to test at the bottom of https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/7790264715?pr=640 (this would be helgeerbe/development plus the Home Assistent integration I comleted). Your feedback is welcome (put it in #640).

ButterBetzi, 2024-02-05:
flashed it and it works! Thank you! :)

commit 18651138 (2024-02-09): Feature: JK BMS Home Assistent integration

* pylontech HA integration: remove unused method/variable

* make MqttHandlePylontechHassClass::publishConfig() private.
  there are no outside users of that method.

* rename to MqttHandleBatteryHass

* battery HA integration: merge methods and bring back forceUpdate().
  even though the forceUpdate() method was not in use before, it makes
  sense to implement it and use it when the battery config changes.
  rather than controlling a separate flag, it now changes the _doPublish
  flag of the class, which also triggers publishing the device config to [... cut]
closed 2024-02-09 by helgeerbe