hoylabs/OpenDTU-OnBattery #507: DPL seems to sometimes not shut down the inverter

schlimmchen, 2023-10-17:
### What happened?

Has anybody ever observed that his/her battery was depleted until the BMS had to protect it, even though the DPL stop limit was in place and well above a level where the BMS protection thresholds would kick in?

Unfortunately, I observed a hand full of times now, that my battery "suddenly" was depleted, even though it really should not have been. Right now it sits at 3.06V per cell. The BMS must have kicked in at 3.000V while discharging.

I never had the chance to observe the behavior while the battery was being used even though the stop threshold was reached. The battery was always already at 0% SoC and the BMS shut off discharging every time I saw this issue. So I am not 100% sure that maybe the BMS estimated a state of charge of 16%, the DPL was happily discharging (stop threshold at 15%), and suddenly the BMS decided that 0% would be a more accurate estimate for the SoC. This makes even less sense now that I observed the issue while the stop threshold was set to 25% (I am trying to keep the battery at a greater charge for the cold and dark months ahead). That's why I am assuming that the inverter did indeed discharge the battery until the BMS had to protect it.

Also, this issue only occurred a couple of times now, even though the battery is discharged to the stop limit practically every day for the last 4 weeks at least.

Shame on me for still not having data logging, so I could confirm that the battery was actually discharged gradually rather than the BMS reporting a big jump down from 16% or even 25% to 0% SoC...

The only explanation I can find for this issue in the code is the DPL shutdown timeout. It is set to 10 seconds. However, even with a polling interval of 1 second, I know that the inverter data is sometimes very out of date, such that the bar in the web app becomes red. It is absolutely conceivable, that 10 seconds is too short. On the other hand, if the inverter shutdown timed out, the DPL is free to do another full loop, where it should detect that the stop treshold is reached, 0W should be the new power limit (assuming no solar power available), so the shutdown() function is called once more. It then turns out the inverter is still producing, so it receives another shutdown command.

### To Reproduce Bug

Hard to tell, I wish I knew.

### Expected Behavior

Inverter should be off as soon as stop threshold is reached.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

b461aff6

### Anything else?

I will keep HomeAssistent running (it was off since the last reboot of my PC :man_facepalming: ) and will log the UART output. I hope to capture information when this issue occurs again. In the meantime I am curious if somebody else experienced this.

JaMo523, 2023-10-18:
> Has anybody ever observed that his/her battery was depleted until the BMS had to protect it, even though the DPL stop limit was in place and well above a level where the BMS protection thresholds would kick in?

So far i didn't noticed any problems with the DPL SoC stop limit. I use 20% and saw that the battery is on the next day directly to 21%. Maybe i can not see the problem because of the capacity of 7 kWh.

Did you thought about communication problems? Maybe the DPL doesn't gets the SoC and switches to voltage stop limit.

schlimmchen, 2023-10-18:
> Did you thought about communication problems? Maybe the DPL doesn't gets the SoC and switches to voltage stop limit.

Actually, yes. However, the communication with the JK BMS is very reliable. I think I can rule that out. There could be other reasons why the voltage stop limits are used as a fallback, i.e., code bugs. However, my voltage thresholds are even more conservative/restrictive, so that also makes no sense to me.

MalteSchm, 2023-10-19:
@schlimmchen 
Good that you opened this ticket. I also had this case recently. What I found was:
- The inverter ran with high power until the battery was empty. 
- The inverter switched off or was cutoff by the BMS (unclear) but it stopped producing power
- The openDTU UI showed the inverter being not reachable (for long long time)

I think I manually started the Huawei PSU then to charge the battery. This caused the inverter to start again and I believe (but I'm not certain) that I was not able to turn it off using the openDTU UI. I then did a quick ESP restart (maybe too quick as I was not able to analyze the issue) which brought back control.

I thought if an extended watchdog that monitors if the Inverter is not reachable for a long time might be a good thing to have.

schlimmchen, 2023-10-19:
Thanks for commenting, @MalteSchm. The last time this happened to me the OpenDTU UI showed the inverter with power set to the lower power limit.

> I then did a quick ESP restart (maybe too quick as I was not able to analyze the issue) which brought back control.

That's scary. That should not be necessary...

> I thought that an extended watchdog which monitors if the Inverter is not reachable for a long time might be a good thing to have.

Oh, okay, you mean that the DPL observes whether or not the inverter it should be controlling is responsive or not, and if it is not for a longer time, the whole system is restarted? I see how that makes sense, however there are new problem then: The ESP should not restart over and over again if the battery is offline for maintenance, or if the BMS kicked in, or the fuse blew and must be reset.

I've had this multiple times now and I think that I did not always restart the ESP. That's a gut feeling only... I know that I updated the firmware for other reasons at least once when the battery was already drained. Like I did when I created this issue... Other than that I cannot remember that I could only fix the situation by restarting. It fixed itself once the sun was up and the MPPT was producing power and charging the battery.

Today it didn't do it wrong. It switched to solar passthrough once the stop threshold was reached, as intended. Logs are recorded and HomeAssistent is running. I hope to gain more insight once the issue occurs again -- whenever that will be, it can certainly be days.

Feldsalat, 2023-10-20:
Ich hatte die Tage gesehen, dass mein Inverter nur 15 Watt einspeist, obwohl mehr möglich und gefordert war. 
Bei Home Assistant im Feld "Limit Persistent Absolute" stand 48W, was gepasst hätte. Der Power Limiter zeigte "reusing old limit" in der Konsole. Habe dann einen anderen Sollwert eingetippt und es funktionierte wieder alles wie erwartet. Das passt zwar nicht ganz auf das Problem hier, aber vielleicht ist es ja etwas in der Richtung.

spcqike, 2023-10-20:
@Feldsalat das klingt eher nach #478, welches bereits behoben sein sollte. Wenn das Problem nochmal auftritt, beobachte mal die Konsole, ob nach spätestens 60s das Limit erneut gesendet wird. 
Durch Fehler in der Übertragung kann es passieren, dass ein eventuell neues Limit, deine 48W, nicht beim Inverter ankommen. Ändert sich die Leistungsanfordernung nicht, wird auch kein neues (das gleiche) Limit übertragen. Das passiert nun aber aller 60s.

Feldsalat, 2023-10-20:
@spcqike  Danke für die Info. Habe länger kein Update gemacht, da alles lief und ich anderes zu tun hatte.

schlimmchen, 2023-10-20:
Danke @spcqike, sehr gut aufgepasst ☺️

schlimmchen, 2023-11-01:
It happened again this afternoon and this time I have data and learned something.

Background: Our battery is installed in our shed. It is not heated. I thought I would add heating, but I decided against it. Instead I reconfigured to BMS to be very conservative: Shut off charging/discharging if the battery is near 0°C. Also, I setup the DPL to keep the battery at 40%, since the battery should have a significant charge while it is basically unused (stored). The BMS is setup for a very conservative cut-off voltage of 3.18V per cell (keep a rather high minimum charge in the cells). That became a problem just now.

We were cooking a little, and the battery was used since it is rather sunny, so the HM-1500 was putting out 1500W AC. The battery voltage dropped below 51.2V. One cell was obviously lower than 3.18V, so the BMS shut off. What the BMS also did was reset the SoC to 0% (from 43%) in that moment.

That probably is what happened to me multiple times now, since the BMS under-voltage-protection-thresholds have always been set to very conservative values by me.

[image]

So I now have a LiFePo4 battery at 3.31V per cell that my BMS thinks has 5% SoC... Very interesting. This means the JK BMS SoC reading only makes sense if the voltage thresholds are setup "correctly".

I am confident that this is my issue and there is no issue with the DPL. For that reason, I am closing this issue. @MalteSchm may open a new one for his observation.

closed 2023-11-01 by schlimmchen

spcqike, 2023-11-02:
@schlimmchen that makes sense, as thats what nearly every battery powered device does.  some years ago i wonderd, why my smartphone shut down at like 10% SoC. yeah, i read and learned, that the device measures and calculates the charged and used energy in Wh (or Ah?) and uses this Wh to calculate the SoC. but as batterys degrade over time, it happens that the total capacity gets lower. if you never "drain" your battery to "0" (shut down because of under voltage protection), and fully charge it afterwards, your SoC state will not be true later. 

for a consumer device like a smartphone, that makes totally sense. if it dies at 10% SoC and you than charge it to 100%, the device knows the remaining capacity and can calculate the SoC states correctly, again.

i never thought about this for a bigger battery (bought or DIY with BMS and stuff), but i wouldn't have guessed, that it calculates the SoC based on your own thresholds. but this is important to know and somewhat logical.

can you set thresholds for voltage and SoC? 

in your case:
if you charge your battery to 100%, the SoC range 0-100 should match your cell voltage 3.18V-3.65V, shouldn't it? so if you still use openDTU's SoC level like 20%, you would stop discharging at ~3.2V (and only use like 65% of your rated capacity) 

> I am confident that this is my issue and there is no issue with the DPL.

are you sure? if my consideration is correct, and this "wrong" cutoff has allready happened once, the BMS should now its "new" SoC states allready.

schlimmchen, 2023-11-02:
> are you sure? if my consideration is correct, and this "wrong" cutoff has allready happened once, the BMS should now its "new" SoC states allready.

I recently fiddled again with the thresholds, to make the battery "fit for the winter". So yes, I think this is my own fault.