hoylabs/OpenDTU-OnBattery #532: Enhanced Battery Monitoring and Protection Features for Smartshunt

swingstate, 2023-11-20:
### Is your feature request related to a problem? Please describe.

**Feature Request for Smartshunt Enhancement; building up on #474:**

Description:
The Smartshunt is capable of measuring the midpoint voltage and its deviation, which is particularly vital for batteries configured in series (such as 2x 12V or 2x 24V). This functionality plays a key role in monitoring battery health and performance, thus should be included in onBattery and made available through MQTT.

### Describe the solution you'd like

**Proposed Enhancements:**

**1. Improved Protection Logic in onBattery Module:** The onBattery module could be upgraded to include advanced protection mechanisms. This would involve:
   - Automatically halting charging or discharging processes if the midpoint voltage deviates excessively.
   - Stopping charging or discharging based on a user-defined Ampere-hour (Ah) threshold, ensuring better control over battery usage.

**2. Expansion of Sensor Data for Battery Monitoring:** To offer a more comprehensive understanding of battery health, the following sensor data points should be incorporated:

   - Minimum and Maximum Battery Voltage: To track voltage fluctuations and prevent overcharging or deep discharge.
   - Midpoint voltage: To track heat of the battery bank
   - Midpoint voltage deviation: To track drift between the batteries
   - Instantaneous Power: For real-time power usage monitoring.
   - Depth of Deepest Discharge: To assess battery strain and longevity.
   - Date of Deepest Discharge: For historical data analysis and battery health assessment.
   - Last Full Charge: To monitor charging cycles and optimize battery maintenance.

**3. Integration with MQTT for Enhanced Monitoring and Automation:** All these values should be accessible via MQTT. This integration will facilitate advanced monitoring and enable automations based on battery status.

**4. Web UI Improvements:** Adding Instantaneous Power and Midpoint Voltage data (only for serial configurations) to the web UI will enhance user experience by providing easy access to key battery information.

This request is inspired by the ESPHome intrgration for VE.Direct -[ Code example](https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp). 

### Describe alternatives you've considered

The ESPHome based integration of a SmartShunt into MQTT / Homeassistant quires a second Shunt in the circuit. 

### Additional context

Additional Context Enhancement:

Over the past few months, I have conducted extensive testing with two shunts integrated within a single circuit (Since the shunt sensor data made available in onBattery is still limited).
This dual-shunt configuration has proven to be operationally viable, yet it comes with its own set of challenges. Firstly, incorporating an additional shunt incurs extra expenses, an important factor to consider for cost-efficiency. 

Moreover, there's a noticeable impact on the circuit's overall performance. The inclusion of a second shunt introduces added resistance and complexity, which can affect the efficiency and energy flow within the circuit. Adding a second shunt also increases the risk for failures within the circuit. 

Another critical observation is the variation in measurement accuracy between the two shunts. Due to their slightly different placements within the circuit, each shunt records differing values. Over time, this disparity in measurements tends to amplify, leading to a significant drift in the data accuracy. This divergence in readings poses a challenge for precise monitoring and may impact decision-making based on these measurements.

Extending the SmartShunt integration in onBattery can help to streamline circuit configurations and improve Battery safety.

schlimmchen, 2024-12-01:
@swingstate Aren't points 2 through 4 completed in the meantime?

Regarding point 1: I suggest to implement a discharge current limit based on the midpoint deviation to achieve the result you are asking for (stop discharging the battery through the DPL inverters). What would be the logic/formula to decide whether or not to prevent discharging the battery? Does that need a configuration item or can we make an informed decision based on the info we have, such as the midpoint deviation and the total voltage, so we can calculate a relative midpoint deviation and trigger at some reasonable threshold.

swingstate, 2024-12-08:
Agree to 2 - 4.

Regarding 1 - the Victron Smartshunt can provide the mid-point deviation (DM). It shouldn't exceed 1%, ideally be maximum 0,5%. I suggest implementing a logic which starts to lower the discharge (also need to consider the charge current at some point) current starting at 0.4% and then stop discharging at 1% deviation. I think making this a default protection mechanism for batteries, which the user can enable, would make sense. Also there should be default values which the user should be allowed to change. 

Example steps: 

At 0,4% - reduce current by 25% and limit current to that new limit until deviation reduces
at 0,6V minus another 25%
0,8V minsur 25
1% stop discharge. 

Same logic could be applied for charging.

schlimmchen, 2025-02-01:
The SmartShunt battery provider could read the midpoint deviation and impose a discharge limit based on that value. The discharge limit is a feature which is part of OpenDTU-OnBattery now. Possibly one needs to take care the discharge limit is imposed even if the user did not enable "limit discharge current". The user must be informed somehow that the battery discharge is inhibited because of the midpoint deviation.

Also, very importantly, one needs to determine first whether or not the midpoint deviation is actually available, as the SmartShunt not always is setup such that it can measure this, or it protects a singular battery where no such reading is even possible.