hoylabs/OpenDTU-OnBattery #537: Dynamic Power Limiter - target power from grid mismatched

DerWe, 2023-11-26:
### What happened?

I just set up openDTU on Battery on an openDTU Fusion board, used to control a Hoymiles HMS1600.
When I set up the Dynamic Power Limiter (Target Power to 30). But when it starts regulating the power limit, the Grid meter idles about -50 to -80 watts, which is approximately 100 Watts lower than the actual target.

This even happens when I setup a lower power limit for the Inverter. In this case, it exceeds the Power limit and continues till it reaches -50 to -80 watts.

[image]

The data from the grid meter is retrieved by http every 3-5 Seconds.
When disabling the Dynamic Limiter and setting a limit manually, it is working perfectly.

What logs can I provide for further Information?

### To Reproduce Bug

Connect to HMS1600 and setup the Power limiter to the following Settings:
[image]

### Expected Behavior

The Power is regulated within the Limits and the configured Target is reached

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

dd8446d

### Anything else?

I used the self compiled version, because there I can compile it for openDTU Fusion. The standard versions seemed not to be compatible with this board

schlimmchen, 2023-11-26:
> I used the self compiled version, because there I can compile it for openDTU Fusion. The standard versions seemed not to be compatible with this board

Files `opendtu-onbattery-generic_esp32s3_usb.bin` and `opendtu-onbattery-generic_esp32s3_usb_factory.bin` in every release are meant for the processor on the OpenDTU-Fusion v2 board. Did you configure your `pin_mapping.json` properly?

> What logs can I provide for further Information?

Enable DPL verbose logging. Then copy a portion of the output of the web console: In the WebApp click `Info`, then `Console` and paste it here in between a pair of three backticks ` (or use the "Code" button in the toolbar above the textbox).

DerWe, 2023-12-01:
Ich hab das und ein paar andere Dinge mal ausprobiert:

1. Ich habe heute mal per Web UI auf die von @schlimmchen vorgeschlagene Version `opendtu-onbattery-generic_esp32s3_usb.bin` gewechselt. Daraufhin habe ich ihn neu starten lassen.
2. Ich habe den DPL wieder aktiviert und die Werte etwas variiert:

```
17:09.10 - DPL AN - Target +30W  - upper Power limit 600
17:17.20 - DPL AN - Target -30W  - upper Power limit 600
17:22.20 - DPL AN - Target +100W - upper Power limit 600
17:25.50 - DPL AN - Target +100W - upper Power limit 100
```

(Die Daten des Netzstromzählers kommen im Home Assistant nur alle paar sekunden an. Die DTU fragt diese aber per http request direkt von dem Tasmota SML Microcontroller ab und hat somit aktuellere Werte)
[image]

Hier nochmal ein paar auszüge aus den Logs. Die vollen Logs hab ich mir gespeichert und kann ich euch gerne noch geben, falls hier das wichtige nicht dabei ist.

```
17:14:36.242 > [    806.777] DPL: waiting for sufficiently recent power meter reading
17:14:39.928 > PowerMeterClass: TotalPower: 183.00
17:14:39.983 > [DPL::loop] ******************* ENTER **********************
17:14:40.029 > [DPL::loop] battery interface disabled, SoC: 0 %, StartTH: 80 %, StopTH: 20 %, SoC age: 810 s
17:14:40.132 > [DPL::loop] dcVoltage: 48.50 V, loadCorrectedVoltage: 48.67 V, StartTH: 48.00 V, StopTH: 47.00 V
17:14:40.233 > [DPL::loop] StartTH reached: yes, StopTH reached: no, inverter is producing
17:14:40.338 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:14:40.384 > [DPL::loop] battery discharging allowed, PowerMeter: 183 W, target consumption: 30 W
17:14:40.438 > [DPL::setNewPowerLimit] requested: 327 W, (re-)sending limit: 327 W
17:14:40.541 > [DPL::loop] ******************* Leaving PL, calculated limit: 327 W, requested limit: 327 W (updated from calculated)
17:14:40.593 > TX ActivePowerControl 865.00 MHz --> 51 91 54 67 85 80 16 82 20 81 0B 00 0C C6 00 00 0D E2 ED 
17:14:40.744 > [    810.759] DPL: waiting for a power limit command to complete

...

17:14:47.819 > [    818.288] DPL: waiting for sufficiently recent power meter reading
17:14:50.302 > PowerMeterClass: TotalPower: -51.00
17:14:50.345 > [DPL::loop] ******************* ENTER **********************
17:14:50.395 > [DPL::loop] battery interface disabled, SoC: 0 %, StartTH: 80 %, StopTH: 20 %, SoC age: 821 s
17:14:50.438 > [DPL::loop] dcVoltage: 48.20 V, loadCorrectedVoltage: 48.61 V, StartTH: 48.00 V, StopTH: 47.00 V
17:14:50.480 > [DPL::loop] StartTH reached: yes, StopTH reached: no, inverter is producing
17:14:50.522 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:14:50.574 > [DPL::loop] battery discharging allowed, PowerMeter: -51 W, target consumption: 30 W
17:14:50.617 > [DPL::setNewPowerLimit] requested: 325 W, last limit: 327 W, diff: 2 W, hysteresis: 10 W, age: 10445 ms
17:14:50.660 > [DPL::loop] ******************* Leaving PL, calculated limit: 325 W, requested limit: 327 W (kept last requested)
17:14:50.702 > [    821.116] DPL: the system is stable, the last power limit is still valid
17:14:50.746 > [DPL::loop] ******************* ENTER **********************
17:14:50.796 > [DPL::loop] battery interface disabled, SoC: 0 %, StartTH: 80 %, StopTH: 20 %, SoC age: 821 s
17:14:50.842 > [DPL::loop] dcVoltage: 48.20 V, loadCorrectedVoltage: 48.61 V, StartTH: 48.00 V, StopTH: 47.00 V
17:14:50.996 > [DPL::loop] StartTH reached: yes, StopTH reached: no, inverter is producing
17:14:51.042 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:14:51.238 > [DPL::loop] battery discharging allowed, PowerMeter: -51 W, target consumption: 30 W
17:14:51.404 > [DPL::setNewPowerLimit] requested: 325 W, last limit: 327 W, diff: 2 W, hysteresis: 10 W, age: 10705 ms
17:14:51.451 > [DPL::loop] ******************* Leaving PL, calculated limit: 325 W, requested limit: 327 W (kept last requested)
17:14:51.496 > [DPL::loop] ******************* ENTER **********************
17:14:51.539 > [DPL::loop] battery interface disabled, SoC: 0 %, StartTH: 80 %, StopTH: 20 %, SoC age: 821 s
17:14:51.581 > [DPL::loop] dcVoltage: 48.20 V, loadCorrectedVoltage: 48.61 V, StartTH: 48.00 V, StopTH: 47.00 V
17:14:51.623 > [DPL::loop] StartTH reached: yes, StopTH reached: no, inverter is producing
17:14:51.665 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:14:51.708 > [DPL::loop] battery discharging allowed, PowerMeter: -51 W, target consumption: 30 W
17:14:51.751 > [DPL::setNewPowerLimit] requested: 325 W, last limit: 327 W, diff: 2 W, hysteresis: 10 W, age: 11220 ms
17:14:51.906 > [DPL::loop] ******************* Leaving PL, calculated limit: 325 W, requested limit: 327 W (kept last requested)

...

17:15:49.809 > [DPL::loop] ******************* ENTER **********************
17:15:49.868 > [DPL::loop] battery interface disabled, SoC: 0 %, StartTH: 80 %, StopTH: 20 %, SoC age: 880 s
17:15:49.902 > [DPL::loop] dcVoltage: 48.10 V, loadCorrectedVoltage: 48.51 V, StartTH: 48.00 V, StopTH: 47.00 V
17:15:49.957 > [DPL::loop] StartTH reached: yes, StopTH reached: no, inverter is producing
17:15:50.000 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:15:50.161 > [DPL::loop] battery discharging allowed, PowerMeter: -49 W, target consumption: 30 W
17:15:50.205 > [DPL::setNewPowerLimit] requested: 333 W, last limit: 324 W, diff: 9 W, hysteresis: 10 W, age: 28213 ms
[... 2 lines cut ...]
[... cut ...]
17:20:55.631 > [DPL::loop] SolarPT disabled, Drain Strategy: 0, canUseDirectSolarPower: no
17:20:55.674 > [DPL::loop] battery discharging allowed, PowerMeter: -38 W, target consumption: -30 W
17:20:55.716 > [DPL::setNewPowerLimit] requested: 588 W, last limit: 589 W, diff: 1 W, hysteresis: 10 W, age: 9989 ms
17:20:55.759 > [DPL::loop] ******************* Leaving PL, calculated limit: 588 W, requested limit: 589 W (kept last requested)
17:20:55.802 > [   1186.225] DPL: the system is stable, the last power limit is still valid

...

```

DerWe, 2023-12-18:
Nach einiger Zeit, habe ich soeben herausgefunden, woher die Probleme kommen.

Ich habe zunächst weitere Tests mit manuellen Limits vorgenommen. Dabei ist mir aufgefallen, dass der HMS-1600 4T sich in unterschiedlichen Leistungsbereichen unterschiedlich verhält.

Alle 4 Eingänge Verbunden, Manuelles Limit
- 0W bis 232W Tatsächliche Leistung entspricht dem Limit
- 232W bis 599W Tatsächliche Leistung schwankt -+100W gegenüber dem Limit
- 600W bis 1600W Nicht Getestet

**Für den DPL bedeutet das, dass er nicht vernümpftig regelt, weil der Hoymiles von Haus aus so stark schwankt :(**

Nur 2 Eingänge Verbunden (1+2 oder 3+4), Manuelles Limit
- Manuelle Limits werden exakt eingehalten. Durch die getrennten Engänge entspricht die Tatsächliche Leistung nun immer 50% des Limits (Soweit klar, da der Hoymiles ja die Tracker und nicht den AC out drosselt)

Ob ich nun ein Montagsgerät erwischt habe oder ob das bei allen HMS-1600 4T so ist, ist fraglich. In der Community findet man dazu leider recht wenig. Mal schauen wie ich damit jetzt umgehe... Das Issue hier hat sich damit aber zum Glück erledigt :)

Testweise habe mal probiert, das Verhältnis aus angeschlossenen Eingängen und Tatsächlichem Limit im Code zu Implementieren. Dazu habe ich testweise in der src/PowerLimiter.cpp ganz stumpf den Rückgabewert der Funktion "PowerLimiterClass" mal 2 genommen.

```
int32_t PowerLimiterClass::calcPowerLimit(std::shared_ptr<InverterAbstract> inverter, bool solarPowerEnabled, bool batteryDischargeEnabled)
{
...

return newPowerLimit*2;
}
```

Mit der Änderung im Code und nur noch 2 Verbundenen Eingängen funktioniert es richtig gut und er pendelt konstant um den Vorgegebenen Wert. Auch das maximale Limit hält er ein, wenn man in den Einstellungen den Faktor 2 berücksichtigt.
Ich weiß nicht ob die Funktion die richtige Stelle dafür ist. Vielleicht könnte man die Anzahl der verbundenen Eingänge aber auch als richtige Einstellung des DPL implementieren. Es gibt sicher einige Varianten wie sowas möglich wäre.

schlimmchen, 2023-12-18:
Danke für deine Rückmeldung. Hast du den Teil im DPL gesehen, der das Power Limit anhand tatsächlich produzierender DC-Kanäle skaliert? Warum funktioniert das in deinem Fall nicht richtig? Wäre ehrlich gesagt schön, wenn diese Skalierung bei dir auseinanderfliegt, ich finde die Skalierung nämlich sehr problematisch.

Es ist im Übrigen nicht ernsthaft vorgesehen, trotz der Skalierung im Code, dass der Inverter mit nur einem Teil der Eingänge gespeist wird, insbesondere weil wir gar nicht wissen, wie der Inverter das Limit auf die MPPTs verteilt.

Hat dein Inverter vier MPPTs statt zwei?

Manos1966, 2023-12-18:
@schlimmchen  Korrekt. Der Hoymiles HMS-1600-4T hat vier MPPTs 
Schoenes Teil...  Max. Eingangsspannung: 65V (anstatt 60V der HM Serie)
Max. Eingangsstrom: 4*12,5A (anstatt 4*11.5A)
Max. Eingangskurzschlussstrom: 4*20A  (anstatt 4*15A)

DerWe, 2023-12-18:
Das dachte ich mir schon, dass es dabei bestimmt ein Problem gibt. Den skalierenden Code habe ich mir gerde angesehen. Ich schätze mal es dauert eine ganze Weile bis er merkt, dass 2 Eingänge nicht mehr angeschlossen sind. Die Internen Kondensatoren im Wechselrichter halten ihre Ladung recht lange. Das letzte Umstecken liegt erst etwa 3h zurrück, noch sind sie mit 7-10V geladen.
Das wäre zumindest eine Erklärung.

Ich werde erstmal wieder die normale Firmware draufspielen und dann morgen in aller frische schauen ob die automatische Skalierung dann Funktioniert. Die Änderung war wie gesagt nur ein Test den ich heute erst gemacht habe. Das hätte ich auch nicht so lassen wollen.

Die internen Schwankungen zeigt er sobald mehr als 2 Eingänge verbunden sind und er in en beschriebenen Bereichen arbeitet. Wenn die schwankungen eintreten sieht man auch, dass er nur auf den Eingängen 1&2 schwankt und Werte über dem prozentualen Kanallimit zieht (wenn er es denn intern überhaupt so berücksichtigt). Kanal 3&4 Bleiben immer konstant.

Genau, wollte ich auch gerade schreiben. Hatte ihn auf grund des breiteren Spannungsbereiches und der Stärkeren Stromverträglichkeit gekauft.

spcqike, 2023-12-19:
@schlimmchen 
> Wäre ehrlich gesagt schön, wenn diese Skalierung bei dir auseinanderfliegt, ich finde die Skalierung nämlich sehr problematisch.

Was siehst du denn als Alternative? Wenn jemand, warum auch immer, nur einen Eingang am WR belegt (sei es ein Paneel oder eine Batterie, wenn man ggf. nur <= 300W erzeugen will und halt einfach einen >HM-600 genommen hat), braucht man die Skalierung, oder? (Ginge es nach mir, wäre sie sogar dynamischer und flexibler, um auch mit batterielosen Anlagen und unterschiedlicher Ausrichtung/Verschattung möglichst immer den gewünschten AC-Limit Wert zu erreichen 😄 aber das Thema hatten wir bereits anderswo, ja meine "Nische" 🙈 )

Für mich wäre die sauberste Lösung, wenn die Wechselrichter nicht die MPPT Eingänge sondern den AC-Ausgang entsprechend drosseln, aber das tun sie nicht und das liegt auch nicht in den Händen der Community.

@DerWe 
> Ich schätze mal es dauert eine ganze Weile bis er merkt, dass 2 Eingänge nicht mehr angeschlossen sind. Die Internen Kondensatoren im Wechselrichter halten ihre Ladung recht lange. Das letzte Umstecken liegt erst etwa 3h zurrück, noch sind sie mit 7-10V geladen.
Das wäre zumindest eine Erklärung.

das kann ich mir nicht vorstellen. Wie hast du das denn gemessen? ja, der Kondensator mag die Spannung relativ lange halten, aber wenn der WR einspeisen will und die Kondensatorspannung > 22V ist (22V war doch die Startspannung, oder?) sollte er Versuchen auf dem Eingang was zu erzeugen. Damit würde er den Kondensator entladen. auf 0V, oder im Zweifel unter seine Abschaltspannung (16V?). Das sollte dann auch als "nicht Produktion" erkannt werden.

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/fb2ca28692fe8b197c0092ab25792a1c71a9067b/src/PowerLimiter.cpp#L531-L536C6

es wird ja geprüft, ob der Eingang > 2W erzeugt. 

Auch für den HMS mit 4 Trackern, der ja scheinbar anders arbeitet als die HMT mit 2 Trackern, sollte das reichen. man kann ja aus 10V, wo der Eingang gar nicht aktiviert wird, keine 2W erzeugen. zumindest nicht dauerhaft 😄 

Um dem Regelverhalten des HMS auf den Grund zu kommen, wie ist denn dein Aufbau? Hast du alle Eingänge mit Paneelen belegt oder mit einer BAtterie? oder gar eine Mischung? 
Kannst du ggf. bei mehreren Laststufen (50, 100, 150, 250, 500, 1000 oder so?) die Spannungen und Leistungen/Ströme an den einzelnen Eingängen und die Gesamtleistung (AC) des Wechselrichters aufnehmen? 

Ich vermute eigentlich, dass der Wechselrichter mit 4 Eingängen und 4 MPPT "genauso" arbeitet, wie die mit 2 Eingängen und 2 MPPT. Die Eingänge werden gleichmäßig gedrosselt. Aber wissen tu ich es natürlich nicht :D (schön wäre es, würde Hoymiles einfach den AC Ausgang auf den gewünschten Wert drosseln und den Wechselrichter "sein Ding" machen lassen) Beim HMT mit 4 Eingängen und 2 Trackern, ist es ja nicht so "einfach", da dort ja immer 2 Eingänge intern parallel sind ,oder?

DerWe, 2023-12-19:
@spcqike 
> das kann ich mir nicht vorstellen. Wie hast du das denn gemessen?

Ich hatte die Spannungen der Einzelnen Strings von OpenDTU betrachtet. Diese blieben lange erhalten. Der Wechselrichter hat entsprechend erst eine ganze weile Später das Event "211 PV-X No input" gemeldet.

Nun muss ich aber auch dazu sagen, dass ich beim abklemmen der Strings 3&4 den Fehler gemacht habe nur die + zu trennen. (Ich dachte passt schon --> Stromkreis offen). Falsch gedacht. Offenbar sind alle Masseleitungen der DC Seite intern verbunden. Dazu kommt, dass der Strom intern im HMS-1600 wohl auch auf den einzelnen DC Masseleitungen gemessen wird.
Die Konsequenz: Ein teil des Stromes ist auch über die inaktiven Eingänge 3&4 Abgeflossen, wurde dort gemessen und auch in der DTU dort angezeigt. Ich hatte mich schon immer gewundert warum er bei 3&4 noch eine Leistung anzeigt - Jetzt weiß ich es.

> es wird ja geprüft, ob der Eingang > 2W erzeugt.

Mit der Erklärung vorweg erscheint mir auch logisch warum er die inaktiven Strings nicht erkannt hat.
Als Gegenprobe hab ich die Masseleitung bei 3&4 heute auch abgemacht und wieder den DPL gestartet. Nun funktioniert es mit nur 2 verbundenen Eingängen auch mit der Standardfirmware richtig schön. 1000 Dank für all die Mühen, die ihr dort schon reingesteckt habt!
[image]

> Um dem Regelverhalten des HMS auf den Grund zu kommen, wie ist denn dein Aufbau?

Ich hatte ursprünglich alle 4 Eingänge per Batterie betrieben, aktuell zwecks troubleshooting nur 2, die anderen beiden hängen nun wirklich in der Luft.
Als Batterie kommt eine Pylontech zum Einsatz, die auch per CAN mit der DTU verbunden ist. Zwischen Hoymiles und Pylontech gibt es einen Schütz, und ein Zeitrelais, mit denen automatisch über Vorladewiederstände alle Eingänge vorgeladen und Zugeschaltet werden können. Außerdem sind alle Eingänge noch einzeln mit einer 13A Sicherung in der + Leitung abgesichert.

>Kannst du ggf. bei mehreren Laststufen (50, 100, 150, 250, 500, 1000 oder so?) die Spannungen und Leistungen/Ströme an den einzelnen Eingängen und die Gesamtleistung (AC) des Wechselrichters aufnehmen?

Klar! mache ich die Tage und lade ich dann hoch. In der Tat ist es so, dass bei 2 Verbundenen Eingängen, die Leistung gleichverteilt ist. Sind alle 4 Verbunden, haben String 1&2 eine andere Leistung, als 3&4.
Das könnte tatsächlich zu dem Problem der instabilität führen. Ich hatte auch schon probiert Shottky Dioden hinter jeden Eingang zu schalten um mögliche Rückwirkungen zu verhindern, allerdings hat das nicht geholfen.

@schlimmchen 
>Wäre ehrlich gesagt schön, wenn diese Skalierung bei dir auseinanderfliegt, ich finde die Skalierung nämlich sehr problematisch.

Ich bin gerade ziemlich froh, dass es diese Skalierung gibt und dass sie nun auch wie gewollt funktioniert. Klar wäre es am besten, wenn der Wechselrichter AC-Seitig regelt. Aber da haben wir ja leider keinen Einfluss drauf.
Das heißt jetzt nicht, dass ich es so lassen will. Ich fände es viel schöner wenn es auch mit alle 4 laufen würde und werde weiter sachen ausprobieren. Allerdings läuft es damit schon mal und das auch erschreckend gut!

schlimmchen, 2024-04-12:
@DerWe Kannst du einmal zusammenfassen, was wir hier diskutieren? Was ist das offene Problem?

DerWe, 2024-04-13:
@schlimmchen Entschuldigt bitte, ich dachte ich hätte das Issue schon geschlossen.

Das Problem konnte ich am Ende beim HMS 1600 4t eingrenzen. Wenn man diesen an einer Batterie betrieben hat, zeigte sich die folgende Anomalie: (y Achse Tatsächlich ausgegebene Leistung; x Achse Wert des Power Limiters)
[image]

**Wann zeigte sich der Fehler:** Wenn man die "-" Eingänge von Eingang A&B auf das selbe Potential wie An Eingang C&D legt --> Passiert ja automatisch wenn man alle Eingänge per Batterie betreibt.

_Vermutung warum:_ Misst man die Eingänge ist ersichtlich, dass die Eingänge wohl nicht Potentialfrei sind, sondern die "-" Leitung intern im Wechselrichter auf ein Potenzial gelegt sind. Dadurch kommt es wohl dazu, dass Srom über Eingang "A+" reinkommt am Ende über Eingang "C-" abfließt. (Gemessen mit einer Stromzange konnte man das auch dort nachweisen) Die internen Messergebnisse konnte er wohl nicht richtig interpretieren und regelte sich dann falsch ein.
 
**Wann geht es:** Wenn man Ihn nur an den Eingängen A&B oder C&D einzeln betrieben hat ging es problemlos. Aus den Erfahrungen heraus würde ich den HMS 1600 4t aktuell nicht zum Einspeisen von Batterie empfehlen.

closed 2024-04-13 by DerWe

Manos1966, 2024-04-13:
Nur zur Sicherheit: 
Hatten alle deine minud Kabel zwischen Hoymiles und Batterie die gleiche Länge gehabt ? 🤔

DerWe, 2024-04-13:
@Manos1966
Ich verstehe wie du darauf kommst, das Diagramm lässt es vermuten... Allerdings sind alle 1,2m lang in 6mm2 Kupfer (Ja ich weiß, 4mm2 hätte wohl auch gereicht😅)
Um auch mögliche Übergangswiderstände auszuschließen hatte ich auch mal die Kabel von A&B mit denen von C&D vertauscht. Der Logik nach hätte die Schieflast sich dann ja umkehren müssen. Dem war allerdings nicht so.

DonJohnLong, 2024-04-13:
Ich hab einen hms-2000 4t und der läuft perfekt 🤔, kann so ein problem nicht feststellen. Vieleicht inverter defekt?

spcqike, 2024-04-13:
> Um auch mögliche Übergangswiderstände auszuschließen hatte ich auch mal die Kabel von A&B mit denen von C&D vertauscht

komplett getauscht oder nur am wechselrichter? Und das andere Ende so gelassen wie es ist?