hoylabs/OpenDTU-OnBattery #548: Configuration Huawei AC-Charger: please reduce the minimal value for "Minimale Leistung"

dragricola, 2023-12-10:
### Is your feature request related to a problem? Please describe.

Inzwischen habe ich openDTUonBattery mit einem Huawei AC-Charger erfolgreich zum Laufen gebracht. Vielen Dank für die Zurverfügungstellung dieses sehr schönen Projektes mit dessen Hilfe sich die Wirtschaftlichkeit einer PV-Anlage mit Batterie sehr steigern lässt!
Mir ist jedoch aufgefallen, dass sich bei der Konfiguration für den AC-Charger leider keine Werte für die minimale Lestung kleiner 100 W einstellen lassen. Das führt dazu, dass sich der Charger frühestens bei einem Überschuss von 100W einschaltet und, wenn die Leistung des AC-Chargers unter 100W fällt, sofort wieder ausschaltet. Geringere Überschüsse als 100W landen somit nicht in der Batterie. Die folgende Abbildung zeigt das Verhalten bei Überschussleistung um 100 W.
[image]
rot: PV-Leistung, gelb: davon im Hausnetz verbraucht, dunkelblau: Leistung des AC-Chargers, Blau: ins öffentliche Netz zurückgespeist, grün: aus dem öffentlichen Netz bezogen. Die grüne Fläche ist ein Maß für die aus dem Netz bezogene und die blaue Fläche ist ein Maß für die nicht in die Batterie sondern ins öffentliche Netz eingespeiste Energie. Im Bereich um 100W kommt es zum mehrfachen Ein- und Ausschalten.

### Describe the solution you'd like

Aus welchem Grund besteht diese Einschränkung?
Im manuellen Betrieb (MQTT Topic: huawei.cmd.mode = 0) kann der Huawei problemlos auch kleine Leistungen von wenigen Watt einstellen und die Regelung reagiert auf Änderungen in der PV-Produktion oder Laständerungen im Hausnetz sehr schnell (wie im Verauf zu sehen). Falls keine gravierenden Gründe dagegensprechen, bitte ich darum, diesen Minimalwert in der nächsten Version herunterzusetzen (auf z.B. 10W) um auch bei geringer Solareinstrahlung, wie es im Winter leider öfter vorkommt, die Batterie öfter und mehr laden zu können. Um ein zu häufiges Ein- und Ausschalten zu vermeiden, wäre auch eine gewisse Hysterese (z.B. +/-10% vom Minimalwert) hilfreich.

MalteSchm, 2023-12-19:
Drei Kommentare dazu:

Der Wirkungsgrad des Netzteils geht bei kleinen Leistungen ziemlich in den Keller. Daher hatte ich da mal 100W als minimale Leistung vorgesehen. Ist glaube ich nur eine Limit in der UI

Die Abschaltung des Netzteils ist so gelöst: liegt der Strom unter 1A wird nach 1 Minute abgeschaltet. Das bedeutet aber im Moment auch 50W minimal. 

Eine Hysterese wäre sinnvoll. Eigentlich müsste das dann in der UI einstellbar sein. Das ist dann eine größere Sache

Evtl komme ich zwischen den Jahren mal dazu das Limit auf 50W zu ändern

dragricola, 2023-12-20:
> Der Wirkungsgrad des Netzteils geht bei kleinen Leistungen ziemlich in den Keller.

So schlecht ist der Wirkungsgrad bei kleiner Leistung gar nicht. Ich habe mal gemessen:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 12">
<link id=Main-File rel=Main-File
href="file:///C:\Users\HANS-P~1\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\HANS-P~1\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<style>

</style>
</head>

<body link=blue vlink=purple>

Stromlimit / A | P_AC / W | U_DC / V | I_DC / A | P_DC / W | eta / %
-- | -- | -- | -- | -- | --
< 0,5 | 5,0 | 51,00 | 0 | 0,0 | 0,0
0,55 | 43,0 | 51,06 | 0,6 | 30,6 | 71,2
1 | 68,0 | 51,50 | 1 | 51,5 | 75,7
1,5 | 102,9 | 51,59 | 1,55 | 80,0 | 77,7
2 | 133,4 | 51,36 | 2,1 | 107,9 | 80,9
3 | 192,6 | 51,70 | 3,05 | 157,7 | 81,9
4 | 232,0 | 51,77 | 4,1 | 212,3 | 91,5
5 | 284,7 | 51,87 | 5,1 | 264,5 | 92,9

</body>

</html>

P_AC wurde am Eingang des Huawei gemessen mit einem Leistungsmesser, der auch kleine Leistungen über 1W sehr genau anzeigen kann. Die anderen Werte sind aus openDTUonBattery und dem BMS der Batterie, deren Plausibilität und Genauigkeit ich mit Stromzange und Voltmeter überprüft habe. 
Ein leerlaufender Huawei hat 5 W Leerlaufverluste. Deshalb ist es aus energetischer und wirtschaftlicher Sicht sinnvoll, den AC-Charger laden zu lassen, wenn er einen DC-Strom größer 0,5A entsprechend 25W abgeben kann (Werte kleiner gleich 0,5 A können nicht eingestellt werden). 
>Eine Hysterese wäre sinnvoll. 

Aus meiner Sicht wäre folgende Hysterese sinnvoll: Huawei minimale Einschalteschwelle bei 50W, minimale Ausschaltschwelle bei 30W. 
>Eigentlich müsste das dann in der UI einstellbar sein. Das ist dann eine größere Sache

Nicht unbedingt! Die minimale Ausschaltschwelle von 30W könnte man auch fest programmieren, um das nicht im UI machen zu müssen. Eventuell braucht man gar keine Ausschaltschwelle konfigurierbar vorzugeben, weil der AC-Lader bei Sollwerten kleiner 0,5 A ohnehin nicht lädt. Dann braucht man den Huawei nur abzuschalten, wenn die Netzleistung einen positiven Schwellwert überschreitet oder für die Dauer einer gewissen Wartezeit keine Anforderung größer 0,5 A erfolgt ist.
>Evtl komme ich zwischen den Jahren mal dazu das Limit auf 50W zu ändern

Das wäre super!

MalteSchm, 2023-12-29:
Ich habe da gerade mal kurz reingesehen. Das Limit für die automatische Leistungsregelung ist erst mal nur eine Einschränkung der UI für den "Automatikmodus". Die UI lässt da nur gewisse Werte als Eingabe zu.

Da man per MQTT auch einen niedrigen Strom einstellen kann habe ich ein wenig mit 1.1 - 1.5 A gespielt (60W - 75W Ausgangsleistung). Da schwankt der Ausgangsstrom aber schon ziemlich. Berücksichtigt man jetzt noch die Verluste machen Werte unterhalb von 75W Eingangsleistung eigentlich keinen Sinn. 

Zum Ein-/Ausschalten ist zu sagen das ich denke das die Sache so bleiben sollte wie es ist. Anderenfalls gäbe es Netzbezug, das ist dann wieder anderen Nutzern nicht recht. Das komplette Ein-/Ausschalten über den Slot Detect Pin erfolgt ja mit einem gewissen Zeitversatz, d.h. da ist eine Hysterese eingebaut. Das Netzteil schaltet also nicht permanent

Lange Rede kurzer Sinn: Ich habe die minimale Leistung im UI auf 50W geändert. Damit das theoretisch geht muss der Schwellwert für den Strom auf 0.75A gesenkt werden. Wird dieser Wert dauerhaft unterschritten (z.B. Batterie voll) schaltet das Netzteil ab. 

Ob das ganze funktioniert müsstest aber du testen. In meinem Setup liege ich da nie in diesem Bereich.

Der Branch ist hier: https://github.com/MalteSchm/OpenDTU-OnBattery/tree/huawei_power_limit_change

Image kommt gleich

dragricola, 2023-12-29:
> Ob das ganze funktioniert müsstest aber du testen. 

Vielen Dank! Der Test folgt morgen - sofern das Wetter mitspielt.

> Das komplette Ein-/Ausschalten über den Slot Detect Pin erfolgt ja mit einem gewissen Zeitversatz, d.h. da ist eine Hysterese eingebaut. Das Netzteil schaltet also nicht permanent

Leider reagiert mein Huawei AC-Charger nicht auf den Slot detect Pin. Er läuft immer, sobald er mit AC-Spannung versorgt wird. Ich schalte deshalb mittels Sonoff DUALR3 ( mit Tasmota), der durch das Slotdetect-Relais gesteuert wird, den AC-Charger AC-seitig ein und aus.

dragricola, 2023-12-30:
[image]
Hier ist der Test von heute mit der Einstellung "Minimum output power: 50W" im AC-Charger-Setup. (Legende siehe oben, zusätzlich neu: Batteriespannung in braun). 
Wie zu sehen, arbeitet die Rückspeisung nun auch im Bereich zwischen 50W und 100W Energieeinspeisung stabil und ohne Probleme. Ich behaupte, dass man den einstellbaren Bereich problemlos noch weiter zu kleineren Leistungen erlauben kann. Mein Vorschlag ist, den Wert für Minimum output power frei wählbar zu machen. Man muss ja diese zusätzliche Freiheit nicht nutzen, sollte es dadurch Probleme geben und ein temporärer Netzbezug ist, wie bereits erwähnt, vor allem im Winter nicht tragisch, solange dadurch der Hoymiles Inverter nicht anspringt. Letzteres kann man aber durch entsprechendes Setzen der Werte im Dynamic Power Limiter Setting verhindern. Ich würde dann je nach Jahreszeit ein anderes Setup für den AC-Charger und den Dynamic Power Limiter wählen. 
Nochmals vielen Dank für das schnelle Eingehen auf meine Wünsche.

dragricola, 2023-12-30:
Ich vergaß, das der Huawei keine Werte kleiner 0,5A einstellen kann. Somit wäre eine Mindestleistung von 0,5A x 54,4V = 27,2 W möglich. Aufgerundet wäre ein Mindestwert von 30W sinnvoll, was ich gerne noch einmal testen würde.

bf4ed, 2023-12-30:
Ja für mich wäre es auch sinnvoll unter 100W einzustellen zu können. Am ende des Tages verschwende ich die letzten Elektronen ins Netz :)
Wo ich eigentlich die 50-99W in die Batterie laden könnte.

MalteSchm, 2023-12-30:
Irgendwann fehlt dann das Kriterium das Gerät über den Slot Detect Pin auszuschalten.
Das Problem ist folgendes: Wenn die Batterie voll ist sinkt der Strom und irgendwann muss die automatische Abschaltung greifen. Diese schaut z.Z. nur auf den Strom und wenn dieser längere Zeit unter 0.75A sinkt wird abgeschaltet. Wenn man mit der Leistung weiter runter geht greift das irgendwann. Da das Netzteil in diesem Bereich nicht mehr verlässlich arbeitet ist meine Sorge das das irgendwelche anderen Probleme nach sich zieht.

dragricola, 2023-12-31:
OK das leuchtet mir ein. Leider hatte ich wegen fehlender Sonne noch nicht das "Problem" die neue Batterie wieder voll zu kriegen. 
Ich denke jedoch, dass dank meiner Konfiguration mit dem Sonoff DUALR3 das Abschalten bei voller Batterie notfalls auch mein ioBroker per Software auslösen könnte, wenn das Abschalten mit dem Slot-Detect Relais nicht funktionieren sollte.
Auf jeden Fall ist das Verhalten bei kleiner PV-Einstrahlung schon jetzt mit einer auf 50W reduzierten Minimum output power viel besser als zuvor. Es wäre schön, wenn diese Änderung auch bald ins nächste offizielle Release einfließen könnte.

MalteSchm, 2023-12-31:
PR is here: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/574

commit d948439b (2023-02-02): Fix #548: The first device profile within the json file was not correctly applied