hoylabs/OpenDTU-OnBattery #576: API /api/livedata/status  returns empty response body with http 200 occasionally

CaCu15, 2023-12-31:
### What happened?

I'm using the http API to implement a charging control in ESP Home currently. I'm trying to use the API **/api/livedata/status** to retrieve some information about the current status of the battery, the charger etc.  Now I've got the problem, that the API occasionally delivers NO result although the http return code is 200 indicating successful completion of the call. A lot of times the call delivers the expected result, but sometimes  it does not return any result. Sometimes 4-5 calls in a row succeed,

### To Reproduce Bug

Execute the following **curl** multiple times in a row. 

` curl -v  http://<ip of the opendtu>/api/livedata/status`

A lot of times the call succeeds as shown in the second code fragment below, but in 10-20% of the cases the call does not return any result:

**Unsuccessful call:**
```
 curl -v  http://192.168.3.17/api/livedata/status
*   Trying 192.168.3.17:80...
* Connected to 192.168.3.17 (192.168.3.17) port 80 (#0)
> GET /api/livedata/status HTTP/1.1
> Host: 192.168.3.17
> User-Agent: curl/7.88.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 4
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
<
* Closing connection 0
```

**Successful call:**
```
 curl -v  http://192.168.3.17/api/livedata/status
*   Trying 192.168.3.17:80...
* Connected to 192.168.3.17 (192.168.3.17) port 80 (#0)
> GET /api/livedata/status HTTP/1.1
> Host: 192.168.3.17
> User-Agent: curl/7.88.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 1586
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
<
{"inverters":[{"serial":"114183731764","name":"HM600","order":0,"data_age":1,"poll_enabled":true,"reachable":true,"producing":false,"limit_relative":5,"limit_absolute":30,"AC":{"0":{"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":233.5,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"Power DC":{"v":2.799999952,"u":"W","d":1},"YieldDay":{"v":465,"u":"Wh","d":0},"YieldTotal":{"v":351.7850037,"u":"kWh","d":3},"Frequency":{"v":49.97999954,"u":"Hz","d":2},"PowerFactor":{"v":0,"u":"","d":3},"ReactivePower":{"v":0,"u":"var","d":1},"Efficiency":{"v":0,"u":"%","d":3}}},"DC":{"0":{"name":{"u":""},"Power":{"v":1.5,"u":"W","d":1},"Voltage":{"v":48.5,"u":"V","d":1},"Current":{"v":0.029999999,"u":"A","d":2},"YieldDay":{"v":238,"u":"Wh","d":0},"YieldTotal":{"v":169.4920044,"u":"kWh","d":3}},"1":{"name":{"u":""},"Power":{"v":1.299999952,"u":"W","d":1},"Voltage":{"v":48.5,"u":"V","d":1},"Current":{"v":0.029999999,"u":"A","d":2},"YieldDay":{"v":227,"u":"Wh","d":0},"YieldTotal":{"v":182.2929993,"u":"kWh","d":3}}},"INV":{"0":{"Tempe* Closing connection 0
rature":{"v":18,"u":"°C","d":1}}},"events":10}],"total":{"Power":{"v":0,"u":"W","d":1},"YieldDay":{"v":465,"u":"Wh","d":0},"YieldTotal":{"v":351.7850037,"u":"kWh","d":3}},"hints":{"time_sync":false,"radio_problem":false,"default_password":false},"vedirect":{"enabled":false,"total":{"Power":{"v":0,"u":"W","d":1},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":2}}},"huawei":{"enabled":true,"Power":{"v":0,"u":"W","d":2}},"battery":{"enabled":true,"soc":{"v":20,"u":"%","d":0}},"power_meter":{"enabled":true,"Power":{"v":578,"u":"W","d":1}}}
```

### Expected Behavior

API call should return result (a JSON) if http code is 200 - or return an http code in the range > 400 or >500 if an error occured during processing in OPenDTU.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2023.12.27.post1

schlimmchen, 2024-01-04:
@helgeerbe Is this a symptom of #591?

helgeerbe, 2024-01-04:
Yes, I think so. The zero size JsonDocument is send over the websocket and should result in `HTTP/1.1 200 OK`.

With release 2024.01.04 I added a log entry on the console output. No help at the moment, but you will know why.
I don't know, if you use mqtt. It might help a little bit if you extend the period from 5 to 15 seconds and disable TLS encryption, if you have enabled it (this consumes a lot of memory).

Last trick would be to compile the code with the maximum of interveners set to 2.

But before you take action, check the console output if you find an entry there.

schlimmchen, 2024-04-12:
I assume that this is no longer an issue. Thomas refactored the API such that this API endpoint merely lists the available inverters. Their data is now pushed through the websocket and chunked. There should now always be enough memory (and sufficiently large buffers configured) to return an answer at this endpoint.

Please reopen this issue if the issue persists in the last release.

closed 2024-04-12 by schlimmchen

CaCu15, 2024-04-16:
Right, never experienced this problem again. Thx for fixing this!