hoylabs/OpenDTU-OnBattery #580: [Request] avoid sending of unnecessary power limits

Fribur, 2023-12-31:
### Is your feature request related to a problem? Please describe.

Before elaborating here in GitHub: do you have a Discord for OpenDTU onbattery? Or do you use the ahoy/OpenDTU discord (which handle?) Chat there might be more efficient.

### Describe the solution you'd like

I noticed a new powerlimit is sent no matter what every 60 seconds "to avoid staleness"? That causes inverter to do +-1W changes in powerlimit every 60 seconds (effectively ignoring hysteresis). Not sure if this is healthy for the inverter, and certainly not needed to change a powerlimit from "270W" to "271W".

```
// (re-)send power limit in case the last was sent a long time ago. avoids
// staleness in case a power limit update was not received by the inverter.
 auto ageMillis = millis() - _lastPowerLimitMillis;
if (diff < hysteresis && ageMillis < 60 * 1000) {
...}
```
Is it not possible to check the current _actually configured/accepted_ power limit per inverter like the following, and send a new limit _only if really needed_?:

```
int currentLimit = round(inverter->SystemConfigPara()->getLimitPercent() * inverter->DevInfo()->getMaxPower() / 100);
auto currentDiff = std::abs(effPowerLimit - currentLimit );
if (diff < hysteresis && currentDiff < hysteresis ){
   return;
}
```
It looks like SystemConfigPara is updated anyhow every 120 seconds via the HoyMiles main loop..so this value should be trustworthy, hence "staleness" of the limit should be no issue?:
```
// Fetch limit
if (((millis() - iv->SystemConfigPara()->getLastUpdateRequest() > HOY_SYSTEM_CONFIG_PARA_POLL_INTERVAL)
		&& (millis() - iv->SystemConfigPara()->getLastUpdateCommand() > HOY_SYSTEM_CONFIG_PARA_POLL_MIN_DURATION))) {
	_messageOutput->println("Request SystemConfigPara");
	iv->sendSystemConfigParaRequest();
}
```

schlimmchen, 2024-01-04:
See 116da051146df9f5714cb18a2895703d426fb3d0.

> Not sure if this is healthy for the inverter,

If regularly setting new power limits is unhealthy, the whole project is in big trouble :speak_no_evil:

Fribur, 2024-01-04:
Not sure I follow the reasoning “ _we then
would have to decide how much deviation is okay, which is unreasonably
complicated._” It is literally 1 line of code: `if (diff < hysteresis && currentDiff < hysteresis )`

Basically just establishes if the new calculated limit and the possibility 120second old limit as reported by the inverter are both within the hysteresis tolerance. If either one of them is not, a new limit is issued. What is the point of the hysteresis  when the inverter is anyway adjusted 1W up or down every 60 seconds? Might as well just remove the hysteresis option and just force updates every 20 seconds or so. 

I will most definitely keep this change in fork, unless I lern of a scenario where this could lead to trouble.

schlimmchen, 2024-01-05:
The user-configurable hysteresis is used to suppress limit updates if they are close to the last limit that was set and we assume that the last limit was indeed set.

You are asking to re-use the same setting for something different, which is a mechanism that tries to determine whether or not the expected limit was indeed received by the inverter.

I don't think a user-setting should be used to control different things. Assume a user sets the hysteresis to 0 W or even 1 W or 2 W. Depending on the inverter output power (higher absolute values cause higher relative deviations), a new limit will be sent every time due to rounding errors (remember that we set an absolute value while the inverter answers with a relative value).

>> we then would have to decide how much deviation is okay, which is unreasonably complicated

> It is literally 1 line of code

I disagree. The user-configurable hysteresis is not suited here. So tell us: Where do we draw the line to decide whether or not the last limit was actually received by the inverter? You set an absolute limit of x, the inverter applies it and reports a relative limit y. We can derive an absolute value: inverter max power * y = z. Now how much may z deviate from x for us to assume that the limit x was set and resetting x will have no actual effect?

> I will most definitely keep this change in my fork, unless I lern of a scenario where this could lead to trouble.

Did you read #478 (which is linked in the commit message)?

Fribur, 2024-01-05:
> Was verursacht das der Limiter bei 9%, nachdem der HM300 12h aus war, fest hängt. (https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/478)

This cannot happen with my proposed change, because ` if (currentDiff < hysteresis )` will make DPL realize that the inverter has an inappropriate PowerLimit, and will consequently issue a new PowerCommand. 

Really sorry for not understanding your concern, you might be right and I am simply not getting it.  :-(

If you do not mind, lets dissect this a bit, and please comment where my thinking is off. I am renaming the variables a bit to make the point clearer: 

- current code that checks if a new PowerCommand needs to be issued:

```
auto assumedDifference = std::abs(effPowerLimit - _lastRequestedPowerLimit);
auto hysteresis = config.PowerLimiter.TargetPowerConsumptionHysteresis;
if (assumedDifference < hysteresis)
```
The issue with this code is that DPW has not idea if `_lastRequestedPowerLimit` was swallowed by the inverter. So we can do 2 things (1) re-issue a current powercommand every 60 seconds (currently done) or (2) verify if _lastRequestedPowerLimit was accepted by getting the current limit from the inverter

- as for option 1: current code that checks if the last PowerCommand is outdated:

```
auto ageMillis = millis() - _lastPowerLimitMillis;
 if (ageMillis < 60 * 1000)
```

- as for option 2: my proposal of an _additional_ check a new PowerCommand needs to be issued:
```
auto currentLimit = round(inverter->SystemConfigPara()->getLimitPercent() * maxPower / 100);
auto actualDifference = std::abs(effPowerLimit - currentLimit );
auto hysteresis = config.PowerLimiter.TargetPowerConsumptionHysteresis;
if (actualDifference < hysteresis)
```

So I am not seeing how I am using hysteresis for something different compared to what it already used for:  check if the current calculated powerlimit AND the powerlimit as reported by inverter are within a threshold to restrict how often the update command is sent. 

If the user sets a too small value, the DPW limiter will spam the inverter no matter what with new Powerlimits, because `ageMillis < 60 * 1000` will not prevent that.

schlimmchen, 2024-03-19:
I have not re-read your last post, please excuse my ignorance.

I am going to close this now based on 8b6e57cda75c2e5577cd2a197afdf3a84b855e3b which is part of release 2024.03.07. The linked change will test the calculated limit against the limit reported by the inverter. Only if these values deviate by more than the hysteresis, a limit update is sent. There is no interval-based forced re-submission of a limit.

As far as I can tell, that addresses your concern fully. Please come back if you suspect that something is missing.

closed 2024-03-19 by schlimmchen