hoylabs/OpenDTU-OnBattery #585: http based Power Meter: mDNS bug prevents resolving of local host names

Fribur, 2024-01-02:
### What happened?

Configuring an http based Shelly EM Power Meter leads to the following symptoms:

1. http://shellyem-485519d72dd4/emeter/0 -->error in WiFiGeneric.cpp line 1582 "DNS Failed for.."
2. http://shellyem-485519d72dd4.local/emeter/0 -->same error as under (1). In addition crash of OpenDTU (something with watchdog) when using "Run test" function
3. http://192.168.1.94/emeter/0 -->works

### To Reproduce Bug

see above. 

Bug can be resolved by replacing in HttpPowerMeter.cpp the code in HttpPowerMeterClass::httpRequest before `prepareRequest(timeout, httpHeader, httpValue);` by the following code:
```
    String urlProtocol;
    String urlHostname;
    String urlUri;
    u_int16_t port;
    extractUrlComponents(url, urlProtocol, urlHostname, urlUri);
    //MessageOutput.printf("url %s, urlProtocol %s, urlHostname %s, urlUri: %s\r\n", url, urlProtocol.c_str(), urlHostname.c_str(), urlUri.c_str());
    response[0] = '\0';
    error[0] = '\0';

    // secureWifiClient MUST be created before HTTPClient
    // see discussion: https://github.com/helgeerbe/OpenDTU-OnBattery/issues/381
    std::unique_ptr<WiFiClient> wifiClient;    
    if (urlProtocol == "https") {
      port= 443;
      auto secureWifiClient = std::make_unique<WiFiClientSecure>();
      secureWifiClient->setInsecure();
      wifiClient = std::move(secureWifiClient);
    } else {
      port= 80;
      wifiClient = std::make_unique<WiFiClient>();
    }

    //hostByName in WiFiGeneric fails to resolve local names. issue described in 
    //https://github.com/espressif/arduino-esp32/issues/3822
    //and in depth analyzed in https://github.com/espressif/esp-idf/issues/2507#issuecomment-761836300
    //in conclusion: we cannot rely on httpClient.begin(*wifiClient, url) to resolve IP adresses.
    //have to do it manually here. Feels Hacky...
    IPAddress ipaddr((uint32_t)0);
    bool httpResult;
    //first check if the urlHostname is already an IP adress
    if (!ipaddr.fromString(urlHostname))
    {
        //no it is not, so try to resolve the IP adress
        mdns_init ();
        ipaddr = MDNS.queryHost(urlHostname);
        if (ipaddr == INADDR_NONE)
        {
            httpResult=false;
            MessageOutput.printf("Error resolving IP: %s\r\n", ipaddr.toString().c_str());
        }
        else
        {
            urlHostname = ipaddr.toString();
            //MessageOutput.printf("resolved IP %s\r\n", ipaddr.toString().c_str());
            httpResult  = httpClient.begin(*wifiClient, urlHostname, port, urlUri);
        }
    }
    else
    {
        //urlHostname is already an IP adress, so just create wifiClient and httpClient
        httpResult  = httpClient.begin(*wifiClient, urlHostname, port, urlUri);
    }
       
    if (!httpResult){        
    //if (!httpClient.begin(*wifiClient, url)) {
      snprintf_P(error, errorSize, "httpClient.begin(%s) failed", url);
      return false;
    }
```

The cause is that hostByName in WiFiGeneric fails to resolve local names. Issue is described in 
 https://github.com/espressif/arduino-esp32/issues/3822 and has been analyzed in depth in https://github.com/espressif/esp-idf/issues/2507#issuecomment-761836300. This solution here feels hacky, and I hope someone with better knowledge how to configure WIFI and HTTP clients in ESP32 can come up with a better way to do it. 

### Expected Behavior

Expect that PowerMeter can be configured as IP or name, local or remote.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

08bc181

Fribur, 2024-01-04:
Turns out mDNS can be turned on for the Network interface under "Network Settings". Who would have thought. the current HttpPowerMeter can still not deal with it (I am curious if it works for anybody else?)...so I ended up converting the above into a proper solution, and while doing so significantly refactoring the code. E.g. removing a redundant JSON library makes the build 20kB smaller, risk of heap fragmentation should be smaller due to snprintf to build strings instead of "stringA+stingB". I hope you find this an improvement and are open to including it in the project. I can create a pull request if preferred. 
[image]
[HttpPowerMeter.cpp.txt](https://github.com/helgeerbe/OpenDTU-OnBattery/files/13835363/HttpPowerMeter.cpp.txt)
[HttpPowerMeter.h.txt](https://github.com/helgeerbe/OpenDTU-OnBattery/files/13835364/HttpPowerMeter.h.txt)

schlimmchen, 2024-01-04:
> E.g. removing a redundant JSON library makes the build 20kB smaller

Wow, that sounds interesting! However, I would expect changes to files other than `HttpPowerMeter.*` to remove the lib?

Please create a Pull Request, so your changes can be evaluated.

Fribur, 2024-01-04:
Done https://github.com/helgeerbe/OpenDTU-OnBattery/pull/594

Fribur, 2024-01-18:
Resolved

closed 2024-01-18 by Fribur