hoylabs/OpenDTU-OnBattery #598: Alternative adjustable AC charger with MeanWell HLG-600H-54AB

Linos88, 2024-01-06:
### Is your feature request related to a problem? Please describe.

Sorry, I didn't manage to create a request... 

### Describe the solution you'd like

 First of all, I would like to thank all the developers for this project and the time they have invested. It fits perfectly into our time.

About the proposal: 
A popular power supply manufacturer offers an adjustable AC / DC power supply that can be used to charge batteries. This is available with different voltages and power. The identifier "AB" stands for I and V adjustable with potentiometer and 3in1 dimming function.  

As an example for 48V LiFePO4 the [MeanWell HLG-600H-54AB](https://www.meanwell.com/Upload/PDF/HLG-600H/HLG-600H-SPEC.PDF) with an adjustable voltage of 45.9V - 56.7V and an adjustable power of 0W - 600W with a control voltage of 0-10V.  

The control voltage could be generated by the [Shelly Plus 0-10V Dimmer](https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus10V/) and receive its commands directly from OpenDTU_onBAT. 

or alternatively 

ESP32 with PWM 0-3.3V and a [PWM digital-to-analog converter to 0-10V] (https://www.google.com/aclk?sa=L&ai=DChcSEwjOxYeUysmDAxWqQUECHWBwB1UYABAMGgJ3cw&gclid=EAIaIQobChMIzsWHlMrJgwMVqkFBAh1gcAdVEAQYAiABEgLaN_D_BwE&sig=AOD64_3rowaNSo4fD_dQDGcqnm8jA_GEgA&ctype=5&q=&ved=2ahUKEwj1zYCUysmDAxX9YPEDkWB1wQ9aACKAB6BAgBECI&adurl=).

DaBaBW, 2024-04-21:
I like to use a HLG, too.
For the adjustment it would be easier to simply PWM an optocoupler on the DIM port of the HLG.
No need of 0-10V

fellpower, 2024-06-01:
i have all that components - and here for testing, if the code is ready.