hoylabs/OpenDTU-OnBattery #601: Using HM WR in parallel

gitisgreat2023, 2024-01-10:
### Is your feature request related to a problem? Please describe.

With the Huawei the highest power Hoymiles that can be combined is the HM-1500. The HMT/HMS cannot be used (for example the HMT-2250), as the Huawei and CMT2300A chip cannot be combined (SPI bus limitation) with the ESP32 chip.

A solution would be to use several HMs in parallel, for example a HM 1500 and a HM800.

### Describe the solution you'd like

Using two HM in parallel. Power limiting can be proportional. For example 50% of the combined wattage ist need: both HM inverters are limited to 50%.

gitisgreat2023, 2024-01-11:
@tbnobody Can dynamic power limiting on two HMs in parallel be done on OpenDTU-OnBattery level itself, or would a supporting feature at the OpenDTU level be very helpfull?

gitisgreat2023, 2024-01-12:
I have only experience with Matlab, not C++. By looking at the code I try to see how complicated it might be.
Looking at [PowerLimiter.cpp](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/src/PowerLimiter.cpp ) I would say for two powerlimiters in parallel, at least three modifications would be needed:
1) adding a second task, if a second HM with power limiter is configured
2) modifying setNewPowerLimit. The total HM power with power limitation activated has to be calculated and then proportionally (a choice) for the array of powerlimiters active the limit can be set.
3) creating an array of powerlimiters [Configuration.cpp](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/src/Configuration.cpp)

spcqike, 2024-01-12:
@gitisgreat2023 you would need some more modifications. like, what setup does the user have? 
- solar only? are the modules of one inverter shaded? so a simple split of the load between both won't work
- one battery for both inverters? great, not much to add (probably)
- two batteries? well, thats going to be more complicated than

what about solar passthrough? 

yes, to limit 2 inverters in one specific usecase shouldn't be to complicated, but openDTU-onBattery has multiple usecases and setups.

another option for handling multiple inverters would be to start with one inverter, increase the load (output) until it maxes out and use the second inverter only, if the first is at max output. as we still want low switching/limiting times and fast responses. afaik the update of multiple inverters takes longer and the time increases with each inverter added to the setup. so if you utilze both (or more) at the same time, you increase the update times and get a not so fast response anymore. so right know, with one inverter, you can keep your production and consumption close together, if you add more inverters and try to limit them at the same time, your system gets slower.
this was mentioned here https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/246#discussioncomment-6063893

gitisgreat2023, 2024-01-12:
I would use one battery with two inverters, the simplest case IMHO. I would put one on each phase (there's a 400V that I can use).
Yes the second, step up/down would also be one I considered. But for the code change the "proportional" would be the easiest to implement I think. 
On you response time: how much time does it cost to query the power meter, typically? If that is relatively fast, and the HM power limiting is relatively slow, then the performance hit wouldn't be that large if one queries the power meter a second time before power limiting the second inverter and adjust it accordingly.

Thanks for the (other thread)[https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/246#discussioncomment-6063893]. 
@helgeerbe google translate says you considered it to do it in autumn/winter... I also read you concern on people adding 5 HM-1500 to a single wallet.
Would it be an idea to enable to use a second HM to be power regulated? That would cover a lot of cases I would say. And the proportional load, that would be easiest to implement, looking at the code. I would be happy to test it then, of course. And I assume @neolauren-cmd would be also happy to test the 2 WR power limiter...

gitisgreat2023, 2024-01-26:
This issue can be closed.

See [this post ](https://github.com/tbnobody/OpenDTU/pull/1200#issuecomment-1906864693) which progressed much further... a software solution would be best, enabling for example combining the HMT2250 with the Huawei (2.25kW vs 1.5kW of the HM1500).

stefan123t, 2024-02-09:
I have suggested a list of PV-Pools to which you can assign your PV-Inverters & PV-Panels.
That way you can specify a limit for each Pool of PV-Inverters / Panels and control a common limit per pool.

See here for the same discussion upstream https://github.com/tbnobody/OpenDTU/discussions/1608#discussioncomment-7988430

schlimmchen, 2024-03-18:
Support for multiple inverters in the dynamic power limiter is a long standing issue, see #230. It will probably be addressed eventually. There is no need for another issue.

closed 2024-03-18 by schlimmchen