hoylabs/OpenDTU-OnBattery #602: Daily yield on some Converters 0 at night

BytePhilHD, 2024-01-10:
### What happened?

When the connection to converters fails (for example at night), some daily yield values reset to 0, even if they were higher earlier. The total yield values are there during the night. 
I am at the latest version!

### To Reproduce Bug

I have 5 active converters (by the way: Why isnt it possible to add more?), the problem occurs when loosing longer connection to them, for example at night. 

### Expected Behavior

Would it be possible to cache the latest received values of the converter and keep it displayed during night? So if the last received value is 2.000 Wh keep this value displayed during the night instead of resetting it to 0. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	2024.01.7.post1

Manos1966, 2024-01-10:
> I have 5 active converters (by the way: Why isnt it possible to add more?)

It is because of the Memory available on the current ESP32 models.
With all additional functions the OpenDTUonBattery performs, having more Inverters would overfill the memory completely and cause problems.

BytePhilHD, 2024-01-11:
In my case I have one 2250-6T and four 2000-4T, but only the one 2250-6T keeps his last value. The others reset to 0. I am not sure why it is like this, but its very bad. 
Is this a opendtu error or is it just with this type of invertes?

Manos1966, 2024-01-11:
"one 2250-6T and four 2000-4T"    Jesus... 🙈 🤣 

Also wie @spcqike bereits gesagt hat, "Made by design" 😄 
Das war die Logik von Hoymiles: Nachts bekommt ein Wechselrichter kein Strom, geht er aus. Anlass zu erkennen, dass der TagesErtrag erreicht wurde.

Diese Logik hat das Problem verursacht, angeschlossen an einer Batterie, hat der Wechselrichter staendig Strom, dadurch wird nie das "Ende des Tages" erreicht, der TagesErtrag laeuft ewig...
Man hat das Problem geloescht indem man ein Neustart des Wechselrichters programmieren kann (unter Dynamic Power Limiter).

Also, 
1. kann es sein, dass deine 2000-4T programmiert sind, neu zu starten?
2. Ich habe mein (gleiche) Problem geloescht, ich habe eine Dummy Parameter in meinem Homatic System definiert, die regelmaessig den Tagesertrag speichert. So kann ich erkennen falls der Wechselrichter einen Neustart gemacht hat.
Zusaetzlich, nutze ich den Dummy Wert von Mitternacht fuer andere Berechnungen.
3, Ich habe nur HM-Serie Wechselrichter. Keiner zeig das Problem, das du beschrieben hast
Anbei Bild meiner OpenDTU um 20:50
Keine Daten empfangen seit 16:47
Tagesertrag bleibt trotzdem da.
Die Frage ist also, haben die 2000-4T eine andere Logik (woran ich nicht glaube), oder passiert etwas waehrend des Tages und sie starten Neu?
[image]

BytePhilHD, 2024-01-11:
Also ich habe grad nochmal nachgeschaut, alle hatten die selbe Logik: Dieses Tageskorrektur war aus. Hab gar nicht gesehen, dass es die Funktion extra gibt und man die überall aktivieren muss :/. 
Hab jetzt mal bei allen die Funktion aktiviert und werd morgen Abend schauen obs geht, wahrscheinlich aber schon. 
Also der Issue hier war wahrscheinlich unnötig, weil ich einfach die Einstellung nicht gesehen hab, sorry :(

Manos1966, 2024-01-11:
Hoffen wir, dass das die Lösung war...

BytePhilHD, 2024-01-12:
Ok also die Einstellung mit dem Fix Daily Value war tatsächlich die Lösung. 
Jetzt funktioniert alles einwandfrei! Danke euch und sorry für den unnötigen Issue.

Einzige coole wäre noch wenn man einen 6. Converter hinzufügen kann (ja hab tatsächlich 6 :D), der Speicher hat laut System auch noch minimalst Luft, aber ist mir bewusst, dass das gefährlich werden kann. 
[image]

closed 2024-01-12 by BytePhilHD

schlimmchen, 2024-01-12:
Du darfst es gerne ausprobieren. Eine Firmware selbst bauen ist nicht kompliziert. Falls du es versuchen möchtest, tue es. Wenn es klappt sage ich dir, welche Datei du anpassen musst. Es ist nur eine Ziffer, die geändert werden muss...

Manos1966, 2024-01-12:
> Eine Firmware selbst bauen ist nicht kompliziert. 

Kannst du uns ein paar Anleitungen posten (oder direkt in die Wiki schreiben), wie wir das machen können, Bernhard? 
Ich habe VisualStudioCode & GitBash installiert aber danach war ich hoffnungslos verloren... 😭

schlimmchen, 2024-01-12:
Ne, sorry, es braucht nicht noch eine Kopie solch einer Anleitung. Siehe https://tbnobody.github.io/OpenDTU-docs/firmware/compile_vscode/ da sind sogar zwei Videos verlinkt.

Manos1966, 2024-01-12:
> Siehe https://tbnobody.github.io/OpenDTU-docs/firmware/compile_vscode/ da sind sogar zwei Videos verlinkt.

Erstellt: December 28, 2023 23:58:48 
Ich habe Jahre von meinem Leben verloren letzten Sommer.... 🤣 🤣 🤣

BytePhilHD, 2024-01-12:
> Du darfst es gerne ausprobieren. Eine Firmware selbst bauen ist nicht kompliziert. Falls du es versuchen möchtest, tue es. Wenn es klappt sage ich dir, welche Datei du anpassen musst. Es ist nur eine Ziffer, die geändert werden muss...

Könntest du mir evtl sagen wie die Klasse heißt in der ihr die Anfragen sendet bzw. wo man das ändern müsste?

schlimmchen, 2024-01-12:
> Erstellt: December 28, 2023 23:58:48

@Manos1966 Diese Infos gibt es schon seit rund zwei Jahren im Abschnitt "Flashing and starting up", siehe https://github.com/tbnobody/OpenDTU/blob/1d29930dc5c47bb568e3c7367ce19deb2fc725ea/README.md

Die sind lediglich Ende letzten Jahrens aus der README.md in das neue OpenDTU-docs repo gewandert.

> Könntest du mir evtl sagen wie die Klasse heißt in der ihr die Anfragen sendet bzw. wo man das ändern müsste?

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/a012d81427a782aeab848c237963381e400b9410/include/Configuration.h#L26

@BytePhilHD Ändere `INV_MAX_COUNT` auf 6. (Ich habs nicht getestet, aber es müsste mit der Anpassung auf "6" getan sein).

BytePhilHD, 2024-01-16:
@schlimmchen Es scheint funktioniert zu haben mit dem weiteren Konverter. Also konnte ihn noch nicht testen, aber bereits einen weiteren hinzufügen, danke also schonmal im vorraus.
Eine Frage hätte ich noch: Wenn ich etwas am Design änder im Projekt (habs in VS Code) und das dann baue und die Firmware.bin hochlade, warum wird das neue Design nicht übernommen? Also ist in der Firmware.bin rein das Backend oder wie?

schlimmchen, 2024-01-16:
Sehr gut!

Design? Meinst du die WebApp? Die musst du mit nodejs bauen (Unterordner webapp) und dann die Firmware neu kompilieren.

BytePhilHD, 2024-01-16:
Ja, ich meine die WebApp. Aber wo genau soll ich die mit NodeJS bauen? Geht das irgendwo bei Platform IO? Also ich habe unterandem die Datei "webapp/src/components/NavBar.vue" verändert und dann bei Project tasks unter generic_esp32 -> General -> Build gedrückt (die Backend Änderung ging auch aber eben WebApp nicht). Wo genau muss ich da also was bauen?

BytePhilHD, 2024-01-16:
Ah ok, aber danke dir trotzdem :)