hoylabs/OpenDTU-OnBattery #605: [Request] configurable HW-SPI (for Huawei CAN and CMT wireless module)

mager33, 2024-01-13:
### Is your feature request related to a problem? Please describe.

Both use the same HW SPI

### Describe the solution you'd like

Make HW-SPI configurable for each device.

### Describe alternatives you've considered

Hard-assign HW-SPI 2 to CMT and HW-SPI 1 to Huawei and disable other modules the require SPI, like display

gitisgreat2023, 2024-01-22:
Its an abstract request.
You're interested that the Huawei can be used together with HMT/HMS Hoymiles, right? (those need the CMT module)

Currently only the HM series can be combined with the Huawei.

mager33, 2024-01-22:
Yes, that was my idea. I don't see a reason why this could not be possible.

 

As an alternative I'd suggest to use the build-in CAN-controller, as is done here: https://github.com/mb-software/esphome-huawei-r4850 - that would free up the SPI for other parts. Using the build-in CAN also means that a cheaper sn65hvd230 module must be used.

 

Gesendet: Montag, 22. Januar 2024 um 10:30 Uhr

gitisgreat2023, 2024-01-23:
@mager33 yeah, but in those applications the built in CAN controller of the ESP32 is used for the Pylontech communication.

mager33, 2024-01-23:
The Pylontech has CAN and RS485, so we could free the CAN-bus. 
And if the library can use both CAN methods we could add a configuration option: (a) CAN via SPI and external controller and (b) CAN via built-in controller

gitisgreat2023, 2024-01-24:
Which interface is then used to steer the RS485? Some general IO pins of the ESP32?

mager33, 2024-01-24:
Yes, RS485 is standard serial via an external transceiver (MAX485 or similar, available on cheap modules). There are 3 hardware UARTs available: https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/

gitisgreat2023, 2024-01-26:
Relevant discussion [here](https://github.com/tbnobody/OpenDTU/pull/1200#issuecomment-1906864693) on adding an ethernet controller via CAN...

mager33, 2024-01-26:
That looks like a well thought out request, only the title is misleading. We can thus close my request.

gitisgreat2023, 2024-01-26:
Fully agree.

krissfr, 2024-02-29:
Pylontech can use UART protocol like the JKBMS, and i don't think people will use JKBMS and Pylontech a the same time (and if needed, esp32 has 3 UART controllers).
Pylontech using uart is very used with esp devices :
https://esphome.io/components/pylontech.html

So if Pylontech use the UART communication, the Huawei charger can use the CAN bus of the esp, and people can use other SPI for NRF and CMT tranceiver.

AndreasBoehm, 2024-08-01:
I create an pull request to make huawei usable together with HMS: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1144