hoylabs/OpenDTU-OnBattery #610: Dynamic Power Limiter stoppt den Wechselrichter nicht bis die Akku SoC - Start Werte erreicht werden

Manos1966, 2024-01-16:
### What happened?

Ich verstehe es nicht...

Meine Pylontech ist bei 30% SOC und ich will sie heute bis zu 95% laden.
Daher:
- Akku SOC - Start: 90%
- DC Spannung - Start: 52V (Vorsichtshalber)
- Strategie: "Leeren, wenn voll" oder "Leeren zur Nacht" (kein Unterschied)
- Aktuelste Firmware (2024.001.07)
- Neustart der OpenDTUonBattery

Mit diesen Eistellungen, sollte der Wechselrichter aufhoeren zu produzieren und warten, bis die Akku SOC die 90% erreicht, bevor er wieder anfaengt.

**Dies ist leider NICHT der Fall. Der Wechselrichter fängt sofort zu produzieren** 
Neustart der OpenDTU, Neustart des Wechselrichters, Firmware Update : Nichts hat geholfen.

Mache ich etwas falsch, oder ist es ein Bug?

Bilder anbei:
BatterieStand
[image]
PowerLimiter Settings
[image]
Wechselrichter
[image]
Victron
[image]

### To Reproduce Bug

Just normal operation using Dynamic Power Limiter and setting a high Battery Start SOC % limit

### Expected Behavior

"Akku SoC - Start %" muss erst erreicht werden bevor der Wechselrichter starten soll Strom zu produzieren.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.01.7.post1

### Relevant log/trace output

```Shell
RX Period End
12:27:18.726 > Last missing
12:27:18.831 > Request retransmit: 3
12:27:18.962 > TX RequestFrame Channel: 61 --> 15 74 40 20 87 80 16 00 72 83 E1 
12:27:19.032 > RX Period End
12:27:19.154 > Last missing
12:27:19.212 > Request retransmit: 3
12:27:19.295 > TX RequestFrame Channel: 75 --> 15 74 40 20 87 80 16 00 72 83 E1 
12:27:19.384 > [    562.716] DPL: waiting for sufficiently recent inverter data
12:27:19.443 > [Pylontech] Alarms: 0 0 0 0 0 0 012:27:19.572 > [Pylontech] Warnings: 0 0 0 0 0 0 012:27:19.645 > [Pylontech] chargeVoltage: 53.200001 chargeCurrentLimitation: 80.000000 dischargeCurrentLimitation: 80.00000012:27:19.718 > Interrupt received
12:27:19.863 > [Pylontech] soc: 32 soh: 10012:27:19.924 > RX Channel: 40 --> 95 74 40 20 87 74 40 20 87 81 00 01 70 C0 A5 | -80 dBm
12:27:19.982 > [Pylontech] voltage: 50.419998 current: 16.900000 temperature: 10.90000112:27:20.204 > [Pylontech] chargeStatusBits: 1 1 012:27:20.316 > RX Period End
12:27:20.410 > Success
12:27:20.588 > [Pylontech] Alarms: 0 0 0 0 0 0 012:27:20.669 > [Pylontech] Warnings: 0 0 0 0 0 0 012:27:20.878 > [Pylontech] chargeVoltage: 53.200001 chargeCurrentLimitation: 80.000000 dischargeCurrentLimitation: 80.00000012:27:20.958 > Fetch inverter: 116174402087
12:27:21.044 > [Pylontech] soc: 32 soh: 10012:27:21.135 > TX RealTimeRunData Channel: 3 --> 15 74 40 20 87 80 16 00 72 80 0B 00 65 A6 68 18 00 00 00 00 00 00 00 00 98 19 DB 
12:27:21.216 > [Pylontech] voltage: 50.419998 current: 16.800001 temperature: 10.90000112:27:21.272 > [Pylontech] chargeStatusBits: 1 1 012:27:21.347 > [Pylontech] Manufacturer: PYLON   12:27:21.405 > Interrupt received
12:27:21.487 > RX Channel: 75 --> 95 74 40 20 87 74 40 20 87 01 00 01 01 FA 00 3D 00 3D 01 33 01 36 00 02 E4 1B 96 | -80 dBm
12:27:21.593 > Interrupt received
12:27:21.680 > RX Channel: 23 --> 95 74 40 20 87 74 40 20 87 84 13 8A 04 4C 00 D9 00 2F 03 D5 00 4C 0E 37 50 61 A4 | -80 dBm
12:27:21.756 > RX Period End
12:27:21.871 > Middle missing
12:27:21.988 > Request retransmit: 2
[... 237 lines cut ...]
12:27:51.499 > [DPL::loop] Consuming Solar Power Only -> adjustedVictronChargePower: 922 W, newPowerLimit: 134 W
12:27:51.572 > [DPL::setNewPowerLimit] requested: 134 W, last limit: 149 W, diff: 15 W, hysteresis: 25 W, age: 13251 ms
12:27:51.639 > [DPL::loop] ******************* Leaving PL, calculated limit: 134 W, requested limit: 149 W (kept last requested)
12:27:51.730 > RX Period End
12:27:51.809 > Middle missing
12:27:51.904 > Request retransmit: 2
12:27:52.126 > TX RequestFrame Channel: 3 --> 15 74 40 20 87 80 16 00 72 82 E0 
12:27:52.217 > Interrupt received
12:27:52.417 > RX Channel: 61 --> 95 74 40 20 87 74 40 20 87 02 00 02 F4 22 00 01 00 01 01 FA 00 41 00 41 01 48 F1 | -80 dBm
12:27:52.483 > RX Period End
12:27:52.540 > Success
12:27:52.613 > TX AlarmData Channel: 23 --> 15 74 40 20 87 80 16 00 72 80 11 00 65 A6 68 28 00 00 00 00 00 00 00 00 43 56 65 
12:27:52.675 > RX Period End
12:27:52.834 > All missing
12:27:52.891 > Nothing received, resend whole request
12:27:52.969 > TX AlarmData Channel: 40 --> 15 74 40 20 87 80 16 00 72 80 11 00 65 A6 68 28 00 00 00 00 00 00 00 00 43 56 65 
12:27:53.143 > Interrupt received
12:27:53.269 > RX Channel: 3 --> 95 74 40 20 87 74 40 20 87 01 00 01 80 01 0E 19 A0 56 A0 56 00 00 00 00 40 7C 3F | -80 dBm
12:27:53.325 > [DPL::loop] ******************* ENTER **********************
12:27:53.399 > [DPL::loop] battery interface enabled, SoC: 32 %, StartTH: 90 %, StopTH: 12 %, SoC age: 1 s
12:27:53.559 > [DPL::loop] dcVoltage: 50.60 V, loadCorrectedVoltage: 50.77 V, StartTH: 52.00 V, StopTH: 47.00 V
12:27:53.637 > [DPL::loop] StartTH reached: no, StopTH reached: no, inverter is producing
12:27:53.695 > [DPL::loop] SolarPT enabled, Drain Strategy: 1, canUseDirectSolarPower: yes
12:27:53.767 > [DPL::loop] battery discharging prevented, PowerMeter: -44 W, target consumption: -50 W
```

helgeerbe, 2024-01-16:
Du hast Solar-Passthrough an. Wenn die Sonne scheint, wird der Strom direkt in dein Hausnetzt eingespeisst. Und zwar in der Höhe, wie du ihn verbrauchen kannst. Was passiert, wenn du das deaktivierst?

Manos1966, 2024-01-16:
Danke @helgeerbe 
Verdammt! Wieder wie ein Anfaenger rein gefallen 😡 🤦‍♂️ 

Solar-Passthrough aus, alles funktioniert wie erwartet:

[image]

spcqike, 2024-01-16:
wie @helgeerbe sagt, du nutzt solar-passtrough. und das ist (eigentlich) gut, da du Umwandlungsverluste vermeidest und damit die Effizienz erhöhst.

deine Antwort liegt auch im Log

12:27:51.294 > [DPL::loop] **StartTH reached: no**, StopTH reached: no, inverter is producing
12:27:51.359 > [DPL::loop] **SolarPT enabled**, Drain Strategy: 1, **canUseDirectSolarPower: yes**
12:27:51.428 > [DPL::loop] battery discharging prevented, PowerMeter: -48 W, target consumption: -50 W
12:27:51.499 > [DPL::loop] **Consuming Solar Power Only -> adjustedVictronChargePower: 922 W**, newPowerLimit: 134 W

in deinen Screens siehst du es quasi auch. der Victron erzeugt 467W, davon speist du direkt 168W ein. 

warum in deinem Bild die Batterie allerdings mit 455W entladen werden soll, verstehe ich nicht. (oder heißt das laden und entladen wären -xxx W?)

Manos1966, 2024-01-16:
Danke Helge!

und Danke Martin fuer die Beschreibung des Logs!

closed 2024-01-16 by Manos1966