hoylabs/OpenDTU-OnBattery #611: PowerMeter fehlerhaft

Tiese, 2024-01-16:
### What happened?

Version 2024.01.04 ist i.O.
Version 2024.01.16 zeigt den Fehler: [HttpPowerMeter] Couldn't find a value for phase 0 with Json query "StatusSNS/SM/16_7_0"

### To Reproduce Bug

Einstellungen -> Power Meter -> Testen

### Expected Behavior

Es sollte genau wie vor dem Update funktionieren!

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.01.16

### Relevant log/trace output

```Shell
[HttpPowerMeter] Getting the power of phase 1 failed.
18:23:45.645 > [HttpPowerMeter] Couldn't find a value for phase 0 with Json query "StatusSNS/SM/16_7_0"
18:23:45.653 > [HttpPowerMeter] Getting the power of phase 1 failed.
18:23:45.665 > [HttpPowerMeter] Couldn't find a value for phase 0 with Json query "StatusSNS/SM/16_7_0"
18:23:45.675 > PowerMeterClass: TotalPower:  0.00
```

Webranger2, 2024-01-16:
Das gleiche hier.
[image]

`19:04:18.190 > [HttpPowerMeter] Getting the power of phase 1 failed.
19:04:18.202 > HTTP Error connection refused
19:04:18.208 > [HttpPowerMeter] Getting the power of phase 1 failed.
19:04:18.215 > HTTP Error connection refused
19:04:18.221 > PowerMeterClass: TotalPower:  0.00
19:04:18.230 > [HttpPowerMeter] Getting the power of phase 1 failed.
19:04:18.238 > HTTP Error connection refused
`

helgeerbe, 2024-01-16:
Was bekommt ihr zurück, wenn ihr die url im Browser eingebt?

Der eine Fehler sagt, das die JSON Antwort nicht geparst werden kann. Der andere lehnt die Verbindung ab. Es ist aber keine Authentifizierung eingerichtet. Stimmt das?

Tiese, 2024-01-16:
Bei mir erscheint das...
`{"StatusSNS":{"Time":"2024-01-16T20:11:01","SM":{"1_8_0":2251.651,"2_8_0":4.264,"16_7_0":281.24,"36_7_0":128.17,"56_7_0":33.15,"76_7_0":119.92,"96_1_0":"1EBZ0102xxxxxx"}}}`

Webranger2, 2024-01-16:
Bei mir kommt das...
`{"val":641.57,"ack":true,"ts":1705432597646,"q":0,"from":"system.adapter.discovergy.0","user":"system.user.admin","lc":1705432597646,"type":"state","common":{"name":"Power_Consumption","type":"number","role":"value.power.consumption","unit":"W","custom":{"history.0":{"enabled":true,"changesOnly":true,"debounce":"1000","maxLength":"960","retention":"31536000","changesRelogInterval":0,"changesMinDelta":0,"aliasId":"","debounceTime":0,"blockTime":1000,"ignoreBelowNumber":"","disableSkippedValueLogging":false,"customRetentionDuration":365,"enableDebugLogs":false}},"write":false,"read":true},"native":{},"_id":"discovergy.0.60867681.Power_Consumption","acl":{"object":1636,"state":1636,"owner":"system.user.admin","ownerGroup":"system.group.administrator"}}
[image]
`

helgeerbe, 2024-01-16:
@Webranger2 das sieht eigentlich OK aus. Haber bei mir auch gerade basic auth mal deaktiviert. Aber bei mir wird richtig geparst.  Nur um sicher zu gehen. Wenn du mehrmals auf testen klickst, dann kommt immer das gleiche Ergebnis (connection refused)?

Fribur, 2024-01-16:
The first one looks related to the port. I might have done a mistake of passing always port 80 for http, and 430 for https. You clearly need port 8080. Should be easy fix. Sorry about that. Let me check.

The second one looks like json parsing error (we switched libraries)…thanks for posting the json string, that helps to debug why it fails.

Webranger2, 2024-01-16:
> @Webranger2 das sieht eigentlich OK aus. Haber bei mir auch gerade basic auth mal deaktiviert. Aber bei mir wird richtig geparst. Nur um sicher zu gehen. Wenn du mehrmals auf testen klickst, dann kommt immer das gleiche Ergebnis (connection refused)?

ja genau

Fribur, 2024-01-16:
@helgeerbe, @schlimmchen : bzgl dem port problem wuerde ich vorschlagen die HttpPowerMeterClass::extractUrlComponents zu ersetzen. Die Methode zerlegt eine URL in  

1. protocol (http oder https)
2. hostname (IP oder name)
3. uri
4. _port fehlt_(!!!)

Stattdess wuerde ich die Methode neu implementieren so wie hier: https://github.com/espressif/arduino-esp32/blob/da6325dd7e8e152094b19fe63190907f38ef1ff0/libraries/HTTPClient/src/HTTPClient.cpp#L274C12-L274C17
Die Methode extrahiert auch den port correct. Und wir koennen uns daruf verlassen das es geht das es die ofizielle Mthode ist eine URL zu zerlegen. 

Note: wir koennen leider nicht direkt dies httpclient Methode benutzen die nur mit der URL arbeitet, da diese Methode **nicht** in der Lage ist mit mDNS umzugehen. Wir muessen die httpclient::begin function benutzen wo alle parameter uebergeben werden (die hier: https://github.com/espressif/arduino-esp32/blob/da6325dd7e8e152094b19fe63190907f38ef1ff0/libraries/HTTPClient/src/HTTPClient.cpp#L177) , das heist wir haben keine Wahl als alles manuell zu extrahieren. 

Ich schaue mir jetzt das JOSN problem an.

Cartman586, 2024-01-16:
Same Problem here! 
[image]

Cartman586, 2024-01-16:
Please fix, Thanks

Fribur, 2024-01-16:
This should resolve the "connection refused" error report here by user utilizing non standard ports (i.e. not port 80 for http and 430 for https): https://github.com/helgeerbe/OpenDTU-OnBattery/pull/612 

@Cartman586: can you please post the URL you use in your query?

Cartman586, 2024-01-16:
http://192.168.1.119/rpc/shelly.getstatus

JSON em:0/total_act_power

2024.01.07 works

Am 16.01.2024 um 22:46 schrieb Fribur ***@***.***>:

This should resolve the "connection refused" error report here by user utilizing non standard ports (i.e. not port 80 for http and 430 for https): #612<https://github.com/helgeerbe/OpenDTU-OnBattery/pull/612>

@Cartman586<https://github.com/Cartman586>: can you please post the URL you use in your query?

—

Cartman586, 2024-01-16:
> This should resolve the "connection refused" error report here by user utilizing non standard ports (i.e. not port 80 for http and 430 for https): #612
> 
> This does NOT fix the JSON reponse parsing error es reported by @Tiese or @dragricola: could you please post your response?
> 
> @Cartman586: can you please post the URL you use in your query?

http://192.168.1.119/rpc/shelly.getstatus
JSON em:0/total_act_power
2024.01.07 works

Fribur, 2024-01-16:
> http://192.168.1.119/rpc/shelly.getstatus

OK, you are not affected by the port issue. In your case the JSON is not parsed correctly. Not yet fixed. Could you paste the full response as Webranger2 did when you paste you `http://192.168.1.119/rpc/shelly.getstatus` here into your browser? That would help figuring out why it is not correctly parsed.

Cartman586, 2024-01-16:
https://www.facebook.com/share/p/29Jf99zaFtpLHWxk/?mibextid=WC7FNe

> > http://192.168.1.119/rpc/shelly.getstatus
> 
> OK, you are not affected by the port issue. In your case the JSON is not parsed correctly. Not yet fixed. Could you paste the full response as Webranger2 did when you paste `http://192.168.1.119/rpc/shelly.getstatus` here into your browser? That would help figuring out why it is not correctly parsed.
> 
> And please also share the JSON path that you put in here (not quite clear from what you wrote) ![image](https://private-user-images.githubusercontent.com/16464641/297200735-76e44bd9-cfa6-41bd-a15b-148824fa7e38.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NDI3MDMsIm5iZiI6MTcwNTQ0MjQwMywicGF0aCI6Ii8xNjQ2NDY0MS8yOTcyMDA3MzUtNzZlNDRiZDktY2ZhNi00MWJkLWExNWItMTQ4ODI0ZmE3ZTM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE2VDIyMDAwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxMjk2Y2I1NzQ5NDRlNDYzNWRkZTM5YWE5NDQzYWE4Yzc2ZGM2MzM0NzU4NGIyNTM1NjdmNGQ3NTYzZWQ2MzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4luMXJmnyZ8GYcoKLJZ_lYPxQoZ4qXBJoHz_K3bpN4k)

> > http://192.168.1.119/rpc/shelly.getstatus
> 
> OK, you are not affected by the port issue. In your case the JSON is not parsed correctly. Not yet fixed. Could you paste the full response as Webranger2 did when you paste `http://192.168.1.119/rpc/shelly.getstatus` here into your browser? That would help figuring out why it is not correctly parsed.
> 
> And please also share the JSON path that you put in here (not quite clear from what you wrote) ![image](https://private-user-images.githubusercontent.com/16464641/297200735-76e44bd9-cfa6-41bd-a15b-148824fa7e38.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NDI3MDMsIm5iZiI6MTcwNTQ0MjQwMywicGF0aCI6Ii8xNjQ2NDY0MS8yOTcyMDA3MzUtNzZlNDRiZDktY2ZhNi00MWJkLWExNWItMTQ4ODI0ZmE3ZTM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE2VDIyMDAwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxMjk2Y2I1NzQ5NDRlNDYzNWRkZTM5YWE5NDQzYWE4Yzc2ZGM2MzM0NzU4NGIyNTM1NjdmNGQ3NTYzZWQ2MzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4luMXJmnyZ8GYcoKLJZ_lYPxQoZ4qXBJoHz_K3bpN4k)

`
[image]
`

https://www.facebook.com/share/p/29Jf99zaFtpLHWxk/?mibextid=WC7FNe

Cartman586, 2024-01-16:
Works Not

URL get: http://192.168.1.119/rpc/shelly.getstatus
JSON: em:0/total_act_power

Up to version 2024.01.07 it works

http://192.168.1.119/rpc/shelly.getstatus

Fribur, 2024-01-16:
@Cartman586 : sorry, this does not help. Could you please review/edit your last 2 posts? I like the see the full JSON response from your Powermeter when you use google chrome for querying it. Should look something like `{"StatusSNS":{"Time":"2024-01-16T20:11:01","SM":"1_8_0":2251.651,"2_8_0":4.264,"16_7_0":281.24,"36_7_0":128.17,"56_7_0":33.15,"76_7_0":119.92,"96_1_0":"1EBZ0102xxxxxx"}}}`

Cartman586, 2024-01-16:
> @Cartman586 : sorry, this does not help. Could you please review/edit your last 2 posts? I like the see the full JSON response from your Powermeter when you use google chrome for querying it. Should [...]

{"ble":{},"cloud":{"connected":true},"em:0":{"id":0,"a_current":2.145,"a_voltage":228.9,"a_act_power":373.6,"a_aprt_power":491.6,"a_pf":0.81,"a_freq":49.9,"b_current":1.290,"b_voltage":226.5,"b_act_power":150.1,"b_aprt_power":292.3,"b_pf":0.67,"b_freq":49.9,"c_current":1.362,"c_voltage":231.2,"c_act_power":281.6,"c_aprt_power":314.3,"c_pf":0.91,"c_freq":49.9,"n_current":null,"total_current":4.797,"total_act_power":805.281,"total_aprt_power":1098.230, "user_calibrated_phase":[]},"emdata:0":{"id":0,"a_total_act_energy":665029.34,"a_total_act_ret_energy":29346.00,"b_total_act_energy":333918.62,"b_total_act_ret_energy":23075.14,"c_total_act_energy":347418.57,"c_total_act_ret_energy":44.39,"total_act":1346366.54, "total_act_ret":52465.53},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":false},"sys":{"mac":"E05A1B3346BC","restart_required":false,"time":"23:08","unixtime":1705442892,"uptime":3830,"ram_size":256064,"ram_free":86228,"fs_size":524288,"fs_free":184320,"cfg_rev":11,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{},"reset_reason":3},"temperature:0":{"id": 0,"tC":59.0, "tF":138.2},"wifi":{"sta_ip":"192.168.1.119","status":"got ip","ssid":"SweetHome","rssi":-73},"ws":{"connected":false}}

Cartman586, 2024-01-16:
{"ble":{},"cloud":{"connected":true},"em:0":{"id":0,"a_current":2.145,"a_voltage":228.9,"a_act_power":373.6,"a_aprt_power":491.6,"a_pf":0.81,"a_freq":49.9,"b_current":1.290,"b_voltage":226.5,"b_act_power":150.1,"b_aprt_power":292.3,"b_pf":0.67,"b_freq":49.9,"c_current":1.362,"c_voltage":231.2,"c_act_power":281.6,"c_aprt_power":314.3,"c_pf":0.91,"c_freq":49.9,"n_current":null,"total_current":4.797,"total_act_power":805.281,"total_aprt_power":1098.230, "user_calibrated_phase":[]},"emdata:0":{"id":0,"a_total_act_energy":665029.34,"a_total_act_ret_energy":29346.00,"b_total_act_energy":333918.62,"b_total_act_ret_energy":23075.14,"c_total_act_energy":347418.57,"c_total_act_ret_energy":44.39,"total_act":1346366.54, "total_act_ret":52465.53},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":false},"sys":{"mac":"E05A1B3346BC","restart_required":false,"time":"23:08","unixtime":1705442892,"uptime":3830,"ram_size":256064,"ram_free":86228,"fs_size":524288,"fs_free":184320,"cfg_rev":11,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{},"reset_reason":3},"temperature:0":{"id": 0,"tC":59.0, "tF":138.2},"wifi":{"sta_ip":"192.168.1.119","status":"got ip","ssid":"SweetHome","rssi":-73},"ws":{"connected":false}}

Fribur, 2024-01-16:
Perfect. Thanks, that helps. So just to verify the JSON path you use: is this it? 
[image]

Cartman586, 2024-01-16:
> Perfect. Thanks, that helps. So just to verify the JSON path you use: is this it? [...]

 Yes

octavvian, 2024-01-16:
Hi,
same here, tasmota powermeter
before last fw upgrade worked just fine!
[image]
[image]
Any fix?
Thx!

Cartman586, 2024-01-16:
Shelly EM:
URL: http://IP_ADR_DES_SHELLY/emeter/0
JSON: power

Shelly Pro 1pm: 
URL: http://IP_ADR_DES_SHELLY/rpc/Shelly.GetStatus 
JSON: switch:0/apower

Shelly 3em:
URL:  http://IP_ADR_DES_SHELLY/status 
JSON: total_power

Shelly pro 3em 
URL: http://IP_ADR_DES_SHELLY/rpc/shelly.getstatus   
JSON: em:0/total_act_power

Hichi Lesekopf 
URL: http://IP_ADR_DES_HICHI/cm?cmnd=Status%208 
JSON: StatusSNS/MT681/Power_cur

Cartman586, 2024-01-16:
Des

image.png (view on web)<https://github.com/helgeerbe/OpenDTU-OnBattery/assets/16464641/801059d9-599c-4853-84bc-f5a4c2aafea2>

Fribur, 2024-01-16:
@Cartman586: could you maybe edit (delete/clean up) some of your posts above? This thread here becomes difficult to read/follow.

Fribur, 2024-01-16:
click on the 3 dots of your posts (delete/edit):
[image]

Cartman586, 2024-01-16:
> click on the 3 dots of your posts (delete/edit): [...]

Ok, are you already find The bug?

dragricola, 2024-01-16:
> Was bekommt ihr zurück, wenn ihr die url im Browser eingebt?
> 
> Der eine Fehler sagt, das die JSON Antwort nicht geparst werden kann. Der andere lehnt die Verbindung ab. Es ist aber keine Authentifizierung eingerichtet. Stimmt das?

Ich habe wieder Version V24.1.4 installiert. Mein JSON Pfad ist: StatusSNS/SML/total_power. Antwort auf http://192.168.178.122/cm?cmnd=status%208 ergibt:

{
    "StatusSNS": {
        "Time": "2024-01-17T00:00:20",
        "SML": {
            "DJ_TPWRIN": 948.524,
            "DJ_TPWROUT": 1826.695,
            "DJ_TPWRCURR": 0.117,
            "total_power": 117,
            "zählerstand_total": 948524.2,
            "zählerstand_tarif_1": 948513.8,
            "zählerstand_tarif_2": 10.3,
            "wirkenergie_total": 1826695.1,
            "wirkenergie_tarif_1": 1826693.3,
            "wirkenergie_tarif_2": 1.8
        }
    }
}

Fribur, 2024-01-17:
The JSON parsing error is related to replacing FirebaseJSON library with Arduino JSON as it saves a lot of code (= space on the ESP32). HOWEVER, as became evident here, the current implementation cannot access nested keys, and I also see no easy way to re-implement the current complicated nested keys that users appear to use. Unless someone feels like doing that (I do not), I would propose to just go back to FireBaseJSON. I updated pull request https://github.com/helgeerbe/OpenDTU-OnBattery/pull/612 accordingly. That negates some of the aspired benefits of https://github.com/helgeerbe/OpenDTU-OnBattery/pull/594 , but retains the ability to resolve local hosts via mDNS.

dragricola, 2024-01-17:
@Fribur : Bei Build #930 funktioniert das Parsing jetzt!

Fribur, 2024-01-17:
Was ist "build 930"? Meinst du das du https://github.com/helgeerbe/OpenDTU-OnBattery/pull/612 local bei dir kompiliert hast?

Cartman586, 2024-01-17:
Bitte so machen, damit es wieder wie Build 2024.01.07 läuft

Please do this so that it runs like build 2024.01.07 again

Am 17.01.2024 um 01:40 schrieb Fribur ***@***.***>:

The JSON parsing error is related to replacing FirebaseJSON library with Arduino JSON as it saves a lot of code (= space on the ESP32). HOWEVER, as became evident here, the current implementation cannot access nested keys, and I also see no easy way to re-implement the current complicated nested keys that users appear to use. Unless someone feels like doing that (I do not), I would propose to just go back to FireBaseJSON. I updated pull request #612<https://github.com/helgeerbe/OpenDTU-OnBattery/pull/612> accordingly. That negates some of the aspired benefits of #594<https://github.com/helgeerbe/OpenDTU-OnBattery/pull/594> , but retains the ability to resolve local hosts via mDNS.

dragricola, 2024-01-17:
> Was ist "build 930"? Meinst du das du #612 local bei dir kompiliert hast?

Nein, ich hatte das Binary von hier https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/7549306137 heruntergeladen und installiert.

Cartman586, 2024-01-17:
Wird der Fehler heute behoben? Danke

Fribur, 2024-01-17:
Bitte versuche den Build den dragicola 1 post über dir verlinkt hat.

Und bitte clean up deine Messages oben!(unnötige Quotations und Duplikationen) damit andere den thread lesen können.

helgeerbe, 2024-01-17:
Hallo, ich habe den Fix von Fribur ( PR #612 ) in development branch übernommen. Die Binaries findet ihr hier: https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/7555678082

Könnt ihr das bitte testen und Feedback geben. Dann kann ich das in die Release übernehmen.

Cartman586, 2024-01-17:
> Bitte versuche den Build den dragicola 1 post über dir verlinkt hat.
> 
> Und bitte clean up deine Messages oben!(unnötige Quotations und Duplikationen) damit andere den thread lesen können.

Hallo, Link Firmware: https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/7549306137 
funktioniert mit PowerMeter wieder (Firmware 03FB80B).

Tiese, 2024-01-17:
Hallo @helgeerbe,
mit deinem Link funktioniert es 👍

Cartman586, 2024-01-17:
Geht

dragricola, 2024-01-17:
> Hallo, ich habe den Fix von Fribur ( PR #612 ) in development branch übernommen. Die Binaries findet ihr hier: https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/7555678082
> 
> Könnt ihr das bitte testen und Feedback geben. Dann kann ich das in die Release übernehmen.

Ja, der Fix funktioniert auch bei mir wieder!

helgeerbe, 2024-01-17:
Sollte mit der [2024.01.17](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.01.17) gefixt sein. 

@Tiese bitte testen und dann das Issue hier zumachen.

Tiese, 2024-01-17:
Danke!

closed 2024-01-17 by Tiese

Webranger2, 2024-01-17:
Mit der [2024.01.17](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.01.17) geht es bei mir nun wieder