hoylabs/OpenDTU-OnBattery #613: LifePO4 25,6V Leistungsanzeige (Watt) inkorrekt

Cartman586, 2024-01-17:
### What happened?

Hallo, da ich hier Jungs vom Fach und Entwicklung habe folgendes Problem, ließt man oft:

Ich besitze einen Hoymiles HM-800, dieser produziert insgesamt 810,9W (siehe Bild Shelly). Dieser Wert wurde mit einem Messgerät überprüft und ist korrekt, dieser wird auch eingespeist. 

Tatsächlich zeigt die OpenDTUonBattery aber bei 100% Regulierung lediglich 494,6W an. 

Dieses Phänomen passiert nur dann, wenn der Wechselrichter an einer LifePO4 Batterie hängt, in meinen Fall 25,6V 100Ah.

Von 810,9W realen Wert und den falschen 494,6W Wert haben wir einen Unterschied von 61%.

Wäre es bitte möglich, einen Offset in % einzubauen, um dem Problem entgegen zu wirken?

Hängen Solarmodule am Wechselrichter, so sind die Werte korrekt. Nur bei den Akkus stimmt der Wert nicht.

Bitte um Hilfe, danke 

#######

Hello, since I have here guys from the field and development the following problem, you often read:

I have a Hoymiles HM-800, which produces a total of 810.9W (see picture Shelly). This value was checked with a measuring device and is correct, it is also fed in.

However, the OpenDTUonBattery actually only shows 494.6W at 100% regulation.

This phenomenon only occurs when the inverter is connected to a LifePO4 battery, in my case 25.6V 100Ah.

From 810.9W real value and the false 494.6W value we have a difference of 61%.

Would it please be possible to add an offset in % to counteract the problem?

If solar modules are connected to the inverter, the values are correct. Only with the batteries is the value incorrect.

Please help, thank you

### To Reproduce Bug

LifePO4 25,6V Akku anstecken 
Connect LifePO4 25.6V battery

### Expected Behavior

Korrekter Leistungswerr 
[image]
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.01.17

spcqike, 2024-01-17:
Hallo,

kannst du bitte mal beobachten, welchen Wirkungsgrad (und Leistungsfaktor und Blindleistung) der Wechselrichter seinerseits meldet? die Leistungsdaten werden ja nicht wesentlich durch die Software manipuliert, sondern vom Hoymiles gemeldet und angezeigt.

[image]

Im Batteriebetrieb mit 24V Batterien kam es bei anderen Usern bereits zu dem Phänomen, dass ein Wirkungsgrad von lediglich 40-50% angezeigt wurde. Ob die tatsächliche Einspeisung in dem Moment höher war, weiß ich nicht. Bei 48V Systemen taucht das Problem wohl nicht mehr auf.

Wenn auch bei dir der gemeldete Wirkungsgrad so gering ist, sehe ich keinen Grund, die Software einfach um einen Offset oder um x Prozent zu erhöhen. Woher soll die openDTU wissen, dass die 490W bzw 40% Wirkungsgrad grade ein Fehler vom Wechselrichter sind und nicht real anliegen? Damit könntest du ja in der nächsten Minute, wenn der WR wieder korrekt 800W meldet, eine falsche Berechnung von 1.200W bekommen.

btw: dein Wechselrichter meldet viele Netzprobleme. vielleicht lief er wirklich "unrund" und hatte einfach einen verschobenen Leistungsfaktor / Wirkungsgrad
[image]

Cartman586, 2024-01-17:
Hallo,

bei 100% zeigt er bei Opendtu 494,6 Watt, in Wahrheit sind es aber 810,9 Watt bei 100%.

Das Problem ist nicht die opendtu, auch die Hoymiles DTU zeigt zeigt 494,6 Watt.

Durch die „niedrige Lifepo4 Spannung von rund 25,6V kommt der Wechselrichter nicht ganz klar.

Daher die Bitte um einen Offset.

Geprüft durch ein externes Strommessgerät, eingespeist werden wirklich 810,9 Watt

Bitte um Hilfe vielen Dank

grafik.png (view on web)<https://github.com/helgeerbe/OpenDTU-OnBattery/assets/50098985/8efc0581-797f-4312-a8e6-3a306272779b>

grafik.png (view on web)<https://github.com/helgeerbe/OpenDTU-OnBattery/assets/50098985/bcc95007-5f28-4060-8298-280a34024c48>

Cartman586, 2024-01-17:
[
[image]
[image]
](url)

Cartman586, 2024-01-17:
Hallo, 

bei 100% zeigt er bei Opendtu 494,6 Watt, in Wahrheit sind es aber 810,9 Watt bei 100%. 

Das Problem ist nicht die opendtu, auch die Hoymiles DTU zeigt zeigt 494,6 Watt.

Durch die „niedrige Lifepo4 Spannung von rund 25,6V kommt der Wechselrichter nicht ganz klar. 

Daher die Bitte um einen Offset. 

Geprüft durch ein externes Strommessgerät, eingespeist werden wirklich 810,9 Watt

Bitte um Hilfe vielen Dank 
[image]
[image]

Cartman586, 2024-01-17:
Details beim Messen
25,6V 100Ah LifePO4

Measuring device"	OpenDTU/Hoymiles	Abweichung / Deviation	
100%	804,3 W		61%	490,7		39%	Too much
90%	804,3 W		55%	490,7		35%	Too much
80%	804,3 W		49%	490,8		31%	Too much
70%	804,3 W		43%	491,3		27%	Too much
60%	708,4 W		41%	479,5		19%	Too much
50%	418,7 W		47%	392,6		3%	OK
40%	335 W		39%	325,1		1%	OK
30%	244,6 W		29%	232,4		1%	OK
20%	145,1 W		19%	136,1		1%	OK
10%	60,1 W		10%	60,9		0%	OK

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Solarteur, 2024-01-18:
Das ist wohl ein generelles Problem bei "24 V-Batterien". Siehe z. B. https://github.com/tbnobody/OpenDTU/issues/1102#issuecomment-1621138846 oder https://github.com/tbnobody/OpenDTU/discussions/1078#discussioncomment-7461152. Bei höheren Batterie-Spannungen tritt dieser Fehler nicht auf.

Cartman586, 2024-01-18:
Denke ich auch. Darum möchte ich gerne einen Offset für 30%,40%,50%,60%,70%,80%,90,100% festlegen können.

Das Problem ist generell bekannt und so könnte man entgegen wirken.

schlimmchen, 2024-01-18:
Da das offensichtlich ein Fehler in der Firmware der Hoymiles Wechselrichter ist, sollte der auch dort behoben werden. OpenDTU(-onBattery) ist nicht dafür da, Fehler in anderen Komponenten zu kaschieren.

Cartman586, 2024-01-18:
> Da das offensichtlich ein Fehler in der Firmware der Hoymiles Wechselrichter ist, sollte der auch dort behoben werden. OpenDTU(-onBattery) ist nicht dafür da, Fehler in anderen Komponenten zu kaschieren.

Hoymiles wird hier leider nichts anpassen.

spcqike, 2024-01-18:
Ich würde es auch nicht in Software anpassen. Welchem Messgerät willst du denn vertrauen? Woher weißt du, dass der Shelly auch wirklich richtig misst? Vielleicht messen beide richtig, nur eben anders. Im Sinne von RMS und Peak. 

Dazu kommt: du solltest dein Setup noch einmal in Gänze überdenken. 800W bei 25.6V sind immer noch 31.2A. Der HM-800 verträgt wohl laut Datenblatt nur 2x12.5A. Das wäre mir auf Dauer einfach zu heiß, da mehr raus ziehen zu wollen. Bei 25.6V wäre theoretisch bei 640W Schluss. 

Alle Ungenauigkeiten darüber hinaus liegen wohl mehr in der Hardware und Software des wechselrichters begründet. (Irgendwie muss er ja arbeiten. Er wird am Eingang irgendwie Pulsen und Kondensatoren laden und entladen. Nur ist eine Batterie mit >100A eben kein Solarpaneel, was irgendwo natürlich begrenzt.)

Cartman586, 2024-01-18:
Werte wurden mit einen externen strommessgerät geprüft und sind korrekt

helgeerbe, 2024-01-18:
Ich glaube nicht, dass man das über eine einfache Offset Tabelle lösen kann. Dazu ändert sich die Spannung im Betrieb und der Hoymiles ist nun mal über seinen Eingangsstrom begrenzt.  Meine 24 V Batterie liegt zwischen 21.x V (leer) und 26.x V (voll). Welche Wattzahl ist dann 100%?

Cartman586, 2024-01-18:
Aber wie kommen der Sache zumindest mal viel näher als Riesen Differenzen

schlimmchen, 2024-01-18:
> Aber wie kommen der Sache zumindest mal viel näher als Riesen Differenzen

Du findest wir sollten Zeit und Energie in ein Feature investieren, das kaum Leute brauchen, geschweige denn in Ihren Einstellungen sehen wollen, damit irgendwelche Werte **weniger falsch**, also am Ende immer noch **falsch** sind? Inkl. Erläuterungen im Wiki, Support in Tickets, Pflege des Codes, etc.

Wenn du dein Problem lösen willst, solltest du auf ein 48V System aufrüsten. Damit kommen die Hoymiles offenbar klar, auch wenn sie dann immer noch missbraucht sind, da sie schließlich für den Betrieb an Solarmodulen gebaut sind.

Oder nutze den gemessenen Wert deines Shelly am Ausgang des WR um die erzeugte Leistung zu kennen.

Und nochmal: Es ist nicht sinnvoll, Probleme anderer Komponenten durch OpenDTU(-OnBattery) kaschieren zu lassen.

Cartman586, 2024-01-18:
Ich würde es gerne machen, habe aber nicht die Fähigkeiten dazu

Sorbat, 2024-01-18:
Das ist irgendwie unglaublich anstrengend zu lesen

Manos1966, 2024-01-19:
Hallo @Cartman586 
wie du erwaehnt hast, es ist kein OpenDTU Problem, sondern Hoymiles-Problem.
Wie spcqike geschrieben hat, das Problem passiert in Verbindung mit 25V Batterien.
Ein paar Bemerkungen meinerseits:

- Ich glaube die Werte, die deine Shelly liefert, nicht. Ich habe eins deiner Bilder als Beispiel benutzt.
Die Hoymiles HM-Serie nimmt Maximal circa 10,5A pro Eingang (deine Bilder bestätigen das, 10,21A & 10,59A Eingang) 
Watt wird berechnet als Volt x Amperes, 25Vx10,21A=**253W**, 25Vx10,59A=**262,5W** 
Total 253+262,5= **515,5W DC-Leistung**
515,5W DC-Leistung umgewandelt mit 95,19% Effizienz = **490,7W AC-Leistung**

Zusammenfassend: Die Amperes die jeder Hoymiles HM-Eingang annehmen kann, sind aus Erfahrung Fest: 10,5A
Bei einer 25V Batterie sind das also 25Vx10,5A = 262,5Watt pro Eingang, also 262,5Wx2 = **525W insgesamt** (DC-Seite)

Aus 525W DC, mit einer Effizient von 100% bekäme man 525W AC **Bestenfalls** - Mehr als das wäre "Zauberei" - _oder Vielleicht die Lösung des Energieproblems?_ 
(wir wissen der HM hat eine Effizient von 95%, die zu den Besten Effizienzen bei Micro-Wechselrichter gehört)
ENDE DER DISKUSSION (...und willkommen in die 48V-Welt 😄 ) 

Was jetzt (höchst wahrscheinlich) passiert: 
Nimm eine Amperzange und macht eine Messung der Amperes die aus deiner Batterie raus gehen.
Ich befuerchte, du wirst feststellen, anstatt den 10,21A & 10,59A = 20,70A (die die Hoymiles pro Eingang berichtet hat), wirst du feststellen, dass viel mehr A raus fliessen!
Es koennten - ich berechne Rückwirkend:  Shelly 804,3W / 95,19% Effizienz = 844,9W / 25V = **33,79 A** 
Also, FALLS deine Amperzange 33,79 A aus deiner Batterie raus kommen misst, dann stimmen die Zahlen deiner Shelly (und die Hoymiles Berechnungen sind ***** )
Jemand hat es bereits gemessen und festgestellt (ich habe den Namen und Link nicht im Kopf)

[image]

spcqike, 2024-01-19:
@Manos1966 man bräuchte keine einfache Zange sondern eher ein DSO (digitales Speicher Oszilloskop). Der HM wird keinen konstanten Strom ziehen. Er muss das ja irgendwie hoch transformieren und zerhacken, damit er aus 24VDC oder auch 48VDC dann ein 230V Sinus bekommt. 

Also einfach einen passenden Shunt an die Batterie und nachmessen :D

Manos1966, 2024-01-19:
🙈

SW-Niko, 2024-01-21:
Hallo @spcqike ,
Der "zerhackte" Strom (>200kHz) wird fast vollständig durch die internen Kondensatoren abgefangen. Zwischen Batterie und Inverter fließt ein mit DC überlagerter Sinus Strom. C2 in dem Bild. Das habe ich mal mit einem HM-300 und einer 20V DC-Quelle gemessen.
[image]

Es wäre aber trotzdem interessant ob der Mittelwert, gemessen mit einem Multimeter (DC) mit der Hoymiles Anzeige übereinstimmt.

spcqike, 2024-01-21:
@SW-Niko Danke für die Messung und deine Mühe. Sehr interessant. 

Also, extern nachgemessen stimmen scheinbar sowohl der DC Eingang als auch der AC Ausgang. An beiden Seiten erreichst du beim HM-300 mit einer 20VDC Quelle die ~300W. Ist das bei unterschiedlichen Limit Werten genauso?

jetzt wäre es interessant zu wissen, ob der hoymiles die Werte selber auch so misst (der HMS scheint da bei 24V ja Probleme zu haben) und ob die HMS Serie genauso reagiert/regelt.

mich wollte mir im laufe des Frühjahres auch einen HMS-2000 holen. Wir haben auf Arbeit zwar nur ein 2 Kanal DSO, aber vielleicht kann ich den da auch mal durch messen.

Cartman586, 2024-01-25:
Das hört sich plausibel an. Ich habe ja oben verschiedenste Messungen und Vergleiche aufgestellt.
Es werden wirklich die Werte erreicht! 

Die Shelly Werte sind korrekt, wurde nachgemessen. 

Mit dieser plausiblen Informationen können vielleicht die Jungs hier in der Firmware entgegen wirken!

Wäre hier sehr dankbar, wenn man dieses Problem entgegen gehen könnte. 

liebes opendtu on Battery Team, könnt ihr mit der Information von Joe6899 was entwickeln.

kann die Werte bestätigen.

spcqike, 2024-01-25:
ich kann mir immer noch nicht vorstellen, dass man da etwas sinnvolles und zielführendes entwickeln kann. Woran soll man es denn genau festmachen?

deine Tabelle zeigt ja bereits ab 70% Limit deine 800W am AC Ausgang. ab 60% Limit gibt es Probleme mit der Hoymiles internen Messung. Das deckt sich ganz gut mit den erwähnten 10A Strommessbereich. (bei 60% vom HM-800 und 25V Batteriespannung, wären das theoretisch 19.2A, also 9.6A pro Eingang und damit recht nah an den 10A Grenzstrom)

Nun schwankt der Strom dann aber auch je nach Batteriespannung / -ladezustand. die Batterie bricht mit steigendem Strom unterschiedlich stark ein (Alterung, Leitungsverluste) etc. pp. einige Faktoren kann man gut heraus rechnen (Leitungswiderstände), aber eben nicht alle.

Cartman586, 2024-01-25:
Wenn da jetzt jemand eine Idee hätte wäre ich dankbar.

spcqike, 2024-01-25:
Was spricht denn gegen eine Hardwarelösung, damit die Software die falschen Werte des Hoymiles ( der nicht für dieses Lastszenario gebaut ist) gar nicht erst auftreten?

die "einfachste" Idee, die mir einfällt, ist ein SmartShunt und ein Stepup-Wandler.

der Shunt misst die tatsächliche Batteriespannung und ermittelt den SoC (das ist sowieso genauer, als sich auf die Spannung zu verlassen) und der Step-Up erhöht die Spannung in einen Bereich, in dem der HM keine Probleme mehr macht. 

bspw. 
https://github.com/kscholty/SmartShuntINA226
und
https://www.ebay.de/itm/164359689494 (die 1500W Version, da die kleinere ggf. nicht genug Strom am Eingang verträgt, und dann halt einen Wandler pro Eingang, oder einen größeren/stärkeren Wandler)

Alternativ die bereits angesprochene 48V Batterie.

Joe6899, 2024-01-25:
The issue is that the DTU values reported from the inverter cannot be relied on when operating at 24V and above 10A.   But this is just the reporting data. there is no technical issue to fix- everything works fine and  your electric meter of course correctly measures the actual power. 
The issue with adjusting and compensating things in Opendtu is that inverter total yield,efficiency  etc are calculated and sent from the inverter - cannot change this data directly.

spcqike, 2024-01-26:
@Joe6899 well, there is one technical issue. the ac output power is to high, whenever the HM can't measure the in-comming dc current (>10A) correctly.
see https://github.com/helgeerbe/OpenDTU-OnBattery/issues/613#issuecomment-1896302434
with 60% limit (expected to be 480W for the used HM-800), the inverter allready produces 700W. with 70% and more, it will be fully open only.
it works fine up to 50% (400W, 16A in total with his battery with ~25.5V at this moment). after that it will create wrong values and you can't limit to a certain output value very well.

but again, IMO its not an issue, that one can or should fix in software.

Cartman586, 2024-01-26:
Kompromiss? Vielleicht kann man einen weiteren Shelly hinzufügen und einstellen, nimm den Shelly Wert als Wechselrichter Leistung. Wäre das bitte realisierbar?

Danke

Joe6899, 2024-02-07:
I agree that there is a regulation issue  when it gets past the point of accurate current  measurement.  
Calculated  target power setting gets confused and hunts around a lot.
The alternative to using a shelly power measurement is to assume (for regulation purposes) that the delivered power=set power.  i.,e add additional configuration option.

Cartman586, 2024-02-07:
Would you implement/program this?

Cartman586, 2024-02-09:
@helgeerbe Would you implement/program this? Thanks

hoschiking, 2024-02-10:
Ich persönlich nutze auch eine 24V Batterie und habe das Verhalten des Hoymiles ebenfalls festgestellt.
Ich bin gleicher Meinung wie @schlimmchen , das ist fehlerhaftes Verhaltes des Hoymiles, weill dieser eben nicht
für einen Betrieb an einer Batterie entwickelt wurde.
Mein HM-300 lässt sich nur bis ca. 260W steuern, da bis zu dieser Leistung der Eingangsstrom innerhalb der Spezifikation liegt. Darüber hinaus übersteuert dieser und speist ca. 330W, da greift wohl eine interne AC-seitige Leistungsbgrenzug.
In diesem Betriebszustand stimmt dann eben die angezeigte und tatsächlich eingespeiste Leistung nicht mehr überein.
Diese Verhalten wurde auch schon in anderen Foren beschrieben. 
In meinen Augen gibt es dafür nur eine Lösung: Die Maximalleistung im DPL darf nur so hoch eingestellt werden, wie die Spec des Hoymiles es zulässt. Deswegen tritt der Fehler bei 48V nicht auf, da man in dem Fall weit vom Limit des DC-Eingangs ist.

schlimmchen, 2024-02-10:
Danke, @hoschiking, das war mir noch gar nicht klar. Der HM-800 verträgt laut Datenblatt nur 12.5A pro Eingang und hat derer zwei. Bei 24V Nennspannung können damit nur höchstens 600W erzeugt werden. Dass @Cartman586 sich beschwert, dass die Anzeige der Leistung nicht korrekt sei bei 700W und mehr ist schlicht nicht ernst zu nehmen.

@helgeerbe Das Issue kann getrost geschlossen werden.

Cartman586, 2024-02-10:
> Danke, @hoschiking, das war mir noch gar nicht klar. Der HM-800 verträgt laut Datenblatt nur 12.5A pro Eingang und hat derer zwei. Bei 24V Nennspannung können damit nur höchstens 600W erzeugt [...]

Bitte freundlich bleiben, sonst nehme ich dich schlicht auch nicht ernst. Danke

schlimmchen, 2024-02-10:
Das mag harsch formuliert gewesen sein, aber es haben dir inzwischen mindestens ein halbes Dutzend Leute freundlich erklärt, dass OpenDTU-OnBattery dein Problem nicht lösen kann. Und anstelle bei Helge zu bestellen

> Would you implement/program this? Thanks

wäre etwas Einsicht von deiner Seite der Diskussion zuträglich.

Bitte stelle das obere Leistungslimit deines HM-800 auf höchstens 600W, dann brennt dir mittelfristig nichts ab und du siehst auch keine falschen AC Power Werte.

hoschiking, 2024-02-10:
@schlimmchen 
so ist es. Der Wert des externen Messgerätes (~800W) ist dann auch korrekt, nur eben darf der Hoymiles so viel nicht
einspeisen, irgendwann raucht der einfach ab. Der DC-Strom liegt bei 800W weit oberhalb des erlaubten Bereiches und die Leistungsregelung ist praktisch außer Funktion, da der Hoymiles oberhalb des DC-Current Limits den Strom nicht mehr richtig messen kann und zeigt dann irgend einen falschen Wert an. Zum Glück wurde eine AC-seitige Leistungsbegrenzung implementiert, die den Wechselrichter vor dem sofortigen Tod bewahrt.  
Ich empfehle diesen Thread in einem anderen Forum als Lektüre.

https://www.akkudoktor.net/forum/hoymiles-wechselrichter/sammelthema-erfahrungsaustausch-betreiber-hoymiles-mikrowechselrichter-an-akku/

hoschiking, 2024-02-10:
@Cartman586 
Um bei der Sache zu bleiben. Habe mir die Mühe gemacht und deine Screenshots durchforstet. Du hast dir sogar Mühe gemacht und einige Messungen durchgeführt und damit das Verhalten sogar bestätigt. Daten interpretieren ist nicht immer einfach ;-)
Schaue dir doch deine Tabelle an:
Während die angezeigte Leistung ab 43% bei 490W verbleibt und sich nicht mehr ändert (der Shunt wird übersteuert), liegt die tatsächliche Lestung bei bereits 800W (Leistungsbegrenzung des AC-Ausgangs). Es ist alles andere als Zufall......
Würdest du bei dem Versuch noch den DC-Strom loggen, würde dir auffallen, dass ab dem Punkt sich der gemessene Strom nicht mehr ändert. 
[image]

Bei meinem MH-300 liegt der maximale richtig gemessene Strom bei 9,9A.
Jede Leistungseinstellung, die zu einem DC-Strom >9,9A führt,  bewirkt, dass die AC-Leistung auf 330W hoch geht,
dabei bleibt die angezeigte AC-Leistung bei 240W stehen, ~260W DC-seitig (9,9A * Ubat).
 
Jetzt müßtest du genug Daten haben um es endlich einzusehen.

Ich habe mir die Mühe gemacht, weil es augenscheinlich nicht bekannt ist und es viele potentielle Betroffene geben wird.
Schönen Abend

Joe6899, 2024-02-18:
Thankyou for the detailed analysis. 

Personally I do not think marginally exceeding 11.5A is that risky. 
SC current rating is 16A. 
Even if you run you battery as low as 24V the max input current is still going to be less than 15A.
I believe the principle design limitation on the continuous current rating (max 11.5A) is the temperature of the flyback MOSFETs.
Because the MOSFET rds has a positive temp coefficient, the ambient temperature is actually the biggest concern with the current rating. The 11.5A limit is set for an ambient temp when the inverter is placed right behind the PV panel on a hot sunny day.

Did any one ever notice that their inverter did not stay cool when run at max output  power @ 24V?
My HM600 barely gets hand warm, even on a summers day in the shade.

schlimmchen, 2024-02-18:
> SC current rating is 16A.

What is SC?

> Personally I do not think marginally exceeding 11.5A is that risky. [...] the max input current is still going to be less than 15A.

You are kidding, right? How is a 30% increase in current (15A rather than 11.5A) "marginal"?

> I believe the principle design limitation on the continuous current rating (max 11.5A) is the temperature of the flyback MOSFETs.

You cold very well be much better at assessing this than me, but for now I do believe that you are making wild assumptions and that the manufacturer should be trusted with their respective rating. I am not convinced that you know what other parts of the inverter are assuming the advertised maximum input current for safe operation. Certainly their calculating logic assumes it, otherwise the output power value would not break when trying to force-feed too high currents into the inverter.

Joe6899, 2024-02-19:
SC = short circuit current

1. in the measurement example above.  343 watts is deliver on the DC side and 330W is outputted on the AC side. So inverter has an efficiency of 96% when operated with a input current of 13A @ 25C  (as opposed to data sheet spec of 11.5A @ 85C). The  Inverter is clearly showing no indication of distress and imminent release of magic smoke.

Is there anyone whom has observed actual poor efficiency when running hoymiles inverters from 24V ?   It seems to  me that the claims of poor efficiency are based on measuring the input current and then doing an efficiency calculation based on the incorrectly stated AC power in the the DTU.

Joe6899, 2024-02-19:
Permitted current is the SC current - 15A. same for HM300,HM700 and HM800.

As a thought experiment...

 let's consider that instead of a 24V battery, An actual 36V solar panel with SC current rating of 15A is used.
PV SC current characteristic is extremely flat for all PV panels.  24V @ 15A for a 15A SC rated PV panel is not only possible but likely.
Input current at maximum power point is always less than the rated SC current.  But maximum power point is not a given.....

e.g on a bright sunny day in Wuppetal your partner decides to hang out your duvet on your balcony. Inadvertently completely covering your recently installed solar panels.  The hoymiles  MMPT tracker optimizes the PV voltage down to 20V and you end up with  5W of output power.
 Realizing this situation you dash out to the balcony and pull off the cover. Immediately the input current shoots up to 15A on the input capacitor which is currently setting at 20V.

hoschiking, 2024-02-19:
In theory, anything is possible, but in practice, the wrong inverter has been selected for the existing module. No matter which combination and which manufacturer, when selecting the components you always have to choose the right inverter for the module. If you don't do this, it's your own fault. First hit on Google for a typical 300W module:
Zellentyp: Monokristallin
Leistung max.: 300 Watt
Leerlaufspannung max.: 34,05 V
MPP-Spannung max.: 29,25 V
MPP-Strom max.: 10,25 A
Kurzschlussstrom max.: 10,82 A
Modulwirkungsgrad: 19,8 %

Do you see any problem? Me not.
The voltage of typical 300-450W PV-Moduls is nominaly 30-45V.

uli-bs, 2024-02-21:
Ich will denn auch mal meinen Senf dazu geben... ;)
Hier werkelt ein HM-600 mit beiden MPPTs an einem 24V 100Ah LiPoFe4 Akku, flankiert von einem HM-1200 und einem HM-300 jeweils an realen Modulen.
Das Verhalten, dass die tatsächlich abgegebene Leistung deutlich über der vom Hoymiles (-600) angezeigten liegt, habe ich hier natürlich auch, aber nicht permanent, sondern stark von der angeforderten Leistung und der aktuellen Klemmenspannung abhängig. 
Hier äußert sich dies durch, ich nenne es mal "Regelschwingen". Über die Rückkopplung, der realen Verbrauchswerte vom Zähler, wird ja der WR immer wieder im Output begrenzet und die Trägheit der Regelung bewirkt ein ständiges auf/ab der abgegebenen Leistung, aber wie gesagt, nicht in allen Betriebszuständen. Aber z.T mit deutlichen Überschreiten der max Leistung (+30%) des WR an einer nachgeschalteten Messstelle. Wobei mir aufgefallen ist, dass dieser Effekt kleiner geworden ist, seit vor einiger Zeit die Steuerung auf einen andern Kanal (1 -> 0?) umgestellt wurde.
Das Verhalten selbst ist aber so unbestimmbar, dass die Erstellung/Pflege einer "Korrekturtabelle" in meinen Augen nicht zielführend sein kann. Ich verwende daher die DTU für die Momentananzeige und die (geeichten?) Tasmota-Shellys für die Langzeitdaten.

Mehr Sorge macht mir da eher die Tatsache, dass der Code ohnehin immer umfangreicher wird...

Joe6899, 2024-02-21:
I agree on the complexity issue. it seems that  the pre take off check sequence for an airbus 320 is probably shorter and simpler than setting up a OnDTU  battery.     I also use a shelly for power measurement.   From my estimates so far  I donated 3% for free to the energy company.  Woorse case control scenario is the washing machine because it hovers around a 500W limit , I suggested  to the wife,  that we  switch to a wash mangle - but this didn't go down well.   To minimize  energy donations it makes sense to limit max power  to the regulated limit  (<470W) during the winter months and increase back to 600W  in the summer months..   One way to  automatically improve the losses would be implement  additional   hysteresis level for power levels above 470W when 24V battery are used.  But gains are really small.

DDvO, 2024-04-09:
Thanks a lot @Joe6899 for all your comments here!

Your below [analysis of Jan 25](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/613#issuecomment-1910235092) and also your further thoughts and remarks are outstanding.

I was pointed to this thread [just yesterday](https://github.com/tbnobody/OpenDTU/discussions/1078#discussioncomment-9050738) and am delighted to finally have got a to-the-point explanation of that phenomenon. I had came across it about two of months back, too, when measuring the real (not dreamed-up) efficiency of my HM-300 when used with a 24 V battery depending the (relative) output limit setting, [reported on the PV Forum ](https://www.photovoltaikforum.com/thread/221194-hm-400-an-batterie-limitierung-%C3%BCber-opendtu-eigenartig/?postID=3660691#post3660691) in a thread on the same subject.

> I have also observed the same behavior for dual 300W inverter inputs on 600W inverter.
> 
> I believe the reason why the output power appears to be limited to 500W is due to the way the input DC current sensor is designed. Basically at an input current >~10A it reaches a saturation point. No matter ho much more current is drawn, the measured input current value stays the same.
> 
> The DC input current is measured using a LS shunt resistor. The shunt resistor has been set to provide maximum accuracy over the intended operating range. Intended operating range is 35V PV panel. So this means that when peak output power is delivered, the maximum DC input current will always be less than 10A. This is also true at low light conditions - low input PV voltage is seen at low power.
> 
> When using a 24V battery the peak output power results in DC input current more than 10A. The input DC current reading is no longer correct. But importantly the control loop on the output power stage still limits it to the max rated output power - 600W. You really do achieve the max rated power even at 24V. The problem now is that if you used the "measured" input current together with the actual output power, you would get the magical situation of a an efficiency level or more than 100%. To counter this, I believe the software guys at hoymiles have been a bit clever. They cannot know the actual input current for currents >10A. But to make the efficiency seem plausible, they have limited the declared output power with 10A current based on an efficiency of ~95.5%.
> 
> On the discussion on whether it is better to use 24V or 48V. 24V clearly will cause measurement issues at max power, but I think it does have some reliability advantages.
> 
> The input electrolytic capacitors on the hoymiles inverters are rated to 65V. Electrolytic capacitor lifetime is strongly affected by voltage. Rule of thumb - half the voltage doubles the lifetime. The next issue is the max transient input voltage. e.g assume inverter is running at max power and BMS/switch/ or relay is opened. In this instance inductive kick will result in far higher voltage. It is plausible to vastly exceed the 65V rating. So if 48V is used I would be careful not to disconnect the input voltage when under output load.

DDvO, 2024-04-20:
In the light of the outcome of this exchange, I've just re-phrased the description what goes wrong under which circumstances and why this is so at https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/24V-Battery-issues-with-Hoymiles

Also here, further improvements and any useful extensions are welcome.

tehzap, 2024-06-27:
I had a similar problem initially when operating both mppts on a 24V battery. I observed different power input on both channels, although obviously the same is expected. Also the AC power output was fluctuating between the real and an odd number. The real output was measured by a Shelly in series. What solved the issue for me was connecting the negative channels using a Y cable instead of leading two separate negative cables to my busbar. Since then I got identical power output and no more power fluctuations in openDTU.

Manos1966, 2024-06-27:
Thanks for sharing your experience @tehzap it went straight to Wiki 👍 https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/24V-Battery-issues-with-Hoymiles

DDvO, 2024-07-02:
> I had a similar problem initially when operating both mppts on a 24V battery. I observed different power input on both channels, although obviously the same is expected. Also the AC power output was fluctuating between the real and an odd number. The real output was measured by a Shelly in series, which showed that actually only the measurement was fluctuating but not the real power output. What solved the issue for me was connecting the negative channels using a Y cable instead of leading two separate negative cables to my busbar. Since then I got identical power output and no more power fluctuations in openDTU.

Pretty weird.
How can it make a noticeable difference whether you connect the two negative input plugs via a Y cable rather than via two regular cables that are connected a little further away via a busbar?
All these cables have essentially zero impedance, so in both cases you effectively short-circuit the negative input connectors.

tehzap, 2024-07-03:
> Pretty weird. How can it make a noticeable difference whether you connect the two negative input plugs via a Y cable rather than via two regular cables that are connected a little further away via a [...]

Tbh, I have no clue. I just interpreted it such that the inverter is really sensitive to slightly varying resistance. 
Here you can see the difference in channel power output before and after the cable change.

[image]

spcqike, 2024-07-03:
> I just interpreted it such that the inverter is really sensitive to slightly varying resistance.

it indeed is. as the DC(-) ports are connected, it is important that all DC(-) connections have the same resistance. from the inverter to the battery. if one cable is slightly longer, if its missplaced at the busbar, if the connection is loose, ... whatsoever, the resistance is slightly higher. 

the inverter measures the dc current in the negative path. so if one connection has higher resistance, more current will flow through the other connection and therefore it measures more current. even tho thats not the "real" power that channel delivers, its measured as such.

Squint5471, 2025-04-29:
> it indeed is. as the DC(-) ports are connected, it is important that all DC(-) connections have the same resistance. from the inverter to the battery. if one cable is slightly longer, if its [...]

I also observed this issue when using extension cables in 2 of 4 inputs of the hms1800:
- 2 inputs with a y cable to the battery
- the other 2 inputs with a y cable and a 1m extension to the battery

bensch2581, 2025-06-10:
Hallo zusammen,
ich bin eher neu im Thema und habe die Diskussion mit Interesse gelesen. Ich habe bei mir mit einem HM800 sehr ähnliche Beobachtungen an einem 24V-System gemacht - und mich dafür entschieden, ein Wechselrichterlimit auf 400W (max.) einzustellen.
Ich habe meinen Werte aus "empirischen Beobachtungen" abgeleitet. ;-)
Fur eine einwandfreie Funktion des DPL ist es m.E. unerlässlich, dass der Hoymiles verlässlich das einspeist, was man ihm vorgibt?

1.: Wäre es also nicht viel mehr zielführend, ein weiteres "Betriebsprofil" neben "Batterie" und "Solar" wie bspw. "Batterie (24V)" zu hinterlegen, in dem die maximale Leistung auf 50% je Kanal begrenzt wird (natürlich mit einer kleinen Erklärung, warum das Ganze...) Das hätte mir einiges an Lese-Zeit erspart - und dazu gehört natürlich die Akzeptanz, dass ich meinen HM800 (den ich eh außerhalb des vorgesehenen Szenarios an einer Batterie betreibe) eben nur sinnvoll bis ~400W mit dem DPL betreiben kann. ;-) 

2.: By the way: funktioniert der DPL auch sinnvoll mit mehreren WR? Sprich könnte ich zum Skalieren der Einspeiseleistung einen oder mehrere weitere HM-XXX  dranhängen?

spcqike, 2025-06-10:
Zu 1. 
nein. Wozu sollte man das fest vorgeben? Der nächste nimmt eine 36V Batterie und möchte noch ein 36V Profil? Man kann alles vorgeben und einstellen. Das reicht. Beschrieben ist es ebenfalls. 
https://github.com/hoylabs/OpenDTU-OnBattery/wiki/24V-Battery-issues-with-Hoymiles

Dafür gibt es Anleitungen, Wikis und co.

Zu 2.
Ja das geht.

hoschiking, 2025-06-10:
Ich frage mich, ob für 24V Systeme der HMS 400/800 nicht die bessere Alternative wäre? 
Laut Datenblatt soll dieser am Eingang einen höheren Strom vertragen. Die Frage ist nur noch, ob es Erfahrungswerte gibt, wie sich dieser oberhalb von 10A verträgt. Ich habe irgendwo Mal was gelesen, aber sicher bin ich mir nicht.
Gibt es fundierte Erfahrungswerte? 
Lässt sich der bis 14A vernünftig regeln? 
Wer kann das was beitragen?

spcqike, 2025-06-12:
von dem was ich grade finde, nein. der HMS-800 geht auch nur bis 12.5A

[image]

der HMS-1000 kann bis 14A. (sofern das noch das aktuelle Datenblatt ist, was ich da gefunden habe)

ich habs selber nicht getestet, da ich keine Batterie habe, aber ich würde erst einmal davon ausgehen, dass die Inverter bis zu ihrem maximalen Eingangsstrom sauber regeln können.

Dennoch tut man sich mit 24V langfristig keinen Gefallen. Wenn man etwas neu machen oder umstellen will, dann würd ich den Speicher erhöhen.

hoschiking, 2025-06-12:
Ich habe mir für Testzwecke Mal einen HMS 400 über Kleinanzeigen gekauft, der kann laut Aufkleber 14A am Eingang. Da wären also theoretisch bis zu 350 Watt möglich also nicht weit vom Maximum. Auf die 50 Watt kann man verzichten. Testen kann ich leider erst in einigen Wochen, erstmal ist Urlaub angesagt.

[image]

DDvO, 2025-06-12:
> Zu 1. nein. Wozu sollte man das fest vorgeben? Der nächste nimmt eine 36V Batterie und möchte noch ein 36V Profil? Man kann alles vorgeben und einstellen. Das reicht. Beschrieben ist es ebenfalls. https://github.com/hoylabs/OpenDTU-OnBattery/wiki/24V-Battery-issues-with-Hoymiles

Einen speziellen 24 V Modus finde ich nicht lohnenswert, aber wenn ich mich recht erinnere, gibt es eine Einstellung, um den Ausgangsstrom (oder äquivalent die Leistung) zu begrenzten.

hoschiking, 2025-06-12:
Klingt gut, leider ist der Preis nach deiner Werbung auf 30€ gestiegen ;-)

hoschiking, 2025-07-02:
So, ich bin zurück aus dem Urlaub, gesamtes Material für die ersten Versuche ist da.
Allerdings habe ich Verbindungprobleme mit dem HMS-400-1T.
Erstmal hat es auf Anhieb funktioniert und jetzt nach einem Neustart wegen Testaufbau gar nicht mehr -:(
Ich kann nur vorab sagen, die ersten Ergebnisse waren sehr vielversprechend.
Dateils später, wenn der Versuchsbaufbau läuft

[2 more replies]