hoylabs/OpenDTU-OnBattery #615: [Request] oder [Bug]? Blockiert ein ausgeschaltetes Display die Kommunikation zu anderen Komponenten?

dragricola, 2024-01-17:
### Is your feature request related to a problem? Please describe.

Ich möchte gerne die Versorgungsspannung meines Oled-Displays ausschalten, wenn es nicht gebraucht wird. Bei unterbrochener Vesorgung wird offensichtlich jedoch die Kommunikation zum Hoymiles Inverter und zum Huawei AC-Charger gestört: Zwischen den erfolgreichen Datenübertragungen liegen dann oft mehrere Minuten.  Bei angeschaltetem Display werden die Daten wieder ca. alle 2 bis 5 Sekunden übertragen. Vermutlich blockiert die nicht mögliche Datenübertragung zum Display die Datenübertragung zu den anderen Komponenten. Ist das vom Programmablauf erklärbar?

### Describe the solution you'd like

Gegebenenfalls sollte der Programmablauf so geändert werden, dass die Kommunikation auch durch ein abgeschaltetes Display nicht unterbrochen wird und die Abfragen zum Hoymiles Inverter und zum Huawei AC-Charger im gleichen Rhythmus wie mit Display möglich sind.

Tiese, 2024-01-18:
Das habe ich auch vor!
Wenn ich den Stecker vom Display abziehe läuft alles wie gewohnt &#128512;

Beim Trennen der Stromversorgung: "also nur den +Pin" soll sich das anders verhalten?

dragricola, 2024-01-18:
> Wenn ich den Stecker vom Display abziehe läuft alles wie gewohnt 😀

Nein, bei mir leider nicht: Wenn ich den +Pin abziehe, dann stockt sofort die Kommunikation und wenn ich ihn wieder anstecke, dann läuft es sofort wieder wie am Schnürchen! Ich habe es gerade eben nochmals probiert und es ist reproduzierbar. Ich könnte es notfalls auch mit einem Video zeigen.

Ich habe übrigens ein 1,3" OLED Display 1,3 Zoll 4 Pin mit Treiber chip sh1106 und Auflösung 128 × 64. Es ist konfiguriert mit Display Type 3.

Tiese, 2024-01-18:
Ich habe dasselbe Display und kann ebenfalls bestätigen, dass **nur** wenn der **+Pin** getrennt wird, **KEINE Aktualisierung** mehr stattfindet - sogar die **Konsole** bleibt **LEER**!

schlimmchen, 2024-12-01:
Mir gefällt dieses Issue nicht, weil hier im Grund danach gefragt wird, ein Hardwareproblem in Software zu flicken.

Vorstellbar ist es, das Display ausschaltbar zu machen. Das wäre ein Feature dessen Sinn ich verstehe. Ausschaltbar meint hier allerdings, ein schwarzes Bild zu schicken. Die Hintergrundbeleuchtung wäre dann trotzdem noch an. Das wäre dann ein signifikantes Feature, denn das Ausschalten muss irgendwie (ggf. über mehrere Wege) steuerbar sein.

Ich gehe davon aus, dass ihr jeweils ein SPI Display einsetzt? Oder passiert das auch bei Einsatz eines I2C Displays? Falls es en SPI Display ist, wünsche ich mir, dass ihr das einmal versucht das Problem zu reproduzieren mit dem aktuellen Release. Denn der SPI Port Manager und die Arbeiten dazu machen mir Hoffnung, dass das Problem damit schon behoben ist.

Andernfalls... Warum müsst ihr das Display ausschalten? Wenn ihr ein OpenDTU-OnBattery Status-Display nutzt neben eurem Bett und das ist gleichzeitig aber der maßgebliche ESP32, der die Kommunikation mit den Invertern übernimmt und vielleicht auch den DPL, dann finde ich das einen merkwürdigen Ansatz. Stattdessen wäre dann eins der Side-Projects, die genau das machen (OpenDTU-OnBattery Statusdisplay), sinnvoll?

dragricola, 2024-12-02:
Das Problem trat auf, wenn man den + Pin abtrennt. Inzwischen habe ich herausgefunden, dass keine Störung der Kommunikation auftritt, wenn man das Display mit einem Schalter in der GND-Zuleitung statt in der VCC-Zuleitung ausschaltet. Damit ist auch die Hintergrundbeleuchtung aus.  

Du kannst dieses Issue daher gerne schließen. Das Problem lässt sich hardwareseitig lösen.

> Ich gehe davon aus, dass ihr jeweils ein SPI Display einsetzt?

Nein, ich setzte das sh1106 mit I2C Schnittstelle ein. Es hat 4 Pins.

>Warum müsst ihr das Display ausschalten? 

Meine Batterie steht im Keller, da brauche ich das Display nur sporadisch und will mit dem Schalter nur verhindern, dass sich das OLED einbrennt. Inzwischen ist die Gefahr des Einbrennens ja aber auch durch die Pixel-Shift-Funktion abgemildert. 

>Vorstellbar ist es, das Display ausschaltbar zu machen. Das wäre ein Feature dessen Sinn ich verstehe. 

Mit AHOY kann man übrigens die DPU so konfigurieren, dass sich das Display nur anschaltet, wenn sich jemand in der Nähe der DTU befindet. Die Anwesenheit wird durch einen PIR-Sensor detektiert. Das funktioniert sehr gut!

Sorbat, 2024-12-02:
Wie kann man sich das hardwaretechnisch vorstellen mit dem Schalter? Hast du den ins dem Gehäuse platziert? Gibt es vielleicht ein Foto davon?

schlimmchen, 2024-12-02:
> Inzwischen habe ich herausgefunden, dass keine Störung der Kommunikation auftritt, wenn man das Display mit einem Schalter in der GND-Zuleitung statt in der VCC-Zuleitung ausschaltet.

Ah, ja das macht Sinn. I2C braucht jeweils einen Pull-Up an der Takt- und Datenleitung. Außerdem müssen die Leitungen im Idle logisch high sein. Wenn du VCC von deinem Display trennst, ist womöglich eine oder beide der Leitungen ohne Pull-Up (wobei die vielleicht auch im ESP eingeschaltet sind) oder sie werden mit Masse verbunden. Das ist dann ein illegaler Zustand des Busses und damit kommt die Hardware und Software nicht so gut klar. Dass aber, wenn die Leitungen "funktionieren" niemand antwortet, ist sicherlich in Software gut abgefangen, weshalb es dann weiterhin gut funktioniert. Und wenn du die Masse vom Display trennst, kann das Display-Modul auch nicht die Datenleitungen mit Masse kurzschließen. Macht für mich Sinn.

Jedenfalls ist das in der Tat in Hardware zu lösen.

Danke für deine Rückmeldung!

dragricola, 2024-12-02:
@Sorbat : Display und Schalter sind in die Frontplatte meines Batterieschranks eingelassen (Bild 1).
Der Schalter verbindet den GND-Pin des Displays mit GND des 5V-Netzteils. Die DTU liegt im Schrank zusammen mit den Invertern, dem Smart-Shunt, dem Huawei AC-Charger, 5V-Netzteil, den Schaltaktoren für den AC-Anschluss sowie der Temperaturregelung mit den Lüftern (Bild 2). Die Verdrahtung wird noch etwas geordnet, wenn alle Tests abgeschlossen sind und ich nichts mehr verändern will.
Bild 1:
[image]
Bild 2:
[image]

@schlimmchen : Danke, das erklärt alles!