hoylabs/OpenDTU-OnBattery #617: VE.Direct wo aktivieren

eeprom23, 2024-01-21:
### What happened?

Wo kann ich das auslesen eine MPPT von Victron activieren ? rx und tx habe ich am esp verbunden über pin 21 und 22. jedoch finde ich keine option das auslesen zu starten !

### To Reproduce Bug

[image]

### Expected Behavior

habe den esp32 schon mehrfach neu geflashed und auch eine neuere version genutzt. edoch finde ich die option nicht.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

7492acd

### Relevant log/trace output

```Shell
-
```

### Anything else?

-

schlimmchen, 2024-01-21:
In der Web-App unter "Einstellungen" "VE.Direct" wählen und einschalten?!

Aber dein Pin-Mapping sieht noch falsch aus. Welches Hardware-Profil/Pin-Mapping hast du denn ausgewählt im Drop-Down-Menü? Den Teil sieht man nicht im Screenshot.

eeprom23, 2024-01-21:
diese einstellung gibt es leider nicht !
[image]
[image]

Manos1966, 2024-01-21:
Eine Profile Datei fehlt...
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Device-Profiles

Solarteur, 2024-01-21:
Ist da überhaupt OpenDTU-**OnBattery** installiert? Bei mir sieht das Menu unter "Hardware" so aus:

[image]

schlimmchen, 2024-01-21:
Er kommt wahrscheinlich von OpenDTU (nicht OpenDTU-OnBattery) und muss die WebApp aktualisieren.

@eeprom23 Bitte drücke Strg+F5 in deinem Browser, falls meine Vermutung stimmt, dass du zuvor das Upstream-Projekt "OpenDTU" installiert hattest auf diesem ESP32.

Manos1966, 2024-01-21:
@Solarteur 🤣🤣🤣🤣🤣🤣🤣🤣 👍 
Oh Man, ich muss zu Fielmann....
[image]

eeprom23, 2024-01-21:
Vielen Dank, ich bin irgenwie auf die normale Open DTu gekommen ! Danke für eure schnell Hilfe !!

schlimmchen, 2024-01-21:
@eeprom23 Bitte schließe das Issue als erledigt.

closed 2024-01-21 by eeprom23