hoylabs/OpenDTU-OnBattery #624: Bug

Koller417, 2024-01-26:
### What happened?

Fehlermeldung nach Update

### To Reproduce Bug

[HTTP ERROR] Bad Request

### Expected Behavior

Neustart

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.01.26

helgeerbe, 2024-01-26:
Kommt da noch was? [HTTP ERROR] Bad Request kann alles sein. Kein Strom. Netzwerkprobleme.... 
Wie hast du das Update durchgeführt? OTA oder über Kabel. Was passiert, wenn du den ESP  neu startest (stromlos machen, oder Knopf drücken)? 

So kann ich dir nicht helfen. Ausser sagen, dass es bei mir auf 3 ESPs läuft.

Koller417, 2024-01-26:
Nein. Update wird über OTA ausgeführt.Nach Neustart ist die alte Version noch drauf.

Solarteur, 2024-01-26:
Gerade bei mir über OTA auf 2024.01.26 aktualisert - keinerlei Probleme.

schlimmchen, 2024-01-26:
Klingt nach "Binary zu groß", also veraltetes Partitionsschema oder die Firmwarr ist für einen bestimmten ESP32 tatsächlich übergelaufen. Aber das würde wohl beim Build auffallen. Welchen ESP32 hast du genau? Welche Firmware-Datei nimmst du?

Bitte Konfig sichern, die passende factory-Bibary über die serielle Schnittstelle programmieren und Konfig wiederherstellen.

Koller417, 2024-01-26:
Auch über Kabel nicht instalierbar. Version 2024.01.16 läst sich über Kabel und OTA instalieren.

Koller417, 2024-01-26:
ESP32 WROOM 32.
generic_32.bin

Koller417, 2024-01-27:
Version 24.1.21 wird über OTA installiert.

Tiese, 2024-01-28:
Bei mir kam die gleiche Fehlermeldung!
Ich habe auf **_wiederholen_** geklickt, dann ging es reibungslos &#128515;

schlimmchen, 2024-02-10:
@Koller417 Dann hattest du also Erfolg und dieses Issue kann geschlossen werden?

closed 2024-03-18 by schlimmchen