hoylabs/OpenDTU-OnBattery #626: [Request] Make Inverter Shutdown configurable

andreash-esp, 2024-01-28:
### Is your feature request related to a problem? Please describe.

I have two Microinverters on two OpenDTU-onBattery. Both are set for -10W Grid Power with a single Shelly 3EM. No Battery attached.
This works fine most of the time, but sometimes one of the Microinverters detects to much Power delivered to the grid and shuts down. This happens when the other one sends to much power.

I'd like to prevent this shutdown. My goal is to limit Power to the Grid to 10W, but I really do not care if it is more than that for a short time.

### Describe the solution you'd like

Have setting to disable the shutdown of the inverter when the Power Limit is below configured Power Limit.

I guess this should be right at the start of "bool PowerLimiterClass::setNewPowerLimit" 

### Describe alternatives you've considered

A (configurable) timeout limit for the Inverter shutdown might have the same result, i.e. "only shutdown if power is below the powerlimit for a certain time and by a certain amount"

schlimmchen, 2024-01-28:
>I'd like to prevent this shutdown.

Why?

andreash-esp, 2024-01-28:
Because it takes some time for the inverter to wake up when it is needed again.

DaBaBW, 2024-02-10:
There is a lower power limit configurable in DPL settings. I thought this is exactly for your case and the inverter only goes down to this value and does not shut off.

spcqike, 2024-02-10:
@DaBaBW no. Lower limit is a minimum that the inverter needs to produce before you start it. 

Like if you set it to 50W, openDTU onBattery won’t start the inverter if you only need 30W.

DaBaBW, 2024-02-10:
Ok, so I misunderstood this function. Then there is missing a switch to not turn off the inverter and stay at lower limit.

Janorei, 2024-03-05:
#179

schlimmchen, 2024-03-09:
@andreash-esp Schau mal in #733 vorbei. Ich tendiere dazu diesen Request als erfüllt zu deklariere, wenn #733 veröffentlicht wurde. Das würde auch dein Problem lösen, korrekt? Der Schalter lautet nicht "Verhindere Shutdown", sondern er heißt "Inverter wird durch Solarpanele gespeist". Der bewirkt aber insbesondere auch, dass die Leistungsproduktion nicht ausgeschaltet wird, wenn das untere Leistungslimit unterschritten wird.

andreash-esp, 2024-03-10:
Hi, das würde mein Problem lösen, denke ich. Vielen Dank!

closed 2024-03-10 by schlimmchen