hoylabs/OpenDTU-OnBattery #628: Update OpenDTU-OnBattery/docs /hardware_flash.md

Manos1966, 2024-01-29:
### Is your feature request related to a problem? Please describe.

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/docs/hardware_flash.md#mcp2515-can-bus-module

says:

MCP2515 CAN bus modules that are widely available are designed for 5V supply voltage. To make them work with 3.3V / the ESP32 a modification is required. [This modification is described here.](https://forums.raspberrypi.com/viewtopic.php?t=141052)

### Describe the solution you'd like

Add to the text,  that there are ready modules available:

MCP2515 CAN bus modules that are widely available are designed for 5V supply voltage. To make them work with 3.3V / the ESP32 a modification is required. [This modification is described here.](https://forums.raspberrypi.com/viewtopic.php?t=141052)
If the above is too complicated, a (more expensive) version of this with two voltage inputs is also available, the Joy-IT SBC-CAN01 which requires no modifications. [CANbus communication with OpenDTU](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Huawei-AC-PSU#canbus-communication-with-opendtu)

schlimmchen, 2024-02-10:
Copy the hardware_flash.md to a new and appropriate Wiki page and let me know. I'll then create a MR to update that file to point to the wiki instead.

Manos1966, 2024-02-14:
Hi @schlimmchen Bernhard,

es ist hier https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Huawei-AC-PSU#canbus-communication-with-opendtu

commit 7d4a30dd (2024-03-19): hardware_flash.md: move some info to wiki

closes #628
closed 2024-03-19 by schlimmchen