hoylabs/OpenDTU-OnBattery #633: Pylontech not included in HomeAssistant MQTT Autodiscovery

meisterlampe, 2024-02-02:
### What happened?

My Pylontech-Battery is not discovered automatically within Home-Assistant.
I checked with MQTT-Explorer and can see the current battery-values published to the topic `solar/battery`. But the Home-Assistant-Autodiscovery-Information expected in the topic `homeassistant/sensor/dtu_battery_0001` is missing.

### To Reproduce Bug

The bug appears on a fresh openDTU-onBattery-System with MQTT and Pylontech-Battery configured

### Expected Behavior

If Home Assistant MQTT Auto Discovery is enabled. A topic `homeassistant/sensor/dtu_battery_0001` should be created, that contains the information needed by Home Assistant.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

2024.01.26

### Anything else?

I'm not that into C++, but I checked the class `MqttHandlePylontechHass` and for me it seems, that it is never called anywhere.
So I just added it to the main.cpp and Auto-Discovery startet working again. ( https://github.com/helgeerbe/OpenDTU-OnBattery/compare/development...meisterlampe:OpenDTU-OnBattery:addMqttHandlePylontechHassToMain ).

The name of the class `MqttHandlePylontechHass` indicates, that simply adding it to main.cpp is not intended. I guess it is not compatible with the other interface options or is this class used for all interfaces?

madmartin, 2024-02-03:
I propose a "git bisect" to find the commit that broke that feature.

NikGro, 2024-02-07:
I can confirm the issue as well as the fix proposed by @meisterlampe

meisterlampe, 2024-02-10:
I tested the new version and can confirm, that the bug is fixed. Thanks a lot! 👍

closed 2024-02-10 by meisterlampe