hoylabs/OpenDTU-OnBattery #641: Add new SmartSolar MPPT 150/45 Rev3

hhoebel, 2024-02-07:
### Is your feature request related to a problem? Please describe.

I'm owner of a brand new Victron SmartSolar MPPT 150/45 which isn't recognized by OpenDTU-OnBattery (reported as "???")

### Describe the solution you'd like

The console reports the PID as 0xA073, so i suggest to add the line
{ 0xA073, "SmartSolar MPPT 150|45 rev3" },
to VeDirectFrameHandler.cpp

schlimmchen, 2024-02-07:
Why "rev 3"? There already is a "SmartSolar MPPT 150|45" but yours is revision three and that has a new PID?

Can you find a canonical source for the PIDs? This would be cause to update the whole list, not just add one particular product.

Implementing this is straight-forward, you already found the right spot.

hhoebel, 2024-02-07:
Sorry, nein, ich hatte keine Quelle dafür und nur vermutet, das die neuere PID dann Rev3 bedeutet. Ich habe nach deinem Post aber nochmal gesucht und tatsächlich was in der Doku des VE.Direct-Hex-Protokolls bei Victron gefunden - die scheint ziemlich aktuell zu sein:
https://www.victronenergy.com/upload/documents/BlueSolar-HEX-protocol.pdf (ab Seite 4). 
Danke übrigens für OpenDTU OnBattery :-)

PR #643 (2024-02-09): Update VE.Direct product IDs
Fixes typos and errors, and adds previously unknown PIDs.

Closes #641.
commit 7c84621e (2024-02-09): update VE.Direct product IDsfixes typos and errors, and adds previously unknown PIDs.closes #641.

* update VE.Direct product IDs

fixes typos and errors, and adds previously unknown PIDs.

closes #641.
closed 2024-02-09 by helgeerbe