hoylabs/OpenDTU-OnBattery #647: Inverter stays in onState even if MQTT-cmd for disable is active

hhoebel, 2024-02-08:
### What happened?

I'm using the MQTT /cmd/mode=1 to disable the Power-Limiter from within HomeAssistant (e.g. for maintenace-reasons, at undervoltage of the battery and when the battery has been to long not been fully charged). Whenever the inverter gets disconnected and reconnected from/to DC, the inverter starts producing power and never switches off (but the LiveData of OpenDTU shows a limit of 0%). It seems that the Power-Limiter switches off the inverter only after receiving the first MQTT-cmd and doesn't repeat that after the inverter was offline. 
Additionally bug (but with a very similar reason): whenever the Power-Limiter gets re-configured over the Web-API during MQTT-disable is active, the new Power-Limiter-State is set, regardless of the disable. This is because the MQTT-cmd sets the _mode in the Power-Limiter-Class only once, after the cmd has been received and any later re-configuration sets the _mode to the new configured mode, which overrides the previously disable-cmd. As workaround the permanent re-sending of the MQTT-cmd helps keeping the disabeled state of the Power-Limiter. But this doesn't help to solve the behavioud after inverter-restart (after DC disconnect), like described before.
Suggestion: save the MQTT-cmd-mode in a separate variable (not _mode) to keep it active even at reconfig and check permanently if the inverter is still off, when disabled.

### To Reproduce Bug

Main bug (power-cycle of inverter):
1.) Write MQTT /cmd/mode = 1 (disable of Power-Limiter)
2.) OpenDTU switches the inverter off
3.) Power-Cycle the inverter (disconnect/connect DC)
4.) Inverter stays on forever and produces power, even if the MQTT-cmd gets set newly

Secondary bug (changing config of Power-Limiter using Web-API):
1.) Write MQTT /cmd/mode = 1 (disable of Power-Limiter)
2.) OpenDTU switches the inverter off
3.) Re-configure the Power-Limiter and press "save"
4.) Power-Limiter gets active again (if enabled in config) regardsless of the MQTT-disable

### Expected Behavior

The MQTT-cmd-mode should work until the mode has been changed again via MQTT.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.01.26

schlimmchen, 2024-02-08:
I see you took good care to write a quality bug report, thanks. Still, I want to nag that you put two issues into one :speak_no_evil: 

1. You are right. The DPL calls shutdown(), and once the inverter was shut down, the pointer to the inverter is reset to nullptr. The DPL does indeed call shutdown repeatedly if `_mode` is `Mode::Disabled` (what you know as 1 in the MQTT topic), but the shutdown function assumes the inverter is down (as the pointer is nullptr) and does nothing. I gues fixing this is simple: Moving a couple of lines of code further down the DPL `loop()` implementation. However, that will fix your issue, but might very well cause an issue for other people who use mode 1 differently. You might want to disable the DPL and send one shutdown to the inverter, so you can freely modify its limit through MQTT while the DPL is in Mode 1. So I think that this is not a bug, this mode is working as intended.
The Wiki says: "The inverter is shut down and the Power Limiter stops operating, as if it was disabled in the web application." And this is exactly how this mode is implemented.
You are asking for a mode where the DPL is NOT fully disabled, instead it governs the inverter to be shut down repeatedly.
2. `PowerLimiter.setMode(PowerLimiterClass::Mode::Normal);  // User input sets PL to normal operation` This is also by design. The assumption is that if you use the Web UI to change the DPL settings, the DPL should work as configured. Again, if this was to change, other people would notice a regression.

To sum up, I have to say that I deem your bug report invalid.

For 1, I argue that disabling the DPL using Mode::Disable means you have to manage the Inverter yourself, as the DPL is indeed off.

For 2, I added a line in the [Wiki](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT) to clarify the behavior.

hhoebel, 2024-02-09:
Haha, yes, next time I promise to do better (one ticket per bug).
Thanks for the detailed explanation. I didn't have this use case in mind, but it makes sense and explains the behavior, which was quite unexpected for me.
Maybe you'll consider adding an additional mode to /cmd/mode at some point: "3 - Maintenance: The inverter remains switched off". For now i'll change my HA-script to get what i need for my use-cases.

schlimmchen, 2024-02-09:
Mode 3 is already in planning for #583 :grin: 

I am not happy about implementing yet another special mode, especially this one. How should MQTT messages changing the inverter limit or enabling it be handled if the DPL is in this new mode? Or user input in the web application?

If you are doing maintenance, you should physically disconnect the inverter. There should be a physical switch (resetable fuse) for that in your setup :wink:

hhoebel, 2024-02-09:
Yes, this will raise new questions depending on the use case. My intention was just to prevent the inverter from starting again. I now understand that this is actually outside the scope of OpenDTU because it starts automatically after DC disconnection. In my setup the inverter is connected to the battery via a charging resistor and a fuse. The resistance is permanently in the branch and is bridged by the fuse after pre-charging. I know there is a better way to do this, but I have 6 inverter inputs on the battery, which increases the effort by a factor of 6. I don't want to make the AC side switchable, because it's 3-phase and can actually be controlled via software (that was my plan). But you're right - I should add a circuit breaker for the inverter.
Nevertheless, there remains a use case for the maintenance mode on the software side - in winter there are always cases where the inverter should remain off until the battery is fully charged so that it can occasionally reach 100%. Then I at least don't want to accidentally restart the inverter - this leaves the need to deactivate the inverter via software, which with 6 inputs and 3 phases is at least not possible without effort via hardware (Shelly + relay, etc.)

schlimmchen, 2024-02-09:
In #221 we are currently discussing a new software layer, which I think will be called something like "battery protection". The idea is to shut the inverter down (and keep it shut down) and to also prevent enabling it through MQTT or the web application. That new software layer is designed to prevent the BMS from kicking in when discharging. A new voltage threshold will be used to determine whether or not that battery protection is active.

The same software layer could also be engaged through other means, i.e., for other purposes. Like "force a full charge" or "winter shutdown mode" or something. That would personally interest me. And that should help you as well, I guess? How would you need to be able to control this new mechanism?

[However, I will repeat, that you should be able to disconnect your inverter from the battery with a fuse/switch, at least when doing hardware maintenance. In my case the pre-charge switch and resistor are in parallel with another fuse. The latter fuse connect the positive and negative of the battery to a screw terminal. In that screw terminal, all inputs of the inverter are attached to. It is my opinion that this is the only reasonable way to connect the inverter(s) with the battery.]

hhoebel, 2024-02-09:
Yes, such a software layer would be very interesting. That's exactly what I'm currently doing via HA - prevent further discharge in the event of undervoltage without using the BMS shutdown. And in addition, the inverter can also be switched off for an adjustable period of time after the last full charge until it is fully charged again. Furthermore - as you mentioned - optional shutdown during certain calendar periods.
I would need the following settings:
1.) Switch-off threshold in V (e.g. 50V)
2.) Restart threshold in V (e.g. 51V) OR hysteresis for 1.)
3.) Maximum time (days) before a full charge is forced
4.) SOC or voltage value considered as a full charge

I have currently implemented 3.) and 4.) by saving the date and time when the 4.) threshold is reached. Then I compare 3.) with this saved date and deactivate the inverter if the period has been exceeded.

It's great that something like this is already being planned - I wouldn't have submitted a suggestion for improvement because you probably already have enough other topics.

BTW: i can disconnect the battery from the system (with build-in switches (via BMS) of the batteries and with a fused circuit breaker (MCCB)), but not additionally the inverter from the whole system. The inverter is always connected via bus bars with the Victron. I'm not sure if i need the precharging resistors at all (those are my main problem when the inverter restarts directly after maintenance, because they are getting *really* hot when the inverter starts drawing current. This is why i want the inverter to be off until i give it the "go" to restart), because the batteries do have a soft-start, but i don't know the exact ramp-up of this startup. Seems as it would be a good idea to fetch the oscilloscope, insert a shunt and take a look at this startup. Without the resistors, it wouldn't be a problem, if the inverter directly starts after re-conenction to DC. My first attempt with a automatically working Mosfet-Current-Limiter exploded (not really - only smoke) , so i returned to the resistors  :-(
But it's a comfort problem. I simply can bridge the resistors with the fuses to avoid their heat-up (but i forgot that once due to the assumption that inverter would stay off) and i could disconnect the inverter via the 3-phase-circuit-breaker in the distribution box. The system itself is safe in a way than it's completly fused (inputs of inverter, input of victron, output of battery, output of inverter) - it's just the idiot in front of the system not careing about the basic electrical rules (switch off and secure against being switched on again).

schlimmchen, 2024-02-09:
> It's great that something like this is already being planned

That's pure coincidence, but yes, we were discussing this in #221 just a day ago.

> I wouldn't have submitted a suggestion for improvement because you probably already have enough other topics.

Never mind. With or without issues being opened I have far more ideas than time to implement them :wink: 

> because they are getting really hot when the inverter starts drawing current.

Wait, what?! What are you doing? You are supposed to **pre**-charge the inverter after DC power off through the resistor and then dis-engange the resistor after connecting the inverter DC to the battery/power source.

> I simply can bridge the resistors with the fuses to avoid their heat-up

Well, there you say it: The pre-charge resistor shall only be engaged for a few seconds at a time, and every time it is manually operated, as the bring up from a powered-down inverter is always a manual process. At least that's the case for me.

hhoebel, 2024-02-09:
> Wait, what?! What are you doing? You are supposed to **pre**-charge the inverter after DC power off through the resistor and then dis-engange the resistor after connecting the inverter DC to the battery/power source.

Yes, it's a manual process and this is why they got hot - i've assumed the inverter to be (and stay) off, changed the system and the precharge-resistors were already in the line with the inverter starting to consume power. I use 3x (to charge 6 inputs) 15R with 25W and those got really hot quickly. After noticing that, i've bridged them as i normaly do. So, normaly theey don't stay in the line unbridged for longer time. This was the moment i've wished the inverter to be safe off :-(

> Well, there you say it: The pre-charge resistor shall only be engaged for a few seconds at a time, and every time it is manually operated, as the bring up from a powered-down inverter is always a manual process. At least that's the case for me.

Correct - a fool with a tool is still a fool (or better: a fool with knowledge is still a fool). Too much working with electricity and too careless at DC :-(
But the good news is, that the soft-start of the battery seems to be really good (measured the current at soft-start today) - i don't need the resistors at all, so i just leave them bridged all the time in future :-)

schlimmchen, 2024-03-19:
Allow me to close this. As far as I can tell, we found that the DPL behaves as expected/designed and we updated the docs. Feel free to open a new issue with a feature request if you are still missing some functionality.

closed 2024-03-19 by schlimmchen