hoylabs/OpenDTU-OnBattery #652: Einbindung Meanwell NPB......

hoschiking, 2024-02-09:
Hier ein paar Bilder
[image]
[image]
[image]
[image]

_Originally posted by @skippermeister in https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/622#discussioncomment-8360632_

hoschiking, 2024-02-09:
@skippermeister

Zufällig habe ich in einem anderen Thread den Screenshots entdeckt, dass du zur PV-Überschußladung einen Meanwell NPB verwendest. Wie geht das? Soweit ich es sehe, wird bisher nur der Huawei unterstützt, da ich aber mit 24V arbeite, kann ich den nicht verwenden. Ist damit auch eine PV-Überschußeinspeisung mit Ladestromregelung in Abhängigkeit des Überschußes realisierbar? Ich suche immer noch nach einer Lösung dafür.
Falls der Bereich "Issues" nicht der richtige ist, davon gehe ich aus, kann man den Thread verschieben?
Danke dir

schlimmchen, 2024-02-09:
@hoschiking Kannst du dein Gesuch in eine neue Diskussion (kein Issue) verschieben, bitte?

hoschiking, 2024-02-09:
Ja gerne, wenn ich wüßte wie. Ich könnte es auch löschen, finde den Button aber auch nicht.
Sorry dafür. Kann es nicht ein Admin machen?
Oder du erklärst einem Github Neuling freundlicher Weise wie das geht.
Danke dir

hoschiking, 2024-02-10:
Fälschlicher Weise einen Beitrag unter Issues erstellt.
Sollte gelöscht werden.

closed 2024-02-10 by hoschiking