hoylabs/OpenDTU-OnBattery #662: [Request] Hide Solar Passthrough settings if VE.Direct is disabled

schlimmchen, 2024-02-10:
### Is your feature request related to a problem? Please describe.

People don't get it. See #651.

### Describe the solution you'd like

Hide all (full) solar passthrough setting from DPL settings view of the web application if VE.Direct is not enabled, possibly also if the pin mapping is not setup to allow VE.Direct.

spcqike, 2024-02-10:
Hi Bernhard

i really like the idea of reducing the complexity of the webUI and hiding options, that the user can't use, because he hasn't the hardware.

for this new request, i want to kindly point you to one of my very first issues, where you wanted to participate in some way 😄 
https://github.com/helgeerbe/OpenDTU-OnBattery/issues/237#issuecomment-1613242269

without solarpassthrough, users without battery can't use DPL. (at least, thats what i think and have in mind. as i'm not at home (and its dark anyway) i can't test it right now)

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/b794f46ef0807b6f10fe056ae3fb2ee5e980242d/src/PowerLimiter.cpp#L322

that is because calcPowerLimit() only works, if either canUseDirectSolarPower() or _batteryDischargeEnabled is true.
canUseDirectSolarPower() only can be true, if you have valid data from a victronMPPT.

_batteryDischargeEnabled on the other hand will be true, when solarpassthrough is enabled and startthreshold is reached.

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/b794f46ef0807b6f10fe056ae3fb2ee5e980242d/src/PowerLimiter.cpp#L287-L289

so, please keep that in mind 😄 

maybe it's really possible, to also add an option, whether or not a battery is used at all, like mentioned and discussed earlier

schlimmchen, 2024-02-10:
Understood.

Pleas note, though, that `canUseDirectSolarPower()` returns false if `!VictronMppt.isDataValid()`, so regardless of whether solar passthrough is enabled or disabled, is must not make a difference if you have no working VE.Direct: `canUseDirectSolarPower()` will be false in any case.

```
bool PowerLimiterClass::canUseDirectSolarPower()
{
    CONFIG_T& config = Configuration.get();

    if (!config.PowerLimiter.SolarPassThroughEnabled
            || isBelowStopThreshold()
            || !VictronMppt.isDataValid()) {
        return false;
    }

    return VictronMppt.getPowerOutputWatts() >= 20; // enough power?
}
```

> _batteryDischargeEnabled on the other hand will be true, when solarpassthrough is enabled and startthreshold is reached.

Yeah, that's true and that's a problem.

So we actually do need to know whether or not an actual battery is attached to the system or not. That comes back to the issue that the user cannot specify which inverter(s) are attached to the battery and which aren't (and which shall be governed by the DPL).

spcqike, 2024-02-10:
well, while reading and answering #651 i wonderd, why his logs show ``[DPL::loop] battery discharging allowed`` even so passtrough wasn't enabled in the first place.... and i found my mistake:

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/b794f46ef0807b6f10fe056ae3fb2ee5e980242d/src/PowerLimiter.cpp#L261-L290

in line 270 ``_batteryDischargeEnabled `` will be set to true if stopthreshold is not reached and passthrough is disabled and startthreshold is reached. 

so yeah, i guess one should be able to hide passthrough options without VE.Direct. maybe :D

schlimmchen, 2024-02-10:
I overlooked the exclamation mark :man_facepalming:

But still.... Nulleinspeisung without battery currently works by chance, not because it is actually supported... If we know that all inverter(s) governed by the DPL were NOT attached to a battery, we could just force `_batteryDischargeEnabled` to true, meaning "there is power available, just use it". Or similar.

spcqike, 2024-02-10:
> So we actually do need to know whether or not an actual battery is attached to the system or not. That comes back to the issue that the user cannot specify which inverter(s) are attached to the battery and which aren't

> If we know that all inverter(s) governed by the DPL were NOT attached to a battery, we could just force _batteryDischargeEnabled to true, meaning "there is power available, just use it". Or similar.

maybe its possible, to add such options to the inverter itself? like, where you can name it and name the inputs and such. 
maybe one could add a switch or checkbox say "this inverter is PV only" or "this inverter uses a battery". 

and, as the DPL right now only allows one lower and one upper limit, maybe this could become an inverter config, too? i guess this would be needed or comes handy, if we think about managing multiple inverters with DPL.

spcqike, 2024-02-15:
hello @schlimmchen 
as you allready work on this topic (and as i don't know if you get a notification, whenever someone mentions your name), i want to point you towards my findings from today:
https://github.com/helgeerbe/OpenDTU-OnBattery/issues/237#issuecomment-1945520550

```
08:37:04.150 > [DPL::loop] dcVoltage: 42.80 V, loadCorrectedVoltage: 42.80 V, StartTH: 0.00 V, StopTH: 0.00 V
08:37:04.152 > [DPL::loop] StartTH reached: no, StopTH reached: no, inverter is NOT producing
08:37:04.152 > [DPL::loop] SolarPT disabled, Drain Strategy: 1, canUseDirectSolarPower: no
08:37:04.167 > [DPL::loop] battery discharging prevented, PowerMeter: 238 W, target consumption: -20 W
```

without solar-passthrough, a start and stop voltage of 0V won't result in a discharge. i don't know why, but it says ``StartTH reached: no``. i thought, 42.8V is greater than 0V 😄 

i now put 10V for start and 5V for stop, and its working. without passthrough.

commit 5b9b3864 (2024-03-15): Feature: Refactor/Simplify DPL settings

this changeset refactors the web application's DPL settings view. the
DPL settings can be complex, and they shall be presented in a way that
allows users to comprehend their meaning. irrelevant settings are now
hidden or displayed dynamically based on the influencing settings.

* group SoC thresholds into their own card

* hide battery SoC thresholds if battery disabled. if the user did not
  even enable the battery interface, battery SoC values will not be used
  for DPL decisions. in that case we completely hide the respective
  settings from the DPL  [... cut]
commit 9c2d4495 (2024-03-15): Feature: Refactor/Simplify DPL settings

this changeset refactors the web application's DPL settings view. the
DPL settings can be complex, and they shall be presented in a way that
allows users to comprehend their meaning. irrelevant settings are now
hidden or displayed dynamically based on the influencing settings.

* group SoC thresholds into their own card

* hide battery SoC thresholds if battery disabled. if the user did not
  even enable the battery interface, battery SoC values will not be used
  for DPL decisions. in that case we completely hide the respective
  settings from the DPL  [... cut]
commit f960b059 (2024-03-16): Feature: Refactor/Simplify DPL settings

this changeset refactors the web application's DPL settings view. the
DPL settings can be complex, and they shall be presented in a way that
allows users to comprehend their meaning. irrelevant settings are now
hidden or displayed dynamically based on the influencing settings.

* group SoC thresholds into their own card

* hide battery SoC thresholds if battery disabled. if the user did not
  even enable the battery interface, battery SoC values will not be used
  for DPL decisions. in that case we completely hide the respective
  settings from the DPL  [... cut]
commit fa0ece94 (2024-03-16): Feature: Refactor/Simplify DPL settings

this changeset refactors the web application's DPL settings view. the
DPL settings can be complex, and they shall be presented in a way that
allows users to comprehend their meaning. irrelevant settings are now
hidden or displayed dynamically based on the influencing settings.

* group SoC thresholds into their own card

* hide battery SoC thresholds if battery disabled. if the user did not
  even enable the battery interface, battery SoC values will not be used
  for DPL decisions. in that case we completely hide the respective
  settings from the DPL  [... cut]
closed 2024-03-17 by schlimmchen