hoylabs/OpenDTU-OnBattery #663: [Request] Integration eSmart3 Laderegler

Michel83, 2024-02-10:
### Is your feature request related to a problem? Please describe.

Hi,
ist es möglich den eSmart3 zu integrieren ?

### Describe the solution you'd like

Dieser läuft glaube ich bei recht vielen als MPPT zum laden der Batterien, am Ausgang kann direkt ein HM gehängt werden.

alexseuf, 2024-02-11:
Hast du eine Spezifikatione von den ( modbus?) Registern die der eSmart3 über rs485 bereitstellt? Oder ein esp32 Projekt das diese erfolgreich auswertet?  Hast du in der Doku die Belegung der rs485 Schnittstelle gefunden?

Michel83, 2024-02-11:
Hi, ja hier ein Projekt. 
https://github.com/joba-1/Joba_ESmart3

Bei mir läuft er noch über die Solaranzeige wo er auch integriert ist abgefragt über rs485 und Pi. 

[ESMART3 MPPT Solar Controller Communication protocol V14.pdf](https://github.com/helgeerbe/OpenDTU-OnBattery/files/14231686/ESMART3.MPPT.Solar.Controller.Communication.protocol.V14.pdf)

Die belegung ist in dem Github von joba gut dokumentiert.

spcqike, 2024-02-11:
https://www.photovoltaikforum.com/thread/132657-esmart-3-mppt-regler/

dort ist auch aller Hand über den laderegler geschrieben und festgehalten. Von pinouts über Protokolle und vielen vielen negativen Erfahrungen (: 

Michel, wie sind deine Erfahrungen? Vielleicht hat sich die Software von laderegler ja seit 2019 verbessert und die Fehler wie zurücksetzen und stoppen des MPPT und durchreichen der PV Spannung an die Batterie treten nicht mehr auf?

schlimmchen, 2024-02-11:
> durchreichen der PV Spannung an die Batterie
:scream: 

Also. In den letzten Wochen wiederhole ich mich oft aus verschiedenen Gründen. Auch hier gilt (wie bei dem Wunsch, die solare Leistung per MQTT bereitzustellen): Da müsste erstmal eine Abstraktion eingeführt werden, die es erlaubt auszuwählen von wo die MPPT Daten kommen. Das ist erheblicher Aufwand. Und dann muss das Protokoll für diesem eSmart3 korrekt implementiert werden. Da muss sich dann jemand finden, der da Bock drauf hat und die Fähigkeiten. Der letzte Teil ist entscheidend.

Hinzu kommt, dass man mit so viel neuem Code endgültig Gefahr läuft, dass die Firmware zu groß wird bei der aktuellen Paritionierung bzw. Update-Strategie, und dann gibt es ein noch viel größeres Problem zu lösen :man_shrugging:

Michel83, 2024-02-11:
> https://www.photovoltaikforum.com/thread/132657-esmart-3-mppt-regler/
>
> dort ist auch aller Hand über den laderegler geschrieben und festgehalten. Von pinouts über Protokolle und vielen vielen [...]

Also mein Esmart3 von vevor (60a) läuft seit 1,5 Jahren problemlos in einem 12v 220ah System. Natürlich täglich im Lademodus, wenn Batterie voll wird entladen auf 30 Prozent, gesteuert über eine externe Steuerung.

Schlechtes kann ich nicht berichten ausser vielleicht ein recht lauter Lüfter, und klemmen die vielleicht nicht dem Standard entsprechen selbst das einrichten über die App klappte problemlos.

alexseuf, 2024-02-11:
Heißt das das aus eurer Sicht das Project bzgl neuer Hardware am Ende ist?

alexseuf, 2024-02-11:
> Hi, ja hier ein Projekt. https://github.com/joba-1/Joba_ESmart3
> 
> Bei mir läuft er noch über die Solaranzeige wo er auch integriert ist abgefragt über rs485 und Pi.
> 
> [ESMART3 MPPT Solar Controller Communication protocol V14.pdf](https://github.com/helgeerbe/OpenDTU-OnBattery/files/14231686/ESMART3.MPPT.Solar.Controller.Communication.protocol.V14.pdf)
> 
> Die belegung ist in dem Github von joba gut dokumentiert.
> 
> ![Screenshot_20240211-090641_Samsung Internet Beta](https://private-user-images.githubusercontent.com/36561525/303906006-f2b56b2c-99cd-4875-973f-2bb37da41c93.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc2NDMxNDgsIm5iZiI6MTcwNzY0Mjg0OCwicGF0aCI6Ii8zNjU2MTUyNS8zMDM5MDYwMDYtZjJiNTZiMmMtOTljZC00ODc1LTk3M2YtMmJiMzdkYTQxYzkzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjExVDA5MTQwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwNGE4NmE0NWJmOGQ2MTNjMWMxNjU4MWRiNDkzMmEwZWI4Y2M4YzRmZDA0NzRhODAyNzFiMWY4OTUxOWIyZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.A2NE_3Sfy1MQgJho7j_4DZFADVeWlceNRJl86JxfIn8)

Wenn ich aber nach einem aktuellen eSmart3 Google finde ich einen RJ45 Stecker für die rs485 kommunikation.. hast du zufällig auch die Belegung von diese da?

Michel83, 2024-02-11:
@alexseuf 

Jap
[image]

schlimmchen, 2024-02-11:
> Heißt das das aus eurer Sicht das Project bzgl neuer Hardware am Ende ist?

Ne, das hab ich nicht gesagt. Aber es gibt schlimmstenfalls ein lästiges Problem zu lösen, weil erst das Partitionsschema angepasst werden muss, und/oder die Update-Strategie. Oder die Firmware mit allen Features läuft nur noch auf ESP32 mit 4MB Speicher. Oder sonst irgendeine Lösung.

Ich bereue das erwähnt zu haben, das hat mit dem Feature wenig zu tun. Es könnte aber darauf hinauslaufen, dass dieses Feature, weil es recht groß wäre, den Build für den generic_esp32 fehlschlagen lässt, weil dessen Flash dann überläuft.

alexseuf, 2024-02-11:
Ich habe es geahnt wieder eine neue Belegung der RJ45 Schnittstelle..

schlimmchen, 2024-12-01:
Hängt von #457 oder vielmehr von #1408 ab. Anschließend kann man sich nochmal überlegen, ob man die Kommunikation mit solch einem Teil integriert. Wird ohne Hardware zum Testen aber sehr anspruchsvoll. Daher setze ich das Label "contribution needed".

AndreasBoehm, 2025-01-31:
Seit Version [2025.01.23](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2025.01.23) könnt ihr jeden Laderegler (MPPT) mit OpenDTU-OnBattery nutzen der seine Daten über MQTT bereitstellt.