hoylabs/OpenDTU-OnBattery #669: HTTP power meter request interval

schlimmchen, 2024-02-13:
### What happened?

I saw in https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/179#discussioncomment-8452399 that @spcqike knows that there is a config value for the HTTP power meter update interval. That is not configurable via the web app.

### To Reproduce Bug

Look at HTTP power meter setting in web app.

### Expected Behavior

Request interval shall configurable from web app since it is already implemented.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

latest

spcqike, 2024-02-13:
Hey 

if one lets the user change the update interval free, one should also mention the fixed 30s timeout in DPL. For now we had 10s interval as standard with a 30s timeout. 

One could make the timeout like 3 times the interval. Or make it configurable, too. 

Maybe one could also add a „baseload“ to powermeter settings and use this as fallback, like multiple users asked for, when timeout is exceeded. 

Anyway, just making the interval easily configurable by the user can make problems with DPL at some point :)

schlimmchen, 2024-02-13:
> Or make it configurable, too.

We can add so many config items that nobody will ever understand all of them :wink: 

> And the interval does not apply to all power meter sources. AFAIK it’s http+json and SDM1/3P only. So one could hide it there and needs to find a solution for the other timeout handling in DPL

Of course.

The configurable interval should have a maximum value of... let's say 15 seconds. And the DPL will wait three times as long. Or the other way around: The power meter provider will tell when its data is outdated. The last part, however, might be difficult until the whole power meter is refactored similarly to the battery provider interface, i.e., until there is a proper interface.

closed 2024-07-10 by schlimmchen