hoylabs/OpenDTU-OnBattery #670: [Request] JK-BMS publish all data to MQTT

ButterBetzi, 2024-02-13:
### Describe the solution you'd like

Hi,
the data of a connected JK-BMS gets published via mqtt since the last release, nice!
But I relealised that it misses a lot of datapoints that can be read out via bluetooth (e.g. the individual cell voltages).

Why arent all datapoints published?
Please also publish those values.

schlimmchen, 2024-02-13:
Well, first of all, not all values that are accessible via Bluetooth are accessible from the UART interface. At least some values that are available in the Android App are not returned when asking "give me all your data" over the UART. All documented parameters on the UART are indeed returned after such a message. So I don't think that we can ever get more out of the JK BMS over the UART connection.

However, all that are transmitted over UART (enable verbose logging and look at the serial output or console log), are indeed published to the MQTT broker. There might be one or two exceptions, I don't remember for sure. The password is such an exception.

You are asking specifically about the cell's voltages. Those definitely are published to the MQTT broker. So I tend to dismiss your request as invalid.

Are you asking about Home Assistant auto-discovery?

ButterBetzi, 2024-02-13:
No, i'm not.
I only have a very limited amount of datapoints. In the Web-GUI and via MQTT.
If you say that the cell voltages are being publihsed then i've got a bug or wrong settings. Here are all my sensors/datapoints:
[image]
[image]

schlimmchen, 2024-02-13:
> > Are you asking about Home Assistant auto-discovery?

> No, i'm not.

Well, given your screenshot: Yes your are :grin:

Nearly all data is published to the MQTT broker:

[image]

So what data points do you think are important to be auto-discoverable in Home Assistant? To be upfront: I don't think every cell's voltage should be auto-discoverable or even be of particular interest within Home Assistant.

ButterBetzi, 2024-02-13:
> Well, given your screenshot: Yes your are 😁

whooops, seems that i had a brainfart when answering there :D

> So what data points do you think are important to be auto-discoverable in Home Assistant? To be upfront: I don't think every cell's voltage should be auto-discoverable or even be of particular interest within Home Assistant.

My approach would be 'full freedom to the (HASS) user'. I would personally like to have all data auto-discoverable and then decide for myself which entities to hide/deactivate. ATM I only use SoC, Current and Temperature and will deactivate all others, because I dont need them.
But at this specific case I have another ESP32 running with https://github.com/syssi/esphome-jk-bms to get the cells voltages that I want.

So in summary: I dont need e.g. "Alarm: BMS Overtemperature" but "Cell 1 Voltage". Others most likely want/need different datapoints. I pledge for making everything auto-discoverable :)
(But I'm also interested why you think that it shouldnt be that way ;))