hoylabs/OpenDTU-OnBattery #674: DTU Abfrageintervall BUG

Cartman586, 2024-02-16:
### What happened?

Beim einstellen oder ändern des Abfrageintervalls kommt die Fehlermeldung 
"Die Frequenz muss zwischen 860250000 und 923500000 kHz liegen und ein vielfaches von 250kHz betragen!"

### To Reproduce Bug

Einfach den Abfrageintervall ändern und speichern

### Expected Behavior

Speichern des Abfrageintervalls
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.02.09

schlimmchen, 2024-02-16:
Das ist ein Bug aus dem Upstream-Projekt, bitte dort danach suchen, da gab es bereits entsprechende Issues. Da heißt es "Strg + F5" drücken um die WebApp neu zu laden. Allerdings scheint das nicht alles zu sein. Vermutlich ist ein ungültiges Frequenzband per default eingestellt gewesen und neuerdings ist das ungültig wegen der Überprüfung auf die Region.

Cartman586, 2024-02-17:
Danke. Komme hier aber nicht weiter

schlimmchen, 2024-02-17:
Als Workaround könntest du dir deine config.json als Backup herunterladen und editieren. Setze "cmt_frequency" auf "865000000" und "cmt_country_mode" auf "0". Speichere die Datei und lade sie wieder hoch. Versuche dann erneut das Polling-Interval umzustellen. Ich vermute, dass es dann geht. Falls ja, lass es uns bitte wissen.

Cartman586, 2024-02-20:
> Als Workaround könntest du dir deine config.json als Backup herunterladen und editieren. Setze "cmt_frequency" auf "865000000" und "cmt_country_mode" auf "0". Speichere die Datei und lade sie wieder [...]

Danke, komme hier leider nicht weiter

Cartman586, 2024-02-20:
> Warum nicht?
> 
> tobi0171 hat [bestätigt](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/639#issuecomment-1954107298), dass das Problem so zu umgehen ist. Ansonsten ist es wie gesagt ein Problem im Upstream-Projekt, melde ich dort bitte. Für uns ist hier nichts zu tun.

Habs soeben nochmal probiert, jetzt hat es mit dem Workaround geklappt. Danke

uli-bs, 2024-02-23:
Auch von mir, die manuelle Datenänderung hat das Problem beseitigt, der Wert vorher war irgendwas mit "...204"

closed 2024-03-18 by schlimmchen