hoylabs/OpenDTU-OnBattery #675: [Request] Option for Limiting Battery-Power

ButterBetzi, 2024-02-16:
### Describe the solution you'd like

Hi,
my current hardware setup gets pretty inefficient when my (24V) battery outputs more than 300 W. 
Therefore I've currently limited the overall Power Limit to 300 W.
Problem:
When my battery is full, I'm in full solar passtrough and the solar panels produce more than 300 W of power the energy gets lost for me.

Idea -> Provide the possibility to set a power limit for the connected battery.

(I tried to find such an option but was unsuccessful. If there is a possibility to solve my problem otherwise please tell me so :))

spcqike, 2024-02-16:
duplicate of https://github.com/helgeerbe/OpenDTU-OnBattery/issues/657

ButterBetzi, 2024-02-18:
sorry, didnt know that existes.

closed 2024-02-18 by ButterBetzi