hoylabs/OpenDTU-OnBattery #676: Load corrected voltage is too high in some scenarios

schlimmchen, 2024-02-17:
### What happened?

I was working on the DPL code and noticed this problem with the load correction factor.

### To Reproduce Bug

Assume a sunny morning as an example. The household consumption is high (for any reason). The inverter produces at high or even full power, e.g., 1500W. The charge controller is producing 1000W. The battery is only drained at a rate of 500W. However, the load correction factor calculates a corrected voltage based on the 1500W AC output power of the inverter.

### Expected Behavior

We are only interested in the voltage drop of the battery itself to estimate its voltage once we stop drawing current from it. If we assume the battery to bounce back higher than it actually will, because we assumed it was drained at 1500W not the actual 500W, we will have discharged it (well) beyond the stop threshold.

The most common hardware setup for this project is a battery, a Victron MPPT charge controller, and a Hoymiles inverter. This means we can be much smarter: Use the current drain as reported by the BMS, if available, and derive the actual battery load from it. Otherwise, reduce the inverter's AC output power value by the charge controllers output power, if available, to estimate the battery's actual load. Only if neither BMS nor (Victron VE.Direct) charge controller info is available, the inverter's AC output power shall be assumed as the battery load, as a last resort.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

12345678

spcqike, 2024-02-17:
> Use the current drain as reported by the BMS

if you have a BMS that reports the current, doesn’t it report the real battery voltage, too? Wouldn’t it be easier to just use this battery voltage than? Like you mentioned in another issue: get battery voltage from BMS or smart shunt or charge controller or inverter input (in that order) 

I thought a load correction factor is only needed, if you don’t have a good BMS (one that you can get data like current and voltage from) or a smart shunt.

schlimmchen, 2024-02-17:
> Wouldn’t it be easier to just use this battery voltage than?

The battery voltage itself still drops with higher loads due to the cell's internal resistances. The load correction factor is still useful, even if you can calculate with the BMS-reported voltage and the BMS-reported current, to estimate how much the battery voltage will bounce back up once you stop draining. The load correction factor is then tuned to something else, i.e., the battery's internal voltage drop.

spcqike, 2024-02-17:
> The load correction factor is then tuned to something else, i.e., the battery's internal voltage drop.

but does this work? The battery internal restistance and therefore the voltage drop depends on several things like temperature, overall health(SoH) and SoC. 

just a random example discharge image for a single lifepo cell for better explanation:
[image]

as you can see, when it’scompletely discharged, as soon as discharge stops the voltage jumps back by about nearly 0.75V, from 2.0V to ~2.75V. What factor would you like to use? In this case, when it’s nearly empty, the internal resistance is approx 37.5mOhm. I doubt, that it’s the same when it’s full, because the initial drop is only about 0.25V, so 12.5mOhm. I guess A wrong and to high factor will make more problems, than no factor at all. As you can let the openDTU make believe that the battery is always fully charged or even more :) 

As I said in another comment somewhere, I wouldn’t use any factor to compensate for battery internal resistance and voltage drops. I would discharge to like 2.75V and stop there. 
for this example cell, I would loose the last few Ah (0.2-0.3Ah or something), which would be the lower 0%-5% from its SoC. But, that’s fine. As it’s more relaxing for the battery to not discharge it completely. Isn’t it? 

Another point is: if you don’t compensate battery internal voltage drop, you will be able to see poor health or other problems faster. As you always see and monitor the real voltage.

spcqike, 2024-02-18:
Another diagram that shows the internal resistance, based on temperature and SoC

[image]

From out recent discussion 
https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/664#discussioncomment-8485389

TomBrett72, 2024-05-25:
Hallo zusammen,
bin noch nicht lang dabei, deshalb zuerst ein riesiges Dankeschön für dieses super Projekt! Meine Anlage ist eher standard. 2x410Wp, Victron 100/30, 2,5kWh LiFePo, HM600 und natürlich DTU! Möchte mit meinem Beitrag hier die Diskussion wieder anschubbsen. Hauptgrund: Ich stelle ständig am max-Limit rum und greife somit manuell in die Regelung ein. Möchte bei meiner kleinen Anlage einfach nicht immer 400Watt ziehen. Im Winter hatte ich den DPL sogar ganz aus und bin konstant 35Watt gefahren. Lief über Wochen stabil. Und sorry, bevor ich mich mit meinem Englisch blamiere oder ich für den Text 1 Tag brauche, versuche ich es in Landessprache.
@schlimmchen Ja, ich beobachte genau das von Dir beschriebene Verhalten. Sofern man ohne SolarPathThrough fährt, passiert es sogar, daß bei starker morgentlicher Ladeleistung der Einschaltpunkt sehr früh erreicht wird und man dann aus einem fast leeren Akku die MaxLeistung zieht. Meine erste Idee war deshalb eine Leistungsbilanz zu rechnen. Also P[victron] - P[WR] und diese Differenz mal Lastkorrekturfaktor. Das würde dann auch die Spannungsüberhöhung während Laden berücksichtigen. Ob physikalisch richtig bleibt die Frage.
Dann kam @spcqike mit seinem cleveren Ansatz, der die Komplexität reduziert. Bin ich immer Fan von sowas. Und folgend @SW-Niko mit einer Idee, ganz nach meinem oben beschriebenen Grund "Leistungsanpassung". Wie würde ich nun den Knoten lösen? Hab im Bild eine Kennlinie für den Bereich 10%SOC-90%SOC berechnet. Zum einen die Herstellerkennlinie meines Akkus. Bessere Werte hab ich leider nicht, sollten aber ausreichen. Zum anderen die WR-Leistung, bei meiner Konfiguration ja max 400Watt. Mit der Gleichung läßt sich ein weiteres Limit berechnen, welches man in einen bestehenden Minimizer eintragen könnte. Ich kann kein C++ aber ich denke es müßte in Zeile 772 rein (effPowerLimit = std::min<int32_t>(effPowerLimit, maxPower);)
Nun denn... mal sehen, kann schon sein, daß ich noch nicht alles bedacht hab.
[image]

SW-Niko, 2024-06-01:
Hallo,
es geht voran.  Bei dem miesen Wetter habe ich Zeit zum programmieren.
Bei mir läuft jetzt die Testphase.

17:11:39.051 > [Battery Limiter] Mode: 1, Active: 1, Battery Voltage: 25.91, Requested Power: 141 W, Limited Power: 91 W

Das passt soweit. Die Leistung wird soweit begrenzt das die Batteriespannung nicht unter 25.9V fällt.
Wenn die Sonne nicht mehr genügend Energie liefert dann sollte der Inverter abgeschaltet werden.

Noch etwas Geduld. 😀

schlimmchen, 2025-02-01:
Superseded by #1399.

closed 2025-02-01 by schlimmchen