hoylabs/OpenDTU-OnBattery #684: OpenDTU Anzahl der Wechselrichter 10???

Guppy82, 2024-02-21:
### Is your feature request related to a problem? Please describe.

Hallo, ich brauche mal Hilfe

Ich habe eine OpenDTU und möchte mehr als 5 Wechselrichter anlernen.

Das System sagt mir aber das nicht mehr als 5 gehen.

Laut Open DTU sind aber 10 möglich. Gibt es da eine Einstellung die ich ändern kann oder muß

Habe HMS und HM Wechselrichter.

Hoffe mir kann einer Helfen.

### Describe the solution you'd like

Läuft auf einem ESP 32

spcqike, 2024-02-21:
Hallo,

die Info dazu findest du in der README des Projekts. direkt auf der Startseite

> Warning
> 
> In contrast to the original openDTU, with release 2023.05.23.post1 openDTU-onBattery supports only 5 inverters. Otherwise, there is not enough memory for the liveData view.

Nein, da kannst du nichts machen, außer die originale openDTU zu nehmen, oder mehrere openDTUs aufzusetzen.

Guppy82, 2024-02-21:
Was soll den die originale openDTU sein? Oder wo bekomme ich die her? 
Oder meinst du die von Hoymiles?

Habe auch schon im Forum gesehen dass mehr als fünf angemeldet worden sind!

spcqike, 2024-02-21:
Damit meine ich den Upstream
https://github.com/tbnobody/OpenDTU

die openDTU-onBattery ist ein Fork der "Originalen", der um Batteriemanagement und Stromzählerauswertung inkl. Einspeiselimitierung erweitert wurde.
Aufgrund der zusätzlichen Anpassungen musste allerdings die Anzahl der maximal koppelbaren Wechselrichter reduziert werden.

> Habe auch schon im Forum gesehen dass mehr als fünf angemeldet worden sind!

WIe gesagt, bei der "openDTU" geht das auch, bei "openDTU-onBattery" ging es bis Mai 2023.

helgeerbe, 2024-02-21:
In der letzten Version, wurde die api für den Live view geändert. Es könnte sein, dass man wieder auf 10 Inverter erlauben könnte. Schaue ich mir mal an m

Guppy82, 2024-02-21:
Ok Das wäre Super!!!
Muß ich da dann was umstellen?

schlimmchen, 2024-02-22:
> Muß ich da dann was umstellen?

Nein. Falls Helge das umsetzen kann, wird es ein neues Release geben, das dann mehr Wechselrichter verwalten kann.

Manos1966, 2024-02-23:
Hallo @Guppy82 

Schau mal hier bitte:
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Compile-your-own-Firmware-with-Visual-Studio-Code#how-to-change-the-max-number-of-inverters-that-opendtuonbattery-can-handle-default-is-5

helgeerbe, 2024-03-22:
Habe mal eine PR #788 vorbereitet, um den Wert wieder auf 10 Inverter zu ändern.

schlimmchen, 2024-03-22:
Das nächste Release lässt wieder 10 WR zu, wie im Upstream.