hoylabs/OpenDTU-OnBattery #685: Live-Daten: "Leistung / Netz" aktualisiert sich nicht

Forstgrund, 2024-02-21:
### What happened?

Die Version 2024.02.19.post1 aktualisiert die Live-Daten Anzeige von "Leistung / Netz" nicht. 
Flasht man zurück auf 2024.02.09 funktioniert es perfekt

### To Reproduce Bug

Normal Install 2024.02.19.post1 without Battery....

### Expected Behavior

Die aktualisierung sollte automatisch erfolgen wie es immer war

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.02.19.post1

spcqike, 2024-02-21:
duplicate of https://github.com/helgeerbe/OpenDTU-OnBattery/issues/682 ?

Ralph161065, 2024-02-21:
Gleiches Problem bei zwei verschiedenen Installationen

der-peterh, 2024-02-23:
WebApps verstehe ich zwar nicht, aber einen 'Dreher' könnte es in webapp/src/components/InverterTotalInfo.vue Zeile 118 geben:
 export default .... 
 hier werden props huaweiData und powerMeterData überkreuzt mit type: Object as PropType<PowerMeter> und type: Object as PropType<Huawei>
... hier ist dann aber auch das Ende meines Verständnisses erreicht - yarn, js und vue sind sowas von nicht meine Welt :(

schlimmchen, 2024-02-23:
Ich hab mir den Code im Frontend noch nicht angeschaut, aber es wäre wohl an der Zeit zu bemerken, dass der Fehler bei mir beispielsweise nicht auftritt (MQTT PowerMeter), und daher der Fehler in der Firmware zu suchen ist. Alle betroffenen melden Probleme mit dem HTTP PowerMeter. Da gab es auch Änderungen in den letzten Wochen, die durchaus dazu geführt haben können, dass aktualisierte Werte nicht mehr in die WebApp kommen oder überhaupt gar nicht abgeholt werden.

der-peterh, 2024-02-23:
Mein ShellyEM3 läuft über MQTT und der Fehler tritt nur beim Auto Refresh auf: HM aktualisiert sich, PowerMeter nicht. Beim manuellen refresh (ziehen im GUI) wird hingegen aktualisiert.
Über MQTT schreibt die Firmware ebenfalls aktuelle Werte.
Ich vermute also entweder der (Websocket oder so was) wird falsch befüllt oder falsch ausgelesen...

danny1605, 2024-02-24:
Gleiche Problem hier auch.

Forstgrund, 2024-02-24:
Meine Daten kommen vom Hichi am Zähler und in Tasmota kann ich sehen wie sich der Wert ständig ändert....nur mit der aktuellen Firmware bleibt der Wert stehen. Aktualisiere ich die Webseite zieht er sich den richtigen Wert vom Hichi. Ich hab jetzt wieder 
Firmwareversion 2024.02.09 laufen....da geht alles wie es sein soll. Ich habe noch keine Batterie angeschlossen

ingrimsch, 2024-02-24:
Kann ich bestätigen:

Site Refresh holt alle Daten - hier funktioniert es
Autorefresh aktualisiert die "Leistung / Netz" Kachel im Live View nicht

Version: 2024.02.19.post1, 
Settings: ohne Batterie nur das http/json Powermeter (Tasmota / Hichi) konfiguriert

cerise21, 2024-02-25:
Hier ebenso (mit 2024.02.19; selbstcompiliert wegen pinmapping für RS485-Anschluss eines SDM630-Powermeters):
Keine automatische Aktualisierung von 'Leistung/Netz'.
Zusätzlich ebenso keine Aktualisierung von 'MPPT-Gesamleistung' und (wahrscheinlich auch) von 'Ladezustand'.
In der Detail-Ansicht unter 'SmartSolar xxx' erfolgt die Aktualisierung der 'Eingang'- und 'Ausgang'-Werte wie gewohnt…

schlimmchen, 2024-02-25:
Ich sehe das Problem nun auch, wenn ich 2024.02.19.post1 installiere. Zuvor war ich auf einem Branch unterwegs, der die letzten Upstream-Änderungen noch nicht enthielt.

> hier werden props huaweiData und powerMeterData überkreuzt mit type: Object as PropType und type: Object as PropType

Stimmt, das ist ein alter Fehler, der vermutlich deshalb keine Auswirkungen hat, weil die Objekte PowerMeter und Huawei gleich aufgebaut sind.

Der PowerMeter Wert im JSON des websockets wird aktualisiert, kommt also bis in die Web App, die im Browser läuft. Dann läuft irgendetwas schief.

Aha, hier: Im Upstream Projekt wurde die API insofern umgestellt, als dass die Daten der Inverter nur bröckchenhaft, oder genauer gesagt, nur pro Inverter übermittelt werden, um all zu große JSON Objekte in der Firmware zu vermeiden (wenn ich das alles richtig verstanden habe). In diesem Zusammenhang wird im Frontend nicht mehr die Nachricht aus dem Websocket stumpf als `this.liveData` interpretiert. Wir müssen die OnBattery-spezifischen Teilobjekte der Nachricht aus dem Websocket ausdrücklich übernehmen. Der Fehler betrifft also auch die Live-Header-Werte der Batterieschnittstelle, der Huawei-Schnittstelle, und der VE.Direct-Schnittstelle.

mastertape, 2024-02-25:
Wollte es auch grade melden… Ich bedanke mich bereits jetzt für den fix

hhoebel, 2024-02-26:
Ich habe den Fix in der aktuellen Version 2024.02.19.post1 getestet und "Leistung/Netz" aktualisiert bei mir noch nicht - der Rest (Victron + Pylontech) geht jetzt.

schlimmchen, 2024-02-26:
Also das hier diskutierte Problem ist mit der 2024.02.19.post1 erst in die Firmware reingekommen. Und meine Änderungen in #696 sind noch nicht veröffentlicht. Was hast du getestet, @hhoebel?

hhoebel, 2024-02-26:
Da ich wegen dem Problem mit dem falschen Kanalcount des DPL bei meinem HMT2250 aktuell die von dir erwähnte Firmware in gepatchter Version betreibe, habe ich deinen Fix dort mit eingespielt, compiliert und dann gestestet. Entweder ist das aber nicht alles, was man ändern muss oder die Änderungen benötigen noch was anderes als nur einen Build in VSCode, um im BIN-File zu landen (bin noch neu im Kompilieren des Projekts).

schlimmchen, 2024-02-26:
Du hast garantiert vergessen, die web app zu bauen. Gehe in einer Kommandozeile in den Unterorder "webapp" dieses Repos und führe "npm run build" aus (oder noch genauer: Installiere npm (nodejs), führe "npm install" aus, und dann "npm run build").

Und/Oder schau hier: https://www.opendtu.solar/firmware/compile_webapp/

hhoebel, 2024-02-26:
Danke Dir! Sowas in der Art habe ich auch irgendwann vermutet. Kaum macht man es richtig, funktioniert es auch schon :-)

ingrimsch, 2024-02-26:
Getestet mit dem Build von https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8037760180
--> funktioniert! :)

PR #696 (2024-02-29): Fix: copy OnBattery-specific data from live view websocket
closes #685.
closes #682.
commit c560d1d9 (2024-02-29): fix: copy OnBattery-specific data from live view websocket (#696)

closes #685.
closes #682.
closed 2024-02-29 by schlimmchen