hoylabs/OpenDTU-OnBattery #691: Refresh time

danny1605, 2024-02-24:
### What happened?

The scheduled refreshing time doesnt work. 
[image]

### To Reproduce Bug

I dont know 

### Expected Behavior

An Update, which eliminate this bug.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.02.19.post1

### Anything else?

[image]

schlimmchen, 2024-02-24:
Are you complaining that you set a "Abfrageintervall" of 3 seconds, but the live view tells that the data is 12 seconds old?

danny1605, 2024-02-25:
> Are you complaining that you set a "Abfrageintervall" of 3 seconds, but the live view tells that the data is 12 seconds old?

That's right, with this setting I change the value ?

schlimmchen, 2024-02-25:
Not directly, no. The poll interval determines how often the Hoymiles radio controller loops over all registered inverters. In each loop, every inverter is polled sequentially. More inverters means the data age will become bigger. Also, in some rounds, the radio controller might fail to fetch the inverter information. The communication is wireless and hence error-prone. Also, since your HMS-800 is offline, it takes all tries that the radio controller will attempt to reach it until giving up, delaying fetching the inverter data of the online inverter accordingly.

Does this explain why the data age is not 3s when the poll interval is set to 3s?

If so, please close this issue.

closed 2024-02-25 by danny1605