hoylabs/OpenDTU-OnBattery #710: Add RS485 Pylontech support

Manos1966, 2024-03-04:
### Is your feature request related to a problem? Please describe.

Currently threre is CANbus Support for Pylontech & other Batteries
The information CAN provides is good but limited compared to the information available from the RS485 port:
**CANbus Information currently:**
[image]

**RS485 Information:**
[image]
[image]
[image]

### Describe the solution you'd like

Add RS485 support to the OpenDTUonBattery in order to make the additional information available.

A solution exists already, limited to the first Pylontech Battery:
#622 adding RS485 Pylontech support 

https://github.com/skippermeister/OpenDTU-OnBattery

tobi0171, 2024-03-06:
Das wäre super ... ich würde mich freuen

Solarteur, 2024-03-07:
Es gibt dazu einen Fork von skippermeister https://github.com/skippermeister/OpenDTU-OnBattery, der aber leider kein Huawei R4850G2 AC-Ladegerät unterstützt und deshalb für mich keine Option ist.

Andererseits sehe ich auch, dass der verfügbare Speicherplatz immer weiter schrumpft...

Napizaki, 2024-03-08:
Ich fände es ebenfalls super. Vor Allem die Spannungen der einzelnen Zellen (min/max) fände ich interessant. Derzeit prüfe ich sowas immer manuell mit der Battery-View Software. Dieses wäre damit überflüssig.

Manos1966, 2024-03-10:
> Andererseits sehe ich auch, dass der verfügbare Speicherplatz immer weiter schrumpft...

So gesehen, ich muss gestehen, ich habe kein Problem einen extra ESP32 fuer die Beobachtung der Pylontech Zellen zu haben (separat von der Haupt-OpenDTUonBattery Einheit.

Ich gehe davon aus, die Meisten von uns nutzen Homematic Systeme und sammeln Daten ueber MQTT. Dann waere es mir lieber, dass auf den Speicher der ersten OpenDTUonBattery die wichtige Real-Time Aufgaben laufen (z.B. zwei Victron MPPT, zwei Wechselrichter usw...) und auf andere "OpenDTUs" die restliche, non-critical Informationen.

Manos1966, 2024-03-27:
I close this as I see it is more beneficiary to run an existing solution on a separate ESP32 than trying to add this feature to OpenDTUonBattery.

closed 2024-03-27 by Manos1966