hoylabs/OpenDTU-OnBattery #723: [Request] charging to SoC target with defined power after "Immediate charging requested" by Pylontech

gitisgreat2023, 2024-03-07:
### Is your feature request related to a problem? Please describe.

When the SoC, for whatever reason, drops so low that an immediate charge is requested by the Pylontech battery, to avoid damage, currently nothing happens. Though a controllable AC charger is present and could charge the battery again to safe level.

### Describe the solution you'd like

When "Immediate charging requested" switches from "No" to "Yes" the AC charger should charge to an specified SoC Target with a to be specified power. 

### Describe alternatives you've considered

Use an external script (like NodeRed) which monitors via MTTQ the "Immediate charging requested" state and sets via MTTQ the charge power until the target SoC is reached.

schlimmchen, 2024-03-09:
@MalteSchm Is this something you'd like to takle?

MalteSchm, 2024-03-10:
My system tells me: yes I should address that... Here is why:

I just checked my dashboard and found all alarms red. It seemed that the inverter operated at full power with communication between openDTU and the inverter being broken. Soft and hard restarts of the ESP were unsuccessful. I got the system back by issuing an inverter reset command.

I think that resetting the inverter when communication is broken for a long time and starting up the PSU when charging is requested would be countermeasures that should be implemented

schlimmchen, 2024-03-10:
Meh... You, too? There are other users who recently started reporting that their inverter is somehow stuck and must be revived through a reboot command, as even with my more recent changes that repeat limit updates and power requests their inverter somehow manages to keep in standby... This is weird...

The DPL could (quite easily with the recent changes) detect if the inverter is not reaching the desired state for a longer time and trigger issuing a reboot message. Should we do that? Can you open a respective issue if you think that's desirable?

> and an ESP reset

That would be a last resort, i.e., after issuing the reboot command did not do the trick.

gitisgreat2023, 2024-03-11:
@MalteSchm 
That's sounds a bit scary, an inverter going rogue... how low went the SoC on your Pylontech, did he shutdown due to undervoltage? Any idea why it got in that situation?
To be prepared: "I got the system back by issuing an inverter reset command.": how do I do this? This is not possible in the WebUI, right?

MalteSchm, 2024-03-11:
@schlimmchen 

Yes I had that twice now. There could have been other cases that went without me noticing. The inverter is reset in regular intervals at night so the issue will resolve over time. The battery stays fully discharged during this period, potentially even disconnected as the BMS will cut the cord at some point.

I think changing the DPL this way is desirable yes.

@gitisgreat2023 

First time: 0%, yesterday 2%. I think the inverter did shut off at this stage but: Once the PSU started charging the inverter also started up again.
You can reset ("Neustart") the inverter in the web-ui. This is accessible from the buttons in the main page / dashboard. Right next to the buttons to turn the inverter off and on

MalteSchm, 2024-03-11:
> I just checked my dashboard and found all alarms red. It seemed that the inverter operated at full power with communication between openDTU and the inverter being broken. Soft and hard restarts of [...]

Created https://github.com/helgeerbe/OpenDTU-OnBattery/issues/748

MalteSchm, 2024-03-12:
> @MalteSchm 0% and 2% ain't good... yes we definitely need a feature coping with this. We should under all circumstances avoid that the inverter goes rogue and the PSU starts charging, creating a high power circular flow. If one is on holidays that could get dangerous.
> 
> > First time: 0%, yesterday 2%. I think the inverter did shut off at this stage but: Once the PSU started charging the inverter also started up again. You can reset ("Neustart") the inverter in the web-ui. This is accessible from the buttons in the main page / dashboard. Right next to the buttons to turn the inverter off and on
> 
> I'm stupid I guess: next to the turn on/off there's the grid info: ![image](https://private-user-images.githubusercontent.com/138098278/312000588-7328a5d3-9c08-4408-93d7-87791e6cc370.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDI3NDgsIm5iZiI6MTcxMDI0MjQ0OCwicGF0aCI6Ii8xMzgwOTgyNzgvMzEyMDAwNTg4LTczMjhhNWQzLTljMDgtNDQwOC05M2Q3LTg3NzkxZTZjYzM3MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTIwNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNzI0MmIxOGYyYTE0N2Y1OGQ2NDI1OTIxNjI3Y2VhZDE1YTVmZmMyMjUzZjhlNzYzZjhhYTBmNDEzMzgyN2M2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qJz275JcCCAB7I8zu-bTjX_M6y0AwXE1qZEnTE3PqOk)
> 
> ![image](https://private-user-images.githubusercontent.com/138098278/312000728-3411c8e0-27bf-4afc-8f1c-f437e67b466e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDI3NDgsIm5iZiI6MTcxMDI0MjQ0OCwicGF0aCI6Ii8xMzgwOTgyNzgvMzEyMDAwNzI4LTM0MTFjOGUwLTI3YmYtNGFmYy04ZjFjLWY0MzdlNjdiNDY2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTIwNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iODAzZjc3NmFmMDdhYmVkYTQ5YjM0ZTlkMGRkMjNjN2I3MjY2M2YxMTRmZGQ2N2QyMzRhNjQ5ZjBiM2Y3MGRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ebWHgmI98EShfwBP5KQES-XNcVW7mD4UyqZg1fVhxUs)
> 
> Or do you meant the on/off is the inverter reset?

Click on thin on/off button. There is more...

MalteSchm, 2024-03-12:
@schlimmchen 
How do I get the `PylontechBatteryStats` needed for `_chargeImmediately`

This did not work:
`auto stats = static_cast<PylontechBatteryStats>(Battery.getStats());`

schlimmchen, 2024-03-13:
You can't. Battery returns a `std::shared_ptr<BatteryStats>`, i.e., a pointer to the base class `BatteryStats`, and if you were to cast this to a `PylontechBatteryStats`, you would need some fancy `std::dynamic_cast` I guess... The reason I don't know is because I avoid stuff like this at all cost. We shouldn't use it or do it like that. And if we were, we would need to be sure that the instance is actually an instance of the derived `PylontechBatteryStats` before we attempt such a cast.

Your request is valid, of course, so lets work out a solution.

The goal is to know the value of `_chargeImmediately` in `HuaweiCanClass`. The value is generated in `PylontechCanReceiver` and then saved in `PylontechBatteryStats`.

We can't ask PylontechCanReceiver, as there is no public instance. Class `Battery` manages that instance in a private `std::unique_ptr<BatteryProvider>`.

That might sound cumbersome, but this kind of encapsulation, separation of concern, and ownership rules, saves us headaches in other contexts.

So I think the best way is to define a new method `bool needsCharging()` in the `BatteryStats` base class. In particular, this gives the other BatteryStats derived classes a chance to implement the same logic based on their battery provider implementation, i.e., the JK BMS can also implement this method and return true to facilitate this feature. Same for the SmartShunt, which might also want to trigger this feature.

This answer war quite verbose, but maybe you appreciate knowing my train of thought.

Please rebase onto 56353e4f :blush:

gitisgreat2023, 2024-03-14:
> > @MalteSchm 0% and 2% ain't good... yes we definitely need a feature coping with this. We should under all circumstances avoid that the inverter goes rogue and the PSU starts charging, creating a high power circular flow. If one is on holidays that could get dangerous.
> > > First time: 0%, yesterday 2%. I think the inverter did shut off at this stage but: Once the PSU started charging the inverter also started up again. You can reset ("Neustart") the inverter in the web-ui. This is accessible from the buttons in the main page / dashboard. Right next to the buttons to turn the inverter off and on
> > 
> > 
> > I'm stupid I guess: next to the turn on/off there's the grid info: ![image](https://private-user-images.githubusercontent.com/138098278/312000588-7328a5d3-9c08-4408-93d7-87791e6cc370.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDI3NDgsIm5iZiI6MTcxMDI0MjQ0OCwicGF0aCI6Ii8xMzgwOTgyNzgvMzEyMDAwNTg4LTczMjhhNWQzLTljMDgtNDQwOC05M2Q3LTg3NzkxZTZjYzM3MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTIwNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNzI0MmIxOGYyYTE0N2Y1OGQ2NDI1OTIxNjI3Y2VhZDE1YTVmZmMyMjUzZjhlNzYzZjhhYTBmNDEzMzgyN2M2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qJz275JcCCAB7I8zu-bTjX_M6y0AwXE1qZEnTE3PqOk)
> > ![image](https://private-user-images.githubusercontent.com/138098278/312000728-3411c8e0-27bf-4afc-8f1c-f437e67b466e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDI3NDgsIm5iZiI6MTcxMDI0MjQ0OCwicGF0aCI6Ii8xMzgwOTgyNzgvMzEyMDAwNzI4LTM0MTFjOGUwLTI3YmYtNGFmYy04ZjFjLWY0MzdlNjdiNDY2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTIwNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iODAzZjc3NmFmMDdhYmVkYTQ5YjM0ZTlkMGRkMjNjN2I3MjY2M2YxMTRmZGQ2N2QyMzRhNjQ5ZjBiM2Y3MGRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ebWHgmI98EShfwBP5KQES-XNcVW7mD4UyqZg1fVhxUs)
> > Or do you meant the on/off is the inverter reset?
> 
> Click on this on/off button. There is more...

Thanks!
As an interim solution, until these features are built in and released I now are able to access remotely my OpenDTU-on-Battery, for example when I'm not at home.
I activated on my Fritz!Box VPN and know I can easily control my setup remotely as well by phone.
Note: if the phone disconnects from WiFi and no LTE appears (for example on my Google Pixel) while trying to connect to the Fritz!Box VPN server, then force on the mobile data settings of the phone ipv4, see for example [here](https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7560/3618_Google-Pixel-kann-keine-VPN-Verbindung-zur-FRITZ-Box-herstellen/).

Linos88, 2024-03-14:
I also like the original idea with forced loading. This function can also be useful for a long service life in winter. Instead of using the SoC in the limit range, the voltage would be more suitable. The SoC can stop if there is only a small standby consumption and the voltage still continues to fall

Manos1966, 2024-03-14:
@MalteSchm Do you have the dtata stored from the time your inverter went rogue?
Can it be that the Efficiency % rating of the inverter went from 96% down to something around 60%?

This is the reason why I have a SHELLY 1PM connected to the 230V side of the inverter.... 😉

ottelo9, 2024-03-15:
> @MalteSchm Do you have the dtata stored from the time your inverter went rogue? Can it be that the Efficiency % rating of the inverter went from 96% down to something around 60%?
> 
> This is the reason why I have a SHELLY 1PM connected to the 230V side of the inverter.... 😉

Really good idea. I'll probably install a Zigbee relay right now.

CaCu15, 2024-03-17:
> You can't. Battery returns a `std::shared_ptr<BatteryStats>`, i.e., a pointer to the base class `BatteryStats`, and if you were to cast this to a `PylontechBatteryStats`, you would need some fancy [...]

@schlimmchen: 

I would like to extend this approach with the following suggestion: Currently the automatic charge control (Huawei automatic charge control) does NOT consider the feedback from the BMS during charging, especially the "charge currrent limitation" as provided by the Pylontech BMS (don't know abut the other BMSs). In order to support this, it would be good to have an additional method `float getChargeCurrentLimitation()` that returns the respective value returned by the BMS (and currently by REST call ` api/batterylivedata/status`). This would make this value available to the automatic power control in [HuaweiCanClass::loop()](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/src/Huawei_can.cpp) to limit the charge like this:

```
        // Set the actual output limit
        float efficiency =  (_rp.efficiency > 0.5 ? _rp.efficiency : 1.0); 
        float outputCurrent = efficiency * (newPowerLimit / _rp.output_voltage);
        
        // new code
        // assuming getChargeCurrentLimitation returns something > than any current supported by the Huawei, e.g. 100A 
        // if not supported by BMS 
        float chargeCurrentLimitation = Battery.getStats().getChargeCurrentLimitation(); 
        outputCurrent = std::min(outputCurrent, chargeCurrentLimitation);
```

MalteSchm, 2024-03-17:
@CaCu15 

I already have that implemented but still need to test it

MalteSchm, 2024-03-18:
Draft PR in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/767

schlimmchen, 2024-04-05:
I consider this closed as completed since #767 was merged into development and will be released eventually.

closed 2024-04-05 by schlimmchen