hoylabs/OpenDTU-OnBattery #724: Set DPL settings (especially "Upper power limit") via MQTT

Napizaki, 2024-03-07:
### Describe the solution you'd like

In addition to pull request [#677](https://github.com/helgeerbe/OpenDTU-OnBattery/pull/677), it would be good to be able to influence the "Upper power limit" of the DPL via MQTT to react to certain conditions (weather, load (e.g. electric car), ...).

cerise21, 2024-03-17:
It is already possible to modify the DPL upper power limit via POST to the Web API. For instructions see [here](https://www.akkudoktor.net/forum/postid/181338/) and [there](https://www.akkudoktor.net/forum/postid/182583/).
However this approach involves rewriting the config every time (which is probably bad in the long run (?)).

Thus I would appreciate it, if a means to modify the upper power limit (without a config rewrite) were implemented.

schlimmchen, 2024-04-05:
@Napizaki @cerise21 Do you consider settings other than the upper power limit to be important enough to be changed using MQTT? Since #677 there are overrides for all SoC and voltage thresholds.

AS every value that can be overridden in the future will have to be maintained forever, I would like to only add MQTT overrides for settings that people actually want to modify for good reasons. In particular, we'll need to define a respective topic, create and maintain the respective documentation, and need to display a line in the to-be-created DPL card of the live view which indicates the override to be active and with what value.

cerise21, 2024-04-07:
Up to now the upper power limit is the only DPL value that I (sometimes) want to change a couple of times per day to account for special energy usage scenarios (e. g. increase for temporary high loads when battery full or high irradiation; decrease for highly intermittent loads or at night time).

Napizaki, 2024-04-11:
As described above, I would personally be very interested in changing this value dynamically.
This would allow me, among other things, to charge my electric car with full inverter power as soon as it is plugged in. If it is not plugged in, the inverter power is reduced again to e.g. 600W.
For me personally, this would be a great added value.

schlimmchen, 2024-06-10:
Among you who subscribed to this issue: Is there nobody willing to implement this? This is low-hanging fruit :grin: The challenging bits and pieces are all already there, see `MqttHandlePowerLimiter.cpp`. It's about copy'n'pasting a couple lines of code and testing the changes. As a first step, no new DPL card is required in the live view... This is optional and is a second step.

schlimmchen, 2024-06-21:
Completed by #1047 and #1063.

closed 2024-06-21 by schlimmchen