hoylabs/OpenDTU-OnBattery #728: Live-Daten, Anzeige Leistung/Netz

Oldie-driver, 2024-03-08:
### What happened?

Live-Daten, die Anzeige Leistung/Netz wird nur aktualisiert, wenn ich das Browserfenster aktualisiere.

Die Anzeige Inverter Gesamtertrag heute und Inverter Gesamtleistung werden aktualisiert.
Auch die Aktuellen Limits vom HM-400 sowie darunter Allgemein, Phase 1 und der String.

### To Reproduce Bug

einfach OpenDTU mit aktiven Powerlimiter zuschauen und wechselnde AC Lasten haben ( Verbraucher einschalten ). Anzeige Leistung/Netz bleibt bis zur Aktualisierung des Fensters immer gleich trotz geänderter Limit Reglung. Beispiel -15,8W trotz Anpassung des Limits.
Gegenprobe am Stromzähler, ist nun ein anderer Wert.

### Expected Behavior

Ich vermute, die Aktualisierung der Anzeige des aktuellen Netzbezuges wurde im Code übersprungen. Den der Dynamic Power Limiter arbeitet einwandfrei. Habe ich mit Shelly 3EM pro im Browser gegengetestet.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.02.19.post1

IngoEF, 2024-03-08:
Hi,
at least in version **2024.02.19.post1** I confirm this unwanted behaviour.
- Ingo

Napizaki, 2024-03-08:
It works for me in the current version (2024.03.07)

Solarteur, 2024-03-08:
Einfach die aktuellsten Version aufspielen und testen - bei mir klappt das problemlos.

closed 2024-03-08 by Oldie-driver

IngoEF, 2024-03-08:
nope. 
Just compiled and installed newest version **2024.03.07**
No auto-update in Edge nor in Firefox.
The upper part stays un-updated until pressing "load again" in the browser options
while the lower display parts for inverter and victron smartsolar show actualized data all the time.

IngoEF, 2024-03-08:
Even different
[image]

Solarteur, 2024-03-08:
Two screenshots taken **without** (manually) reloading the page (2024.03.07, Firefox 123.0.1)

[image]
[image]

From my point of view everything is ok.